node-hid/node-hid #174: electron issues

khrome, 2016-12-22:
Hey there, I'm just trying to verify a library I write ( https://www.npmjs.com/package/barcode-scanner ) will work in electron, but deps are proving to be an issue.

Electron + HID installation attempts have been frustrating, I first tried all kinds of voodoo in my own project, but lost faith that it really had anything to do with that project (at very least I want to confirm node-hid would install in a vanilla project).

I then tried the `node-hid-toy` repo, which will not compile for me. Out of the box it complains rimraf is missing and then when I manually add that  I wind up fatal on compile. I've tried using the install hook, and manually triggering

Is there something I'm missing?

https://gist.github.com/khrome/eb9721e6ffeb740e371db8822c252d1d

todbot, 2016-12-23:
Hi @khrome,

A few questions to help us figure out what's going on:
1. It looks like you're running Node v4.4.7 on Mac OS X 10.11.6.  Is that correct?
2. Do you have Xcode installed?  If so, what version?
3. What version of Electron are you trying to build node-hid for?
4. Do you have `node-gyp` and `node-pre-gyp` globally? (or really, have you installed the prereqs in: https://github.com/node-hid/node-hid#compiling-from-source )

khrome, 2016-12-23:
I was, but have updated to current with electron + npm (1.4.13 & 6.5.0) at the 4th log. Before I was on (1.4.11 & 4.4.7). It seems like the compiler is either being given the wrong flags or is running with some strange or unexpected defaults. I am, indeed, on OSX 10.11.6. I have both `node-gyp` and `node-pre-gyp` installed globally and am running xcode 8.0 (8A218a).

Let me know if I can provide any other details!

todbot, 2016-12-23:
Hi!
You might try upgrading your node and then uninstalling & re-installing globally node-gyp & node-pre-gyp.  These are the tools responsible for actually doing compilation and they've been known to be finicky with versions of node & electron.

The fact that you're getting so many errors is indicative that something very essential is not getting set up correctly.  Both before and after doing the node upgrades mentioned above, what is output with the following commands:
```
which gcc
which make
which node
which node-gyp
which node-pre-gyp
```

Also, in a different directory, try building node-hid all by itself, by doing:
```
git clone https://github.com/node-hid/node-hid.git
cd node-hid                                        # must change into node-hid directory
git submodule update --init                        # done on publish automatically
npm install                                        # rebuilds the module
node-pre-gyp rebuild                               # rebuilds the C code
```

khrome, 2016-12-23:
~/NPMs/electron-hid-toy:daedalus>which gcc
/usr/bin/gcc
~/NPMs/electron-hid-toy:daedalus>which make
/usr/bin/make
~/NPMs/electron-hid-toy:daedalus>which node
/Users/khrome/.nvm/versions/node/v6.5.0/bin/node
~/NPMs/electron-hid-toy:daedalus>which node-gyp
/Users/khrome/.nvm/versions/node/v6.5.0/bin/node-gyp
~/NPMs/electron-hid-toy:daedalus>which node-pre-gyp
/Users/khrome/.nvm/versions/node/v6.5.0/bin/node-pre-gyp

I got the same output (added to the gist) from a direct clone & install of node-hid

Thanks!

todbot, 2016-12-23:
Dang that's weird in that it's prefectly normal. I'm on the same OS with the same version of node and node-hid compiles fine. Hmm, I'll keep looking around for other clues. Is your system configured interestingly in any way?
Oh, a few other things you can try for more info:
```
gcc --version
pwd         
```
The `pwd` is in the directory of your node-hid checkout, just to let me know if there are any weird characters in your path.  Some have reported issues with spaces or whatnot when using node-gyp.

khrome, 2016-12-23:
gcc --version
Configured with: --prefix=/Applications/Xcode.app/Contents/Developer/usr --with-gxx-include-dir=/usr/include/c++/4.2.1
Apple LLVM version 8.0.0 (clang-800.0.38)
Target: x86_64-apple-darwin15.6.0
Thread model: posix
InstalledDir: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin

`pwd` /Users/khrome/NPMs/node-hid

khrome, 2016-12-28:
Over the holidays I tried upgrading to Sierra, which had no effect. Continuing to investigate...

khrome, 2017-01-10:
Good news, I finally got `electron-hid-toy` working (sort of), unfortunately I cannot tell you where the issue arose (other than something was wrong with `cc`). After doing a clean install of Sierra, everything is compiling normally, though the script output is unexpected. After compiling I get `/build/{module_name}/{configuration}/HID.node` instead of `/build/HID/Release/HID.Node`... if I correct this, everything works normally.

I'll close this after I validate it in the parent project I was originally working on.

Thanks for the attention.

khrome, 2017-01-11:
OK, I can confirm everything except the compile path is working under an electron (under node the compile path seems fine) project and everything is working (as expected) in node.js proper.

Feel free to let this issue dangle for the `/build/{module_name}/{configuration}/HID.node` build issue or close this and open a new one... for now It's a wart I can live with.

danielmcq, 2017-01-15:
I have this issue, too. I noticed that in the `package.json` file under the "binary" section, it looks like this:
```
  "binary": {
      "module_name": "HID",
      "module_path": "./build/{module_name}/{configuration}/",
      "host": "https://github.com/node-hid/node-hid/releases/download/",
      "remote_path": "{version}"
  },
```

I have an electron app which depends on node-hid, and when I run it I get an error about `node_modules/node-hid/build/HID/Release/HID.node` not being found. When i change the `module_path` in `package.json` to `./build/HID/Release/`, my electron app starts up correctly. This may be a problem with npm because I'm assuming that `{module_name}` and `{configuration}` are supposed to replaced when the build script it ran. I don't know it hard-coding is the right way to resolve this issue, but it is one way to solve this problem without having to rename folders every time `npm install` is ran.

todbot, 2017-01-16:
Hi @danielmcq & @khrome, 
Thanks for your looking into this. I've spent the last few days looking into this and re-examining how I thought node-gyp and node-pre-gyp behave.  I think I discovered a fix to the configuration that addresses the issues you're seeing.  I'm still testing the different platforms and environments but I'm hopeful.  If you've been working from a checkout of node-hid and want to try it out, try pulling down the changes and see if they work for you.

danielmcq, 2017-01-16:
Thanks!
Yeah, I'll point my electron project to the repo and try a clean install. I'll report back with the results.

danielmcq, 2017-01-16:
@todbot The latest code in master seems to  have fixed the issue. Thank you for working on that. I'm still not experience with everything involved with node-gyp otherwise I would have tried doing a pull request for you. BTW, will you be bumping either the minor or patch version of node-hid now that you have this code in place?

todbot, 2017-01-16:
Cheers, @danielmcq.  Thanks for testing things out.

khrome, 2017-01-16:
Thanks! @todbot,  @danielmcq

closed 2017-01-16 by khrome