node-hid/node-hid #175: NPM install fails

Kwaadpepper, 2017-01-05:
I install libusb-dev, it used to work before, but few people reported me that it fails to install*; I installed it myself: 
node-hid@0.5.2 install: `node-pre-gyp install --fallback-to-build`

http://paste.ubuntu.com/23744923/

todbot, 2017-01-05:
Which Linux distro are you using?  
Do you have node-gyp and/or node-pre-gyp installed globally?  If so, which versions?

Also, I'm unclear, you were able to get it to install?

todbot, 2017-01-05:
I think the problem you are seeing is because the apt pkg `libusb-1.0-0-dev` is not installed.
Try doing `sudo apt install libusb-1.0-0-dev` and then reinstall your node project.

Kwaadpepper, 2017-01-07:
Sorry i'm using ubuntu 16.04.1
I have these packages installed matching ```libusb```

    libusb-0.1-4:amd64
    libusb-1.0-doc
    libusbmuxd2:amd64     
    libusb++-0.1-4v5:amd64
    libusb-dev
    libusbmuxd4:amd64     
    libusb-1.0-0:amd64
    libusb++-dev                                  
    libusb-1.0-0-dev:amd64
    libusbhid-common

todbot, 2017-01-08:
Unfortunately, npm appears to be very bad about relaying errors when compiling native modules.
Could you try again and copy the last 50 or so lines of what's in your terminal as link that as a gist/paste?

yonicstudios, 2017-01-10:
`rimraf` module seems to be missing as a dependency.
I installed it with `npm install rimraf` before installing `node-hid` and the installation worked fine.

foxindabox, 2017-01-11:
I am getting npm install errors on windows.
Below are the logs

D:\Dev\temp\test\node_modules\node-hid>if not defined npm_config_node_gyp (node "C:\Program Files\n
odejs\node_modules\npm\bin\node-gyp-bin\\..\..\node_modules\node-gyp\bin\node-gyp.js" rebuild )  el
se (node "" rebuild )
Building the projects in this solution one at a time. To enable parallel build, please add the "/m"
 switch.
C:\Program Files (x86)\MSBuild\Microsoft.Cpp\v4.0\V140\Platforms\x64\PlatformToolsets\v140\Toolset
.targets(36,5): error MSB8036: The Windows SDK version 8.1 was not found. Install the required ver
sion of Windows SDK or change the SDK version in the project property pages or by right-clicking t
he solution and selecting "Retarget solution". [D:\Dev\temp\test\node_modules\node-hid\build\hidap
i.vcxproj]
gyp ERR! build error
gyp ERR! stack Error: `C:\Program Files (x86)\MSBuild\14.0\bin\msbuild.exe` failed with exit code:
1
gyp ERR! stack     at ChildProcess.onExit (C:\Program Files\nodejs\node_modules\npm\node_modules\no
de-gyp\lib\build.js:276:23)
gyp ERR! stack     at emitTwo (events.js:106:13)
gyp ERR! stack     at ChildProcess.emit (events.js:191:7)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:215:12)
gyp ERR! System Windows_NT 6.1.7601
gyp ERR! command "C:\\Program Files\\nodejs\\node.exe" "C:\\Program Files\\nodejs\\node_modules\\np
m\\node_modules\\node-gyp\\bin\\node-gyp.js" "rebuild"
gyp ERR! cwd D:\Dev\temp\test\node_modules\node-hid
gyp ERR! node -v v6.5.0
gyp ERR! node-gyp -v v3.3.1
gyp ERR! not ok
npm WARN barcode@1.0.0 No description
npm WARN barcode@1.0.0 No repository field.
npm ERR! Windows_NT 6.1.7601
npm ERR! argv "C:\\Program Files\\nodejs\\node.exe" "C:\\Program Files\\nodejs\\node_modules\\npm\\
bin\\npm-cli.js" "install" "node-hid" "--save"
npm ERR! node v6.5.0
npm ERR! npm  v3.10.3
npm ERR! code ELIFECYCLE

npm ERR! node-hid@0.4.1 install: `node-gyp rebuild`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the node-hid@0.4.1 install script 'node-gyp rebuild'.
npm ERR! Make sure you have the latest version of node.js and npm installed.
npm ERR! If you do, this is most likely a problem with the node-hid package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-gyp rebuild
npm ERR! You can get information on how to open an issue for this project with:
npm ERR!     npm bugs node-hid
npm ERR! Or if that isn't available, you can get their info via:
npm ERR!     npm owner ls node-hid
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     D:\Dev\temp\test\npm-debug.log

todbot, 2017-01-11:
Hi @foxindabox,
The error you're seeing is since there isn't a pre-built native binary for that particular combo of OS & node version, npm is building node-hid from source, and you don't have a C compiler installed.

As a workaround, try installing the prereqs listed in: 
https://github.com/node-hid/node-hid#compiling-from-source
Specifically for Windows, you should be able to do:
* Run `npm install --global windows-build-tools` (in Admin PowerShell)
* Add `%USERPROFILE%\.windows-build-tools\python27` to PATH in Environment Variables
And then try doing `npm install node-hid` again.

foxindabox, 2017-01-11:
Hi @todbot 
I have already followed the above two steps.
Still I am facing this issue.

Kwaadpepper, 2017-01-11:
Here is the output log http://paste.ubuntu.com/23782360/

todbot, 2017-01-11:
Hi @Kwaadpepper, 
Your issue isn't the same as @foxindabox.  I think your issue is related to the error "Aucun fichier ou dossier de ce type"  ("no such file or folder"?)  For some reason npm cannot find the "node" program.  It seems like there is something weird in your environment and "node" isn't in your path.

Kwaadpepper, 2017-01-12:
Well i did ```sudo ln -s /usr/bin/nodejs /usr/bin/node```

it installed sucessfully but, there were small errors : 

    node-pre-gyp ERR! Tried to download(404): https://github.com/node-hid/node-hid/releases/download/0.5.2/HID-v0.5.2-node-v46-linux-x64.tar.gz 
    node-pre-gyp ERR! Pre-built binaries not found for node-hid@0.5.2 and node@4.2.6 (node-v46 ABI) (falling back to source compile with node-gyp)

6pm, 2017-01-14:
404 error:
````
Tried to download(404): https://github.com/node-hid/node-hid/releases/download/0.5.2/HID-v0.5.2-node-v48-linux-x64.tar.gz 

````

todbot, 2017-01-14:
Hi @6pm, that is a warning from npm (specifically the node-pre-gyp tool) as it's looking for a pre-built binary. If it doesn't find one (and there often won't be especially for Linux or newer versions of Node), then npm will use the node-gyp tool to build node-hid for your platform.

6pm, 2017-01-14:
Hello @todbot. I found solution, need install: 
````
sudo apt-get install libusb-1.0-0-dev
````
Thank you for your package!

todbot, 2017-01-29:
I believe this has been fixed in the latest release.

closed 2017-01-29 by todbot