node-hid/node-hid #176: node-hid install on Windows 10 64 bit fails

rtomkins, 2017-01-06:
I require some guidance.
Windows 10, 64 bit, fresh install of node.js version 7.4.0 Current, fresh install of NetBeans 8.2 all.
I confirmed that node.js is installed and works using HelloWorld.js on NetBeans.
I used npm install hid and was unsure if anything actually installed, the version information was 0. I used npm uninstall hid.
I used npm install node-hid and there was a flurry of activity and then errors thrown about no Pythin being being found and more.
I suspect the node-hid install failure means that there is no libusb binary for my Operating System, but my lack of knowledge prevents me from definitively knowing this to be true.

1. Would I get a different end result were I to uninstall node.js and reinstall with node.js version 6.9.4 LTS, in other words, would node-hid then successfully install?
2. I do not want to install python and Visual Studio just to compile libusb, is there an alternative method to facilitate the installation and use of node-hid?
3. Is there an alternative to node-hid that would enable me to communicate over USB with a HID Device Class?

todbot, 2017-01-06:
Hi,
I think the problem is there isn't a precompiled binary of node-hid for Node 7.x.  So node-hid is trying to compile from source and you don't have a C compiler installed.
This goes into more detail about setting up your system:
  https://github.com/node-hid/node-hid#compiling-from-source
In your case you should be able to get by doing:
* Run `npm install --global windows-build-tools` from an Admin Powershell
* Add `%USERPROFILE%\.windows-build-tools\python27` to your PATH Environment variable

rtomkins, 2017-01-07:
I did not want to install all the tools, so I uninstalled node.js v7.4 and installed node.js v6.9.4 and then installed node-hid. This was successful and then I ran show.devices.js and this was also successful.
I may yet go to the effort to make node.js v7.4 work.

closed 2017-01-07 by rtomkins