node-hid/node-hid #177: Error Opening Device

donsingh, 2017-01-24:
My app:
`"use strict";
var HID = require('node-hid');
console.log('devices:', HID.devices());
var device = new HID.HID(1137,41234);
`
And the resulting console output:

`"use strict";
var HID = require('node-hid');
console.log('devices:', HID.devices());
var device = new HID.HID(1137,41234);
`

todbot, 2017-01-24:
Hi, 
Please provide more details about your setup, like:
1. What OS are you on?
2. What version of Node.js are you running?
3. What specific USB HID device are you attempting to open?
4. What is the output of `HID.devices()`?
5. If Linux, are you running as 'root' user or standard user?

donsingh, 2017-01-25:
running my app as root fixed the problem. I tried adding a new rule to `/etc/udev/rules.d` called `81-hid.rules` and ran `sudo udevadm control --reload-rules `but it seems it didnt solved my problem.

-Mint 18
-Node 7.4
-Trying to open a RFID read/write device
-HID.devices() output:
```
devices: [ { 
    vendorId: 1137,
    productId: 41234,
    path: '0001:0008:00',
    release: 1,
    interface: 0 } ]
```

-Running as root made it work. How do i give node permission?

todbot, 2017-01-25:
Hi @donsingh,
I'm not familiar with Mint and how it differs from Ubuntu & Fedora, but it looks like it does use udev, so the config should be very similar.  Do a search around for "Mint 18 udev usb hid" to see if others using gamepads or similar have solved the problem.

As a test, you could try using these rules:
https://github.com/todbot/blink1/blob/master/linux/51-blink1.rules
but replace the 'idVendor' and 'idProduct' values with those for your device.

donsingh, 2017-01-26:
This seems to be an issue with Mint 18. I will try to ask help from their end.

closed 2017-01-26 by donsingh

gcback, 2019-08-11:
Hi..
I executed the sample code presented. 

> my environment
 - pc, windows 10
 - nodejs : v10.15.3
 - run: CMD, PowerShell (in superuser)
 - scanner : HID-USB scanner (vendorId: 59473, productId: 8704)

The result is failure as follows. Can I get a solution or advice for this failure?
---
 { vendorId: 59473,
    productId: 8704,
    path:
     '\\\\?\\hid#vid_e851&pid_2200#6&5fa295d&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}\\kbd',
    manufacturer: 'Manufacturer',
    product: 'Barcode Reader',
    release: 563,
    interface: -1,
    usagePage: 1,
    usage: 6 } ]
C:\Users\gcbac\Projects\git\qr-scanner\node_modules\node-hid\nodehid.js:49
    this._raw = new (Function.prototype.bind.apply(binding.HID,
                ^

Error: cannot open device with vendor id 0xe851 and product id 0x2200
    at new HID (C:\Users\gcbac\Projects\git\qr-scanner\node_modules\node-hid\nodehid.js:49:17)
    at new UsbScanner (C:\Users\gcbac\Projects\git\qr-scanner\node_modules\usb-barcode-scanner\dist\usb-barcode-scanner.js:31:25)
    at Object.<anonymous> (C:\Users\gcbac\Projects\git\qr-scanner\index.js:11:15)
    at Module._compile (internal/modules/cjs/loader.js:701:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:712:10)
    at Module.load (internal/modules/cjs/loader.js:600:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:539:12)
    at Function.Module._load (internal/modules/cjs/loader.js:531:3)
    at Function.Module.runMain (internal/modules/cjs/loader.js:754:12)
    at startup (internal/bootstrap/node.js:283:19)

todbot, 2019-08-12:
Hi @gcback,  
If this USB barcode scanner pretends to be a keyboard, then you cannot use `node-hid` to control it.  Just treat it like a keyboard.

koo9, 2019-10-30:
@todbot seeing the same error here. on the device path, it ends with \kbd, does that mean the scanner act as a keyboard?

JeffersonJiao, 2020-01-22:
Any solution for this? I need something that reads a bar code in node server side.

todbot, 2020-01-23:
If the device is acting like a keyboard, use any common way to read keystrokes.