node-hid/node-hid #179: Serial Number, Manufacturer and Product  missing on linux

rdepena, 2017-01-28:
Serial Number, Manufacturer and Product when getting the device list on linux:

code:

```javascript
var hid = require('node-hid');
hid.devices();
```
Node-hid 0.5.4 on ubuntu 16.10

Tested 0.5.2 and the data is there.

rdepena, 2017-01-28:
a bit more information: 

0.5.2 behaves the same way if I do not pass --driver=hidraw during install, but 0.5.4 does not work even with the hidraw flag.

todbot, 2017-01-28:
Hi,
I believe that information is only available to the root user (or if you've given udev perms to the devices in question)

I'm unclear what issues you're having on Ubuntu 16.10.  I test regularly on that and without hidraw.  Could you give more details as to the issues you're seeing?

mediaslav, 2017-01-28:
Same issue here, Ubuntu 16 and Node-hid 0.5.4. That's all data I see:
```
[
  {
    "vendorId": 3599,
    "productId": 3,
    "path": "0002:0002:00",
    "release": 259,
    "interface": 0
  }
]
```

rdepena, 2017-01-29:
Just another bit of info: 

Same code on 0.5.2  installed with hidraw works, but under 0.5.4 I get the following error:

```javascript
Error: cannot open device with vendor id 0x54c and product id 0x9cc
    at Error (native)
    at new HID (/home/rdepena/Documents/Projects/node-hid-test-basic/node_modules/node-hid/nodehid.js:28:17)
    at repl:1:18
    at sigintHandlersWrap (vm.js:22:35)
    at sigintHandlersWrap (vm.js:96:12)
    at ContextifyScript.Script.runInThisContext (vm.js:21:12)
    at REPLServer.defaultEval (repl.js:346:29)
    at bound (domain.js:280:14)
    at REPLServer.runBound [as eval] (domain.js:293:12)
    at REPLServer.<anonymous> (repl.js:545:10)
```

It's almost as it is ignoring the hidraw flag.

rdepena, 2017-01-29:
Looks like the driver flag is being ignored, I hard coded the driver to hidraw in the [binding.gyp](https://github.com/node-hid/node-hid/blob/master/binding.gyp#L3) file, ran `node-pre-gyp rebuild` and everything started working.

todbot, 2017-01-29:
Hi @rdepena, what is the exact command you are using to install `node-hid`?  If you are having problems with the default pre-built binary for 0.5.4, then you will need to both specify you want to build from source and use a different driver.  To do this when installing, type:

```
npm install node-hid --build-from-source --driver=hidraw
```

Apologies if that wasn't clear.  I'll update the README. The way npm deals with binary modules isn't always that clear.

Also, what is the error you are seeing when just doing `npm install node-hid` in a blank directory?  How is it not working for you?  I do much testing on fresh installs of Ubuntu 16.10 and haven't seen any compilation or functional problems. (As mentioned previously, on Linux you won't be able to get manufacturer and similar strings without either running as 'root' user (via sudo) or giving your user permission via udev)

rdepena, 2017-01-29:
@todbot adding the --build-from-source works, I guess something in the install changed from 0.5.2 to 0.5.4. There was no additional error, everything was related to missing hidraw support (I have the udev rules setup and 0.5.2 works without the build from source flag). 

Just as an additional question; how do I specify the driver for an electron build ?

Thank you!

mediaslav, 2017-01-29:
Look for this output of "node-hid" and "usb-detection" libs on same machine with same rights etc. I believe both them use libusb but pulling different amount of data. Hope that helps.

"node-hid":
```    
        {
            "vendorId": 3599,
            "productId": 3,
            "path": "0002:0002:00",
            "release": 259,
            "interface": 0
        }
```
 "usb-detection": 
```   
        {
            "locationId": 0,
            "vendorId": 7531,
            "productId": 2,
            "deviceName": "EHCI Host Controller",
            "manufacturer": "Linux 4.4.0-59-generic ehci_hcd",
            "serialNumber": "0000:02:03.0",
            "deviceAddress": 0
        },
        {
            "locationId": 0,
            "vendorId": 7531,
            "productId": 1,
            "deviceName": "UHCI Host Controller",
            "manufacturer": "Linux 4.4.0-59-generic uhci_hcd",
            "serialNumber": "0000:02:00.0",
            "deviceAddress": 0
        },
        {
            "locationId": 0,
            "vendorId": 3599,
            "productId": 3,
            "deviceName": "VMware Virtual USB Mouse",
            "manufacturer": "VMware",
            "serialNumber": "",
            "deviceAddress": 0
        },
        {
            "locationId": 0,
            "vendorId": 3599,
            "productId": 2,
            "deviceName": "VMware Virtual USB Hub",
            "manufacturer": "",
            "serialNumber": "",
            "deviceAddress": 0
        }
```

todbot, 2017-01-29:
Hi @mediaslav,
By default node-hid on Linux uses hidapi's 'libusb' driver. libusb requires the device to be opened to get that information and you can only open the device if you have permission (or run as 'root' via sudo).  You can change to hidapi's 'hidraw' driver with `--driver=hidraw` as I mention above if you want that information.  Currently, node-hid defaults to 'libusb' instead of 'hidraw' because until recently, the critical HID APIs to send and get Feature reports was missing in hidraw.  It may time to default Linux to hidraw.  I've opened issue #180 to track this change.

todbot, 2017-01-29:
Hi rdepena,  I still don't understand.  Please post a gist of exactly the output of:
```
npm install node-hid
```
when done in a complete empty directory.  Thanks!

rdepena, 2017-01-30:
Hey @todbot 

There was no installation error, I did not know it at the moment but it boiled down to:

adding the `--driver=hidraw` flag on the install was not enough to get hidraw support.

That was basically the issue I was having but now by passing the `--build-from-source` flag everything works perfect.

Thank you.

todbot, 2019-12-05:
I think this works with modern node-hid

closed 2019-12-05 by todbot