node-hid/node-hid #181: Can't see bluetooth device on Raspberry Pi

Mlgarrido, 2017-02-12:
I know that armv7 is not fully supported, but is works partially (using USB). 

On Raspberry Pi I can't see the device (PS3 Dualshock controller). The device is paired and connected with sixad (1 red led is enabled by default and I can see /dev/input/jsX /dev/input/eventX ...) but I can't list if I create a simple script using node-hid (only see USB keyboard and mouse).

With USB connected I can see and communicate with the device, but with BT connected I can't see the device in list.

Thx!

todbot, 2017-02-14:
Hi,
node-hid on RasPi works well, it's just that we can't produce pre-build binaries easily.  I should change the README text to reflect this.

Bluetooth devices aren't handled by node-hid on Linux by default.  There are two ways to interface with HID devices on Linux: "libusb" (older, more complete, but USB only) and "hidraw" (newer, has issues with older kernels, but BT should work).

You can enable the hidraw driver by doing 

```
npm install node-hid --driver=hidraw
```

Let me know if that works for you.

Mlgarrido, 2017-02-14:
Hi! 

With hidraw doesn't work either. I can only list hidraw0-1, my keyboard and mouse USB, but I cant see bluetooth devices. 

Do I need anything else so that my devices will mount in hidraw?

Thx

todbot, 2017-02-14:
Hi,
No clue. I have zero experience with using Bluetooth HID devices with hidraw or node-hid.  But I have had a lot of headaches trying to get Bluetooth to work with Linux.

For the comments above I made, I  was essentially parroting what's in the HIDAPI README: https://github.com/signal11/hidapi/  node-hid is mostly just a wrapper to HIDAPI.  I have experience with hidraw but only for USB devices and those seem to "just work" as long as the kernel has hidraw support.

I would investigate finding out if your RasPi can even see Bluetooth devices.  Maybe one of these links will help:
https://www.raspberrypi.org/learning/robo-butler/bluetooth-setup/
http://www.makeuseof.com/tag/setup-wi-fi-bluetooth-raspberry-pi-3/

Mlgarrido, 2017-02-14:
Thanks, I will try to install everything from scratch again. I think the problem is the original hidapi. I'll keep investigating and I'll tell you if I can make it work.

Meanwhile i will use the usb.

Thx for your help and your awesome lib!

closed 2017-02-14 by Mlgarrido