node-hid/node-hid #182: Dualshock 4 Bluetooth Can Not Write to HID Device

shubshub, 2017-02-17:
Trying to write to HID Device (Dualshock 4) That is connected via bluetooth doesn't seem to work it just returns an error

When I connect the Dualshock 4 via USB it works perfectly although from what I have logged I can't see much difference in how they are seen by the program

What am I doing wrong? And how can I go about fixing it :) Thanks

shubshub, 2017-02-17:
Windows 10 64 Bit
Node JS 5.4.1
node-hid 0.5.4

todbot, 2017-02-17:
Hmm... not sure. I personally have not tested any Bluetooth HID devices with node-hid (or HIDAPI, the C library node-hid wraps)

Two questions:
1. Does the controller work in Bluetooth mode otherwise? (like with games and such)
2. Is the controller listed if you do (in a clean directory):

```
npm install node-hid
node node_modules\node-hid\src\show-devices.js
```

shubshub, 2017-02-17:
Also yes the controller does work regulary and 
[image]

(Left side is Bluetooth Right side is USB)
I took a picture of the HID Information for both Bluetooth connection and USB Connection

`node_modules\node-hid\src\show-devices.js
module.js:422
  return process.dlopen(module, path._makeLong(filename));
                 ^

Error: The specified module could not be found.
\\?\C:\Users\Christopher\Desktop\Node JS Stuff\Dualshock\nwjs-sdk-v0.20.2-win-x64\data\node_modules\dualshock-controller\node_modules\node-hid\build\Release\HID.node
    at Error (native)
    at Object.Module._extensions..node (module.js:422:18)
    at Module.load (module.js:343:32)
    at Function.Module._load (module.js:300:12)
    at Module.require (module.js:353:17)
    at require (internal/module.js:12:17)
    at Object.<anonymous> (C:\Users\Christopher\Desktop\Node JS Stuff\Dualshock\nwjs-sdk-v0.20.2-win-x64\data\node_modules\dualshock-controller\node_modules\node-hid\nodehid.js:9:15)
    at Module._compile (module.js:397:26)
    at Object.Module._extensions..js (module.js:404:10)
    at Module.load (module.js:343:32)`

shubshub, 2017-02-17:
Everything about the controller works All the buttons work and do events and stuff, I just can't change the LED or Set the Motors with node-hid Like I was able to in USB Mode

todbot, 2017-02-17:
Ah, so it does work for you in Bluetooth mode, just not completely, in that you cannot set the LEDs & motor.  This might be a known issue with DS4 controllers.  Have you tried the `ds4` package? https://www.npmjs.com/package/ds4

shubshub, 2017-02-17:
I would try it but it doesn't seem to be getting updates anymore, The package I'm using right now is still active, But it seems to send data in the exact same way

shubshub, 2017-02-17:
Could it be something to do with the release number being 0?

todbot, 2017-02-17:
Release number of what?  And what package are you using if not `ds4`?

shubshub, 2017-02-17:
I am using the dualshock-controller package by rdepena
https://github.com/rdepena/node-dualshock-controller

shubshub, 2017-02-17:
I'm guessing node-hid just doesnt work with bluetooth devices? Unless its a problem with the way you wrote the write function in HID.cc?
Sorry if I am being annoying D;

shubshub, 2017-02-17:
Okay so I got the Can't Write to HID device error to go away, I guess I'll just keep playing around with the buffer locations until it works

todbot, 2017-02-17:
Sorry, you'll need to be more explicit in what you are trying to do and what errors you are seeing.

From what I can gather, your situation is:
- Win10 64bit (Professional? Home? International?)
- Node.js 5.4.1 (installed via download? installed via nvm?)
- `node-hid@0.5.4`  (installed by hand? as a dep to a package? recompiled from source? if so, what version of Visual Studio?)
- `dualshock-controller` on top of `node-hid` (what version? installed via npm or source?)
- Sony Dualshock 4 controller via both USB and Bluetooth

And then what you are experiencing is:
- DS4 controller on USB works to both receive controller events and set controller state (e.g. `controller.setExtras()` in `dualshock-controller` package
- DS4 controller on Bluetooth works to receive controller events but CANNOT set controller state

So then, questions:
1. Is that a correct summary?
2. Is there anything in the unknowns above that could conceivably cause the issues you're seeing?
3. Have you searched on Google for others having problems setting DS4 controller state via Bluetooth, regardless of if it's using node-hid or not?
4. What *exactly* are you trying to do that you are having issues with?
5. Do you have exact text output of the errors you are experiencing?
6. I have no idea what you mean by "playing around with the buffer locations".  Please clarify.

It is very difficult to remotely diagnose a problem on an unknown configuration about an ill-described problem.  The more information you can give, the easier it will be to figure out what's going on.

shubshub, 2017-02-17:
Its okay, Sorry to have been a bother it was a problem with the buffer While I got the error to go away, I still need to figure out the rest of the buffer locations for the data I'm trying to write

todbot, 2017-02-19:
This is a dupe of #184 and is not an issue with node-hid.  See rdepena/node-dualshock-controller#64 for further development on this issue.

closed 2017-02-19 by todbot

Pecacheu, 2018-01-11:
Think I may have found the issue (though I can't solve it). I believe it is indeed a problem with node-hid, see here: https://github.com/rdepena/node-dualshock-controller/issues/64#issuecomment-357082104

(Putting this here for Goggle search result purposes)