node-hid/node-hid #183: Cannot install node-hid on OSMC

moskovskiy82, 2017-02-17:
This is the log i am getting

#npm install -g node-hid
```
/usr/bin/hid-showdevices -> /usr/lib/node_modules/node-hid/src/show-devices.js

> node-hid@0.5.4 install /usr/lib/node_modules/node-hid
> node-pre-gyp install --fallback-to-build

node-pre-gyp ERR! Tried to download(404): https://github.com/node-hid/node-hid/releases/download/0.5.4/HID-v0.5.4-node-v48-linux-arm.tar.gz
node-pre-gyp ERR! Pre-built binaries not found for node-hid@0.5.4 and node@6.9.5 (node-v48 ABI) (falling back to source compile with node-gyp)
gyp WARN EACCES user "root" does not have permission to access the dev dir "/root/.node-gyp/6.9.5"
gyp WARN EACCES attempting to reinstall using temporary dev dir "/usr/lib/node_modules/node-hid/.node-gyp"
make: Entering directory '/usr/lib/node_modules/node-hid/build'
make: *** No rule to make target '../.node-gyp/6.9.5/include/node/common.gypi', needed by 'Makefile'.  Stop.
make: Leaving directory '/usr/lib/node_modules/node-hid/build'
gyp ERR! build error
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/lib/node_modules/node-gyp/lib/build.js:276:23)
gyp ERR! stack     at emitTwo (events.js:106:13)
gyp ERR! stack     at ChildProcess.emit (events.js:191:7)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:215:12)
gyp ERR! System Linux 4.4.27-7-osmc
gyp ERR! command "/usr/bin/nodejs" "/usr/lib/node_modules/node-gyp/bin/node-gyp.js" "build" "--fallback-to-build" "--module=/usr/lib/node_modules/node-hid/build/Release/HID.node" "--module_name=HID" "--module_path=/usr/lib/node_modules/node-hid/build/Release"
gyp ERR! cwd /usr/lib/node_modules/node-hid
gyp ERR! node -v v6.9.5
gyp ERR! node-gyp -v v3.5.0
gyp ERR! not ok
node-pre-gyp ERR! build error
node-pre-gyp ERR! stack Error: Failed to execute '/usr/bin/nodejs /usr/lib/node_modules/node-gyp/bin/node-gyp.js build --fallback-to-build --module=/usr/lib/node_modules/node-hid/build/Release/HID.node --module_name=HID --module_path=/usr/lib/node_modules/node-hid/build/Release' (1)
node-pre-gyp ERR! stack     at ChildProcess.<anonymous> (/usr/lib/node_modules/node-hid/node_modules/node-pre-gyp/lib/util/compile.js:83:29)
node-pre-gyp ERR! stack     at emitTwo (events.js:106:13)
node-pre-gyp ERR! stack     at ChildProcess.emit (events.js:191:7)
node-pre-gyp ERR! stack     at maybeClose (internal/child_process.js:877:16)
node-pre-gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:226:5)
node-pre-gyp ERR! System Linux 4.4.27-7-osmc
node-pre-gyp ERR! command "/usr/bin/nodejs" "/usr/lib/node_modules/node-hid/node_modules/.bin/node-pre-gyp" "install" "--fallback-to-build"
node-pre-gyp ERR! cwd /usr/lib/node_modules/node-hid
node-pre-gyp ERR! node -v v6.9.5
node-pre-gyp ERR! node-pre-gyp -v v0.6.31
node-pre-gyp ERR! not ok
Failed to execute '/usr/bin/nodejs /usr/lib/node_modules/node-gyp/bin/node-gyp.js build --fallback-to-build --module=/usr/lib/node_modules/node-hid/build/Release/HID.node --module_name=HID --module_path=/usr/lib/node_modules/node-hid/build/Release' (1)
npm ERR! Linux 4.4.27-7-osmc
npm ERR! argv "/usr/bin/nodejs" "/usr/bin/npm" "install" "-g" "node-hid"
npm ERR! node v6.9.5
npm ERR! npm  v3.10.10
npm ERR! code ELIFECYCLE

npm ERR! node-hid@0.5.4 install: `node-pre-gyp install --fallback-to-build`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the node-hid@0.5.4 install script 'node-pre-gyp install --fallback-to-build'.
npm ERR! Make sure you have the latest version of node.js and npm installed.
npm ERR! If you do, this is most likely a problem with the node-hid package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-pre-gyp install --fallback-to-build
npm ERR! You can get information on how to open an issue for this project with:
npm ERR!     npm bugs node-hid
npm ERR! Or if that isn't available, you can get their info via:
npm ERR!     npm owner ls node-hid
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!    

npm-debug.log

5061 error Linux 4.4.27-7-osmc
5062 error argv "/usr/bin/nodejs" "/usr/bin/npm" "install" "-g" "node-hid"
5063 error node v6.9.5
5064 error npm  v3.10.10
5065 error code ELIFECYCLE
5066 error node-hid@0.5.4 install: `node-pre-gyp install --fallback-to-build`
5066 error Exit status 1
5067 error Failed at the node-hid@0.5.4 install script 'node-pre-gyp install --fallback-to-build'.
5067 error Make sure you have the latest version of node.js and npm installed.
5067 error If you do, this is most likely a problem with the node-hid package,
5067 error not with npm itself.
5067 error Tell the author that this fails on your system:
5067 error     node-pre-gyp install --fallback-to-build
5067 error You can get information on how to open an issue for this project with:
5067 error     npm bugs node-hid
5067 error Or if that isn't available, you can get their info via:
5067 error     npm owner ls node-hid
5067 error There is likely additional logging output above.
5068 verbose exit [ 1, true ]

```

moskovskiy82, 2017-02-17:
Same with
 npm install -g node-hid --build-from-source
```
/usr/bin/hid-showdevices -> /usr/lib/node_modules/node-hid/src/show-devices.js

> node-hid@0.5.4 install /usr/lib/node_modules/node-hid
> node-pre-gyp install --fallback-to-build

gyp WARN EACCES user "root" does not have permission to access the dev dir "/root/.node-gyp/6.9.5"
gyp WARN EACCES attempting to reinstall using temporary dev dir "/usr/lib/node_modules/node-hid/.node-gyp"
make: Entering directory '/usr/lib/node_modules/node-hid/build'
make: *** No rule to make target '../.node-gyp/6.9.5/include/node/common.gypi', needed by 'Makefile'.  Stop.
make: Leaving directory '/usr/lib/node_modules/node-hid/build'
gyp ERR! build error
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/lib/node_modules/node-gyp/lib/build.js:276:23)
gyp ERR! stack     at emitTwo (events.js:106:13)
gyp ERR! stack     at ChildProcess.emit (events.js:191:7)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:215:12)
gyp ERR! System Linux 4.4.27-7-osmc
gyp ERR! command "/usr/bin/nodejs" "/usr/lib/node_modules/node-gyp/bin/node-gyp.js" "build" "--fallback-to-build" "--module=/usr/lib/node_modules/node-hid/build/Release/HID.node" "--module_name=HID" "--module_path=/usr/lib/node_modules/node-hid/build/Release"
gyp ERR! cwd /usr/lib/node_modules/node-hid
gyp ERR! node -v v6.9.5
gyp ERR! node-gyp -v v3.5.0
gyp ERR! not ok
node-pre-gyp ERR! build error
node-pre-gyp ERR! stack Error: Failed to execute '/usr/bin/nodejs /usr/lib/node_modules/node-gyp/bin/node-gyp.js build --fallback-to-build --module=/usr/lib/node_modules/node-hid/build/Release/HID.node --module_name=HID --module_path=/usr/lib/node_modules/node-hid/build/Release' (1)
node-pre-gyp ERR! stack     at ChildProcess.<anonymous> (/usr/lib/node_modules/node-hid/node_modules/node-pre-gyp/lib/util/compile.js:83:29)
node-pre-gyp ERR! stack     at emitTwo (events.js:106:13)
node-pre-gyp ERR! stack     at ChildProcess.emit (events.js:191:7)
node-pre-gyp ERR! stack     at maybeClose (internal/child_process.js:877:16)
node-pre-gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:226:5)
node-pre-gyp ERR! System Linux 4.4.27-7-osmc
node-pre-gyp ERR! command "/usr/bin/nodejs" "/usr/lib/node_modules/node-hid/node_modules/.bin/node-pre-gyp" "install" "--fallback-to-build"
node-pre-gyp ERR! cwd /usr/lib/node_modules/node-hid
node-pre-gyp ERR! node -v v6.9.5
node-pre-gyp ERR! node-pre-gyp -v v0.6.31
node-pre-gyp ERR! not ok
Failed to execute '/usr/bin/nodejs /usr/lib/node_modules/node-gyp/bin/node-gyp.js build --fallback-to-build --module=/usr/lib/node_modules/node-hid/build/Release/HID.node --module_name=HID --module_path=/usr/lib/node_modules/node-hid/build/Release' (1)
npm ERR! Linux 4.4.27-7-osmc
npm ERR! argv "/usr/bin/nodejs" "/usr/bin/npm" "install" "-g" "node-hid" "--build-from-source"
npm ERR! node v6.9.5
npm ERR! npm  v3.10.10
npm ERR! code ELIFECYCLE

npm ERR! node-hid@0.5.4 install: `node-pre-gyp install --fallback-to-build`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the node-hid@0.5.4 install script 'node-pre-gyp install --fallback-to-build'.
npm ERR! Make sure you have the latest version of node.js and npm installed.
npm ERR! If you do, this is most likely a problem with the node-hid package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-pre-gyp install --fallback-to-build
npm ERR! You can get information on how to open an issue for this project with:
npm ERR!     npm bugs node-hid
npm ERR! Or if that isn't available, you can get their info via:
npm ERR!     npm owner ls node-hid
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /room/room-assistant/npm-debug.log
root@osmc:/room/room-assistant# cd /root/.node-gyp/6.9.5
root@osmc:~/.node-gyp/6.9.5# cd ..
root@osmc:~/.node-gyp# ls
6.9.5
root@osmc:~/.node-gyp# chmod 777 6.9.5/
root@osmc:~/.node-gyp# npm install -g node-hid --build-from-source
/usr/bin/hid-showdevices -> /usr/lib/node_modules/node-hid/src/show-devices.js

[... 34 lines cut ...]
npm ERR! Linux 4.4.27-7-osmc
npm ERR! argv "/usr/bin/nodejs" "/usr/bin/npm" "install" "-g" "node-hid" "--build-from-source"
npm ERR! node v6.9.5
npm ERR! npm  v3.10.10
npm ERR! code ELIFECYCLE

npm ERR! node-hid@0.5.4 install: `node-pre-gyp install --fallback-to-build`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the node-hid@0.5.4 install script 'node-pre-gyp install --fallback-to-build'.
npm ERR! Make sure you have the latest version of node.js and npm installed.
npm ERR! If you do, this is most likely a problem with the node-hid package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-pre-gyp install --fallback-to-build
npm ERR! You can get information on how to open an issue for this project with:
npm ERR!     npm bugs node-hid
npm ERR! Or if that isn't available, you can get their info via:
npm ERR!     npm owner ls node-hid
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /root/.node-gyp/npm-debug.log

```

todbot, 2017-02-17:
Hi, 

1. What is OSMC?

2. Also, for large text output, please create a gist and link to that.  Otherwise it makes reading issues difficult.

3. The error is saying npm doesn't have root permission. So as a quick fix, try running the same command with 'sudo'.  But it is normally bad practice to run node & npm as 'root'.  It looks like however you installed node is assuming it to be in a root-controlled area.

closed 2019-12-05 by todbot