node-hid/node-hid #184: Can not send Feature Report to Dualshock 4? (Bluetooth)

shubshub, 2017-02-18:
So I've been looking at a few different places on sending data to the Dualshock 4 over Bluetooth, I've done the CRC-32 Hash I've got the report ID and everything according to the program there are no errors, However nothing was happening when I did device.write(buff) atleast not from what I could see

I don't know if I was sending the data wrong if I was missing just the tiniest thing, but reading more into it according to what this is saying here
http://www.spinics.net/lists/linux-input/msg29517.html
`>  #ifdef CONFIG_SONY_FF
>         buf[offset++] = sc->right;
> @@ -1285,7 +1292,11 @@ static void dualshock4_state_worker(struct work_struct *work)
>         buf[offset++] = sc->led_state[1];
>         buf[offset++] = sc->led_state[2];
>
> -       hdev->ll_driver->output_report(hdev, buf, sizeof(buf));
> +       if (sc->quirks & DUALSHOCK4_CONTROLLER_USB)
> +               hdev->ll_driver->output_report(hdev, buf, 32);
> +       else
> +               hdev->ll_driver->raw_request(hdev, 0x11, buf, 78,
> +                               HID_OUTPUT_REPORT, HID_REQ_SET_REPORT);

Ok, now I get it. The USB stuff takes output-reports and BT uses
SET_REPORT. That's exactly the inverse of wiimotes.`

The last bit (The comments at the bottom) state that the USB stuff is output reports (Which I assume is the basic device.write since that worked perfectly over USB) and I'm guessing by SET_REPORT is a send Feature Report? as said here https://github.com/torvalds/linux/blob/7ae123edd37a47e178eb9a6631fe4a7108262c10/include/linux/hid.h#L738

Here is an image of the data I am trying to send to the Feature Report and also it failing
[image]

- For sendFeatureReport native code see Line 343 in HID.cc
- For hid_send_feature_report native code see line 749 in hidapi\windows\hid.c

I've also been reading over http://www.psdevwiki.com/ps4/DS4-BT However it has only been minimal help to me and hasn't actually helped me figure out what I am doing incorrectly.

If anyone or even you have any suggestions that would be excellent :)

todbot, 2017-02-19:
This is not a bug in node-hid, as far as I can tell, but rather figuring out the protocol a DualShock 4 controller needs when in Bluetooth mode.  I just picked up a DS4 controller myself to look into this and can verify the issue you're seeing.  I think instead you should update the issue already open on rdepena/node-dualshock-controller#64

btw, a DS3 controller over Bluetooth does allow setting of LEDs & rumble, if that is a workaround for you.

shubshub, 2017-02-19:
I've updated the issue over there just now, Also I'm using a Dualshock 4 because of the colorable LED which I'm using to make a neat little game out of, According to some things I've read I have to send a Feature Report but it won't let me do that Because I don't know the right protocol I guess

todbot, 2017-02-19:
Hi,
I made some headway.

This repo: https://github.com/todbot/node-hid-ds4-test
is an example of being able to set the rumble and LED on a DS4 over Bluetooth.

However, upon sending the output report to update the rumble & LED, the controller stops streaming data on reportId 1.  It's unclear to me why this happens or how to restart it. Other drivers seem to switch to getting input data from reportId 11, but I've not seen exactly how to enable that either.

This is not a bug in node-hid so I'm closing it for now.  I'll be watching the issue on rdepena/node-dualshock-controller#64 so if you or others can get further, I'll be around to maybe offer some assistance.

closed 2017-02-19 by todbot

connorc0405, 2017-10-29:
Has anyone figured out how to get the DS4 to continue sending input report 0x01 after sending an output report?  Do you need to send a get feature report to make the DS4 start sending input reports again? I am using hidapi on mac.

Pecacheu, 2018-01-11:
Think I may have found the issue (though I can't solve it). I believe it is indeed a problem with node-hid, see here: https://github.com/rdepena/node-dualshock-controller/issues/64#issuecomment-357082104

(Putting this here for Goggle search result purposes)