node-hid/node-hid #186: cross compile not working due to hard coded include path in binding.gyp

FreshXOpenSource, 2017-02-23:
Cross compiling node-hid for instance in buildroot for raspberry leads to an error :

```arm-buildroot-linux-gnueabihf-gcc: ERROR: unsafe header/library path used in cross-compilation: '/usr/include/libusb-1.0'```

gcc is picky about the fixed include path in bindings.gyp

todbot, 2017-02-23:
Hi,
Please give more details on exactly what you're doing to result in this error.  Why are you cross-compiling instead of (like most people) running this directly on the RasPi?

FreshXOpenSource, 2017-02-23:
Hi Tod,

we use buildroot from https://buildroot.org to build an embedded linux firmware running most of the logic on nodejs. as part of this we need node-hid, which we compile by adding it to the additional npm modules in the buildroot interpreter section. 

the make of the buildroot fails with above's message

Lange, 2019-06-11:
I believe I'm encountering the same issue, though I don't have a lot of domain knowledge here so it's possible I am chasing a red herring.

I too am attempting to cross-compile a buildroot image, for the purposes of using WPE Webkit on my Raspberry Pi 3. I am following this guide: https://medium.com/@decrocksam/building-web-applications-for-wpe-webkit-using-node-js-3347146013f3

todbot, 2019-06-11:
hmm yeah this is a problem but I'm not sure how to solve it. `node-pre-gyp` needs to know where libusb libs and headers are to build `node-hid` and I'm not sure how to propagate programmatically those settings to the `binding.gyp` file.  I'll look into gyp variables and see what's possible.

In your cross-compile setup, do you have access to `pkg-config` and does it work correctly for doing things like `pkg-config libusb-1.0 --cflags`?

`node-hid` does compile just fine on Raspberry Pi but that's on the device, not in cross-compile.

Lange, 2019-06-11:
I'm unfamiliar with `pkg-config` or how to use it.

I _think_ the solution might just be changing this:

```
'include_dirs+': ['/usr/include/libusb-1.0/']
```
to this (not sure I have the gyp syntax correct?):
```
'include_dirs+': ['<!(echo %STAGING_DIR%)/usr/include/libusb-1.0/']
```

but, I don't know that for certain, this is just a guess.

Lange, 2019-06-11:
The above change does indeed seem to let the build of `node-hid` succeed under buildroot. I'm going to finish building my buildroot image and then try to actually test this out on my RPI3 and see if it works as expected.

todbot, 2019-06-11:
Basically `pkg-config` is a platform- independent way to specify info about libraries. In your cross-compile setup, I would expect there should be a way to set your PATH so that running the command:
```
pkg-config --cflags libusb-1.0
```
would output the same thing as 
```
echo $STAGING_DIR)/usr/include/libusb-1.0/
```
If so, then the change I just checked into `binding.gyp` should work for you.

Lange, 2019-06-13:
I think it works, yeah. Thank you!

My experiments with buildroot have unfortunately hit an unrelated roadblock, so I can't actually fully test my app. However, `node-hid` does seem to compile at least.

closed 2019-07-25 by todbot