node-hid/node-hid #187: Write fails when first byte is 0x00

Gastonius, 2017-02-27:
Hello everybody,

So, my system: electron, mac, device is Mooltipass (HID for password storage),

Anyway, I'm able to connect and send and retrieve commands just fine. Problem is: Whenever I need to send a command that starts with 0x00 it doesn't send anything, and Mooltipass protocol requires a 0x00 sometimes.

So, 
device.write([0x01,0xA2]) -> Works fine. device.read() gets data from the device, etc.
device.write(**[0x00,0xA2]**) -> Doesn't work, most probably isn't even sending any data.
device.write([0xB,0xA4,etc,etc,etc] -> Works fine.
device.write([0x00,0xA4,etc,etc,etc] -> Doesn't work.

Also, by using a Chrome App, and chrome.hid, same commands work fine and as expected.

Have a nice day, and thanks!

limpkin, 2017-02-27:
following that issue.

Gastonius, 2017-02-27:
Ok, found the issue,

hid_write in hid.c actually skips the first byte if it is a 0, so, my workaround was:

When I need to send [0x00,0x01], I actually send [0x00,0x00,0x01]. Not elegant, but works.

Hope this helps someone.

todbot, 2017-02-27:
Hi @Gastonius,
You're talking about hid_write in hidapi, correct?  We should probably open an issue there.  I believe this is related to the fact that hidapi inherited the libusb technique of putting the ReportId in the first byte of the buffer and that not all HID devices use numbered reports (making the first buffer byte zero)

Actually, I'm wondering if maybe it's an issue with Mooltipass?  Do you have a link to its HID descriptor?  Is it using numbered reports?

todbot, 2017-02-27:
Nah, forget that last paragraph. After looking at it more, I'm thinking it's more an issue with the Mac implementation of `hid_write()` in hidapi

Gastonius, 2017-02-27:
Hi Tod,

Hid.c in libusb I believe it is

No, it is not using numbered reports, and limpkin should have the correct link.

limpkin, 2017-02-28:
Hello,

https://github.com/limpkin/mooltipass/blob/master/source_code/src/USB/usb_descriptors.c#L48 is the link. 

Following the USB spec, report IDs are completely optional when implementing a custom HID endpoint.

todbot, 2017-03-10:
I believe this is an issue within https://github.com/signal11/hidapi  (the C library that node-hid wraps).

However, in looking into how to purchases a Mooltipass (which I couldn't find), I did notice that Mooltipass has their own app that uses node-hid and it apparently doesn't suffer from the issue noted here:
  https://github.com/limpkin/mooltipass/tree/master/mooltiapp

limpkin, 2017-03-10:
Hello,

Yes, we've just pushed mooltipass app with the quick fix we mentioned: https://github.com/limpkin/mooltipass/commit/ab118b6644d0285f515733f7cbee02d96bda06d8

but this is a quick fix...

Gastonius, 2017-03-10:
Hi Tod,  actually it does!

https://github.com/limpkin/mooltipass/blob/master/mooltiapp/app/preload.js#L218 See there I had to add a 0 to every package I sent that way.

Now... from my findings:

In MAC, you have to send a 0 if the first byte is 0: So, [0,100] reaches the device as [100], so I had to send [0,0,100] in order to have [0,100] reaching the device. Now, if you send [1,100], the device will get [1,100]

In Windows, you have to send a 0 in the first byte always.

I guess Limpkin is more acquainted on the lower-level communications & protocol, that's what I experienced.

In libusb you can clearly see here: https://github.com/signal11/hidapi/blob/a6a622ffb680c55da0de787ff93b80280498330f/libusb/hid.c#L1006 

So, it is actually part of hidapi, a "bug" or a "feature", should node-hid add that 0, knowing it will be stripped by the driver?

Have a great weekend!
Gaston

todbot, 2017-03-10:
I think the proper reaction should be that hidapi acts the same across platforms, as it's not apparently for this case.

I'm really reluctant to add this sort of platform-specific hack to node-hid because I can really see it biting us down the road.

I'll see about creating a test case in vanilla hidapi C code to submit as an issue in the hidapi repo. Is there a place I can buy a Mooltipass to test on?

limpkin, 2017-03-10:
Actually we'll send you one for free! just drop me a message at limpkin at limpkin dot fr with your address.

todbot, 2017-03-10:
Wow, thanks @limpkin! I've been following Mooltipass off-and-on since I saw an early version at a Hackaday meetup in Los Angeles. It's come so far since then. Emailing you off-list...

gniezen, 2017-03-22:
@todbot I'm not sure if ef1388d is correct. It doesn't matter if the first byte is 0x00 or not, it's that hidapi's `hid_write` on Windows expects a report ID as the first byte (usually 0x00 when there's only a single report) while Mac does not.

According to an [hidapi issue](https://github.com/signal11/hidapi/issues/255) that's still open, @signal11 said

> It should be the same. Are you using the latest from git?

However, it looks like node-hid is indeed using the latest git snapshot, so it looks like this bug is still there. 

I agree that a platform-specific hack is not ideal, but the only alternative at the moment is that anyone using node-hid (or hidapi directly) will have to write a platform-specific hack in their own code.

limpkin, 2017-03-22:
This is what we have done.
However, report IDs do not make sense when we are talking to a USB HID endpoint.

gniezen, 2017-03-22:
If you're not using the report ID (when you just have a single report) just set it to 0x00, as per the docs linked above. It will be passed on to Windows (which requires it), and hidapi will strip it out before sending it to libusb (which does not require it).

From my copy of _USB Complete Fifth Edition_:
> A HID can support multiple reports of the same type with each report having its own Report ID and contents. [...] If there is no Report ID item, the report used the default ID of 0x00. A descriptor should not declare a Report ID of 0x00.

I would therefore argue that the bug is in Windows, not hidapi or node-hid ;)

limpkin, 2017-03-22:
If I'm not mistaken, report ID only apply to non vendor define HID usage pages.

gniezen, 2017-03-22:
Sure, that may be the case in the USB spec, but it seems Microsoft made some strange decisions in their implementation. From _USB Complete_ again:

> For Windows applications, the report buffer provided to an API function must be large enough to hold the report plus one byte for the Report ID even if using only Report ID 0x00.

From what I gather the decision made in hidapi, in order to not require platform-specific code, is to always require a report ID and strip it away when it's 0x00 for implementations that actually adhere to the USB spec (like macOS and libusb).

FutureOfAI, 2019-06-05:
> Hi Tod, actually it does!
>
> https://github.com/limpkin/mooltipass/blob/master/mooltiapp/app/preload.js#L218 See there I had to add a 0 to every package I sent that way.
>
> Now... from my findings:
>
> In [...]

Yes, I have the same problem as you, It seems hardware automatic ignore first byte. Some app will add 0x00 in the first byte in windows but some app will be not, I test two different windows app, and get their output data:
[image]
The difference between them is if the first byte is 0x00

And, I use STM32F2xx HAL_USB_HID_Custom to receive, my describe function is in the blow, are there some errors in my configure?
##########################################################################
__ALIGN_BEGIN static uint8_t CUSTOM_HID_ReportDesc_FS[USBD_CUSTOM_HID_REPORT_DESC_SIZE] __ALIGN_END =
{
  /* USER CODE BEGIN 1 */ 
  0x06, 0xFF, 0x00,      // Usage Page = 0xFF00 (Vendor Defined Page 1)
  0x09, 0x01,             // Usage (Vendor Usage 1)
  0xA1, 0x01,             // Collection (Application)
  // Input report
  0x19, 0x01,             // Usage Minimum
  0x29, 0x40,             // Usage Maximum
  0x15, 0x00,             // Logical Minimum (data bytes in the report may have minimum value = 0x00)
  0x26, 0xFF, 0x00,       // Logical Maximum (data bytes in the report may have maximum value = 0x00FF = unsigned 255)
  0x75, 0x08,             // Report Size: 8-bit field size
  0x95, CUSTOM_HID_EPIN_SIZE,// Report Count
  0x81, 0x02,             // Input (Data, Array, Abs)
  // Output report
  0x19, 0x01,             // Usage Minimum
  0x29, 0x40,             // Usage Maximum
  0x75, 0x08,             // Report Size: 8-bit field size
  0x95, CUSTOM_HID_EPOUT_SIZE,// Report Count
  0x91, 0x02,             // Output (Data, Array, Abs)
  0xC0    /*     END_COLLECTION	             */
   
}; 
######################################################################
/* USB CUSTOM_HID device Configuration Descriptor */
__ALIGN_BEGIN static uint8_t USBD_CUSTOM_HID_CfgDesc[USB_CUSTOM_HID_CONFIG_DESC_SIZ] __ALIGN_END =
{
  0x09, /* bLength: Configuration Descriptor size */
  USB_DESC_TYPE_CONFIGURATION, /* bDescriptorType: Configuration */
  USB_CUSTOM_HID_CONFIG_DESC_SIZ,
  /* wTotalLength: Bytes returned */
  0x00,
  0x01,         /*bNumInterfaces: 1 interface*/
  0x01,         /*bConfigurationValue: Configuration value*/
  0x00,         /*iConfiguration: Index of string descriptor describing
  the configuration*/
  0xC0,         /*bmAttributes: bus powered */
  0x32,         /*MaxPower 100 mA: this current is used for detecting Vbus*/
  
  /************** Descriptor of CUSTOM HID interface ****************/
  /* 09 */
  0x09,         /*bLength: Interface Descriptor size*/
  USB_DESC_TYPE_INTERFACE,/*bDescriptorType: Interface descriptor type*/
  0x00,         /*bInterfaceNumber: Number of Interface*/
  0x00,         /*bAlternateSetting: Alternate setting*/
  0x02,         /*bNumEndpoints*/
  0x03,         /*bInterfaceClass: CUSTOM_HID*/
  0x00,         /*bInterfaceSubClass : 1=BOOT, 0=no boot*/
  0x00,         /*nInterfaceProtocol : 0=none, 1=keyboard, 2=mouse*/
  0,            /*iInterface: Index of string descriptor*/
  /******************** Descriptor of CUSTOM_HID *************************/
  /* 18 */
  0x09,         /*bLength: CUSTOM_HID Descriptor size*/
  CUSTOM_HID_DESCRIPTOR_TYPE, /*bDescriptorType: CUSTOM_HID*/
  0x11,         /*bCUSTOM_HIDUSTOM_HID: CUSTOM_HID Class Spec release number*/
  0x01,
  0x00,         /*bCountryCode: Hardware target country*/
  0x01,         /*bNumDescriptors: Number of CUSTOM_HID class descriptors to follow*/
  0x22,         /*bDescriptorType*/
  USBD_CUSTOM_HID_REPORT_DESC_SIZE,/*wItemLength: Total length of Report descriptor*/
  0x00,
  /******************** Descriptor of Custom HID endpoints ********************/
  /* 27 */
  0x07,          /*bLength: Endpoint Descriptor size*/
  USB_DESC_TYPE_ENDPOINT, /*bDescriptorType:*/
  
  CUSTOM_HID_EPIN_ADDR,     /*bEndpointAddress: Endpoint Address (IN)*/
  0x03,          /*bmAttributes: Interrupt endpoint*/
  CUSTOM_HID_EPIN_SIZE, /*wMaxPacketSize: 2 Byte max */
  0x00,
  0x20,          /*bInterval: Polling Interval (20 ms)*/
  /* 34 */
  
  0x07,	         /* bLength: Endpoint Descriptor size */
  USB_DESC_TYPE_ENDPOINT,	/* bDescriptorType: */
  CUSTOM_HID_EPOUT_ADDR,  /*bEndpointAddress: Endpoint Address (OUT)*/
  0x03,	/* bmAttributes: Interrupt endpoint */
  CUSTOM_HID_EPOUT_SIZE,	/* wMaxPacketSize: 2 Bytes max  */
  0x00,
  0x20,	/* bInterval: Polling Interval (20 ms) */
  /* 41 */
} ;

/* USB CUSTOM_HID device Configuration Descriptor */
__ALIGN_BEGIN static uint8_t USBD_CUSTOM_HID_Desc[USB_CUSTOM_HID_DESC_SIZ] __ALIGN_END =
{
  /* 18 */
  0x09,         /*bLength: CUSTOM_HID Descriptor size*/
  CUSTOM_HID_DESCRIPTOR_TYPE, /*bDescriptorType: CUSTOM_HID*/
  0x11,         /*bCUSTOM_HIDUSTOM_HID: CUSTOM_HID Class Spec release number*/
  0x01,
  0x00,         /*bCountryCode: Hardware target country*/
  0x01,         /*bNumDescriptors: Number of CUSTOM_HID class descriptors to follow*/
  0x22,         /*bDescriptorType*/
  USBD_CUSTOM_HID_REPORT_DESC_SIZE,/*wItemLength: Total length of Report descriptor*/
  0x00,
};

todbot, 2019-11-23:
I believe this is fixed by move to libusb/hidapi as of v1.0.0.

closed 2019-11-23 by todbot