node-hid/node-hid #188: Raspberry Pi Compile Problems

nathandtrone, 2017-03-05:
So I've been trying to install this on my Raspberry Pi Zero;
uname -a "Linux raspberrypi 4.4.50+ #970 Mon Feb 20 19:12:50 GMT 2017 armv6l GNU/Linux"

I'm fresh out of ideas, and at the end of my rope here. Does anyone or the dev know if I'm stupidly missing something here? Everything else I run on my Pi works perfectly fine, except for this.

I've been trying to get this working for about a week now with no success. I've tried numerous solutions that I know of and have followed another numerous fixes suggested by some other communities in the past. This final time I've fresh installed and tried this:

**Install.sh**
echo "Check your system is up to date:"
sudo apt-get update
sudo apt-get upgrade

echo "Install latest version of node"
sudo wget http://node-arm.herokuapp.com/node_latest_armhf.deb
sudo dpkg -i node_latest_armhf.deb

echo "Check latest version of node is installed"
node -v

echo "Install dependancies for node-hid"
sudo apt-get install g++ curl
sudo apt-get install libssl-dev apache2-utils
sudo apt-get install build-essential
sudo apt-get install pkg-config

sudo npm install node-gyp -g

sudo apt-get install git-core
sudo apt-get install libudev-dev
sudo apt-get install libusb-1.0-0-dev

echo "Install node-hid"
sudo npm install node-hid

**Here's the output of the NPM INSTALL:**

pi@raspberrypi:~/projects $ sudo npm install node-hid
|
\> node-hid@0.5.4 install /home/pi/projects/node_modules/node-hid
\> node-pre-gyp install --fallback-to-build

node-pre-gyp ERR! Tried to download(404): https://github.com/node-hid/node-hid/releases/download/0.5.4/HID-v0.5.4-node-v46-linux-arm.tar.gz
node-pre-gyp ERR! Pre-built binaries not found for node-hid@0.5.4 and node@4.2.1 (node-v46 ABI) (falling back to source compile with node-gyp)
gyp WARN EACCES user "root" does not have permission to access the dev dir "/root/.node-gyp/4.2.1"
gyp WARN EACCES attempting to reinstall using temporary dev dir "/home/pi/projects/node_modules/node-hid/.node-gyp"
make: Entering directory '/home/pi/projects/node_modules/node-hid/build'
make: *** No rule to make target '../.node-gyp/4.2.1/include/node/common.gypi', needed by 'Makefile'.  Stop.
make: Leaving directory '/home/pi/projects/node_modules/node-hid/build'
gyp ERR! build error
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/home/pi/projects/node_modules/node-gyp/lib/build.js:276:23)
gyp ERR! stack     at emitTwo (events.js:87:13)
gyp ERR! stack     at ChildProcess.emit (events.js:172:7)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:200:12)
gyp ERR! System Linux 4.4.50+
gyp ERR! command "/usr/local/bin/node" "/home/pi/projects/node_modules/node-gyp/bin/node-gyp.js" "build" "--fallback-to-build" "--module=/home/pi/projects/node_modules/node-hid/build/Release/HID.node" "--module_name=HID" "--module_path=/home/pi/projects/node_modules/node-hid/build/Release"
gyp ERR! cwd /home/pi/projects/node_modules/node-hid
gyp ERR! node -v v4.2.1
gyp ERR! node-gyp -v v3.5.0
gyp ERR! not ok
node-pre-gyp ERR! build error
node-pre-gyp ERR! stack Error: Failed to execute '/usr/local/bin/node /home/pi/projects/node_modules/node-gyp/bin/node-gyp.js build --fallback-to-build --module=/home/pi/projects/node_modules/node-hid/build/Release/HID.node --module_name=HID --module_path=/home/pi/projects/node_modules/node-hid/build/Release' (1)
node-pre-gyp ERR! stack     at ChildProcess.<anonymous> (/home/pi/projects/node_modules/node-hid/node_modules/node-pre-gyp/lib/util/compile.js:83:29)
node-pre-gyp ERR! stack     at emitTwo (events.js:87:13)
node-pre-gyp ERR! stack     at ChildProcess.emit (events.js:172:7)
node-pre-gyp ERR! stack     at maybeClose (internal/child_process.js:818:16)
node-pre-gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:211:5)
node-pre-gyp ERR! System Linux 4.4.50+
node-pre-gyp ERR! command "/usr/local/bin/node" "/home/pi/projects/node_modules/node-hid/node_modules/.bin/node-pre-gyp" "install" "--fallback-to-build"
node-pre-gyp ERR! cwd /home/pi/projects/node_modules/node-hid
node-pre-gyp ERR! node -v v4.2.1
node-pre-gyp ERR! node-pre-gyp -v v0.6.31
node-pre-gyp ERR! not ok
Failed to execute '/usr/local/bin/node /home/pi/projects/node_modules/node-gyp/bin/node-gyp.js build --fallback-to-build --module=/home/pi/projects/node_modules/node-hid/build/Release/HID.node --module_name=HID --module_path=/home/pi/projects/node_modules/node-hid/build/Release' (1)
npm ERR! Linux 4.4.50+
npm ERR! argv "/usr/local/bin/node" "/usr/local/bin/npm" "install" "node-hid"
npm ERR! node v4.2.1
npm ERR! npm  v2.14.7
npm ERR! code ELIFECYCLE

npm ERR! node-hid@0.5.4 install: `node-pre-gyp install --fallback-to-build`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the node-hid@0.5.4 install script 'node-pre-gyp install --fallback-to-build'.
npm ERR! This is most likely a problem with the node-hid package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-pre-gyp install --fallback-to-build
npm ERR! You can get their info via:
npm ERR!     npm owner ls node-hid
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /home/pi/projects/npm-debug.log

**npm-debug.log**
[npm-debug.log.txt](https://github.com/node-hid/node-hid/files/819075/npm-debug.log.txt)

If you need any other information from me, just ask. Thanks in advance.

nathandtrone, 2017-03-05:
I am not sure what changed but after I had put my project aside for a while, nothing changed at all, I hadn't touched it, but now it's working.. I'm not sure why but it works now, but I have a new issue. It only shows one device. Even though my LSUSB returns:
Bus 001 Device 008: ID 2354:1111
Bus 001 Device 007: ID 0a12:0001 Cambridge Silicon Radio, Ltd Bluetooth Dongle (HCI mode)
Bus 001 Device 006: ID 0bda:817b Realtek Semiconductor Corp.
Bus 001 Device 004: ID 05e3:0608 Genesys Logic, Inc. USB-2.0 4-Port HUB
Bus 001 Device 003: ID 148f:5370 Ralink Technology, Corp. RT5370 Wireless Adapter
Bus 001 Device 002: ID 058f:9254 Alcor Micro Corp. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

So what in the world is wrong with my Pi....

nathandtrone, 2017-03-16:
I still haven't found a solution to this problem. I've checked everything I can think of, and tried to write my own solution and I can't get it to work properly, nor have the time.

Does anyone else have a Raspberry Pi Zero and have this functioning properly? Any information? No replies either which is starting to worry me.

todbot, 2017-03-16:
Hi!
`node-hid` only sees HID devices, it doesn't see any non-HID devices.  From that lsusb, it does'nt look like there are any HID devices except for maybe '2345:1111'

nathandtrone, 2017-03-16:
Oh, ok. Thanks for the reply. What about the path in my Node result, matches the Device 8 on my lsusb, and it has different VendorID and Product ID? Any idea why that might be happening? It's a robotics usb device. I can use python HID to connect to it, but it doesn't seem to work on this.

todbot, 2017-03-16:
It's the same ids, but represented in either decimal or hexadecimal.  lsusb shows the vendorId and productId in hex, e.g. "2354:1111" is 0x2354 / 0x1111. And 0x2354 == 9044, 0x1111 == 4369.

Node.js doesn't have an easy way printing out numbers in hex format, so it does it in decimal.  You can use either decimal or hex format when opening a device with node-hid, but it'll always print them in decimal.

nathandtrone, 2017-03-20:
Ok. So I've found something. If you try to sudo a node js script on pi zero, it doesn't actually sudo it. I force logged in as root and then tried running the script, everything is functioning properly now.

This ticket can be closed.

closed 2019-07-25 by todbot