node-hid/node-hid #189: Problem installing "minidsp". Log indicates error in"node-hid"

ian-robertson, 2017-03-10:
Dear author,
System is Linux 3.13.0-112-generic. Ubuntu 14.04 LTS.
Installed nodejs v4.8.0 and npm v2.15.11  using the source obtained from 
"curl -sL https://deb.nodesource.com/setup_4.x"
Attempting to to install module "minidsp" using "npm install -g minidsp".  
Failure message indicates a possible problem in "node-hid"

"Failed at the node-hid@0.5.4 install script 'node-pre-gyp install --fallback-to-build'.
This is most likely a problem with the node-hid package,
Tell the author that this fails on your system:
node-pre-gyp install --fallback-to-build"

Echo of screen output and /usr/lib/npm-debug.log  attached.
Good luck an many thanks,
Ian

[npm_minidsp_install_screen_output.txt](https://github.com/node-hid/node-hid/files/833781/npm_minidsp_install_screen_output.txt)

[npm-debug.log.txt](https://github.com/node-hid/node-hid/files/833782/npm-debug.log.txt)

todbot, 2017-03-10:
Hi,
Thank you for your complete error log, it is very helpful.  From it, it looks like the problem is with "ascii-progress", which is I believe really a dependency of `node-minidsp`, not `node-hid`. Specifically, this section:

```
make: Entering directory `/usr/lib/node_modules/minidsp/node_modules/ascii-progress/node_modules/get-cursor-position/build'
  CXX(target) Release/obj.target/pos/src/pos.o
../src/pos.cc: In function ‘void Method(const v8::FunctionCallbackInfo<v8::Value>&)’:
../src/pos.cc:259:6: warning: variable ‘ret’ set but not used [-Wunused-but-set-variable]
  int ret, row, col;
      ^
  SOLINK_MODULE(target) Release/obj.target/pos.node
  COPY Release/pos.node
make: Leaving directory `/usr/lib/node_modules/minidsp/node_modules/ascii-progress/node_modules/get-cursor-position/build'
```

Also, as an aside:
* The only version of Node v0.4.x we test on is the last LTS version v0.4.12.  That may be an issue with `node-minidsp` or its dependencies as well.
* I'm not sure but I believe it's bad practice to install node as root and run npm as root.  I've noticed several people having issues when they use 'sudo' with npm.

closed 2017-03-10 by todbot

ian-robertson, 2017-03-11:
Hi Tod,

Thanks for rapid response. I will take your advice and repeat with a 
later version on Ubuntu 16.04.  However, I am not yet convinced that the
problem lies in minidsp since I have generated a similar error when I
try to install node-hid on its own.  I'll also repeat without sudo (and
fix any necessary permissions). 

Best wishes

Ian

------------------------------------------------------------------------
Ian Robertson PhD, CEng, FIET
Telephone  : 07734 325046
Email :  ian.robertson@theiet.org
------------------------------------------------------------------------

ian-robertson, 2017-03-11:
I confirm that, following, Tod's advice, I have now correctly installed minidsp.js  using:

curl -sL https://deb.nodesource.com/setup_7.x | sudo -E bash
apt-get install -y nodejs
npm install -g minidsp

The versions that are now running are:
nodejs - v   v7.7.2
npm -v  4.1.2
minidsp -V  1.0.3

Ian.

ian-robertson, 2017-03-11:
Hi Tod,

I confirm that, following, your advice, I have now correctly installed
minidsp.js using:

curl -sL https://deb.nodesource.com/setup_7.x | sudo -E bash
apt-get install -y nodejs
npm install -g minidsp

The versions that are now running are:
nodejs - v v7.7.2
npm -v 4.1.2
minidsp -V 1.0.3

Thanks again

Ian.

------------------------------------------------------------------------
Ian Robertson PhD, CEng, FIET
Telephone  : 07734 325046
Email :  ian.robertson@theiet.org
------------------------------------------------------------------------