node-hid/node-hid #190: How can I get all the device's report descriptors

rpastro, 2017-03-21:
I have just started evaluating the node-hid module to see if I can use it for my application and I have a very basic question.  How can I get the list of all the report descriptors supported by a device, with the corresponding usagePage, usage and reportIds?

In my application I am only interested in usagePage==11 and usage==5, so I need to find out if the device supports that and if it does, I need to get the corresponding reportId(s).

For instance, here is the information that I receive for a device if I use the chrome.hid API in a Chrome packaged app.

{
    "collections": [{
        "reportIds": [ 1 ],
        "usage": 1,
        "usagePage": 12
    }, {
        "reportIds": [ 2,  4, 5 ],
        "usage": 1,
        "usagePage": 65280
    }, {
        "reportIds": [ 3 ],
        "usage": 5,
        "usagePage": 11
    }],
    "deviceId": 20,
    "maxFeatureReportSize": 1,
    "maxInputReportSize": 32,
    "maxOutputReportSize": 32,
    "productId": 1056,
    "productName": "Jabra SPEAK 510 USB",
    "reportDescriptor": {},
    "serialNumber": "1C48F9E833D5020A00",
    "vendorId": 2830
}

As you can see, the device information contains all the report descriptors under the collections array and I can find out that the reportId is 3.

On the other hand, below is the information that I receive when invoking the HID.devices() API.

 { vendorId: 2830,
    productId: 1042,
    path: '...',
    serialNumber: '1C48F97DDD08x010900',
    manufacturer: '',
    product: 'Jabra SPEAK 410 USB',
    release: 265,
    interface: -1,
    usagePage: 12,
    usage: 1 },

The API only returns the first usagePage/usage and it doesn't return the reportId at all.

Is there any way to get the information I need with the node-hid API?

Thank you for your support.

todbot, 2017-03-21:
Hi,
Which OS are you on?  node-hid will return usage & usagePage info for Mac & Windows, but not Linux. node-hid depends on the hidapi C library which doesn't support this yet, see: https://github.com/node-hid/node-hid#usage-and-usagepage-device-info-fields

rpastro, 2017-03-21:
I need to support Windows and Mac. As you can see below I am getting a usagePage (12) and usage (5).  

The problem is that the response doesn't include the other usagePages and usages which are also supported by the device (see response from chrome.hid API in the original message). The response also doesn't include any of the reportIds.

{ vendorId: 2830,
productId: 1042,
path: '...',
serialNumber: '1C48F97DDD08x010900',
manufacturer: '',
product: 'Jabra SPEAK 410 USB',
release: 265,
interface: -1,
usagePage: 12,     <<< This is what I get with node-hid
usage: 1 }

todbot, 2017-03-21:
what does 'show-devices.js' output?  There should be a separate object for each usage page (i.e. each usage appears as a separate device).  For instance, here's what my Macbook Pro trackpad looks like, which has multiple usages:
```
devices: [ { vendorId: 1452,
    productId: 610,
    path: 'IOService:/IOResources/AppleUSBHostResources/AppleUSBLegacyRoot/AppleUSBXHCI@14000000/Apple Internal Keyboard / Trackpad@14400000/Touchpad@2/AppleUSBTCButtons',
    serialNumber: '',
    manufacturer: 'Apple Inc.',
    product: 'Apple Internal Keyboard / Trackpad',
    release: 549,
    interface: -1,
    usagePage: 1,
    usage: 2 },
  { vendorId: 1452,
    productId: 610,
    path: 'IOService:/IOResources/AppleUSBHostResources/AppleUSBLegacyRoot/AppleUSBXHCI@14000000/Apple Internal Keyboard / Trackpad@14400000/Touchpad@1/AppleUSBMultitouchDriver/IOHIDInterface/AppleUSBMultitouchHIDEventDriver/IOHIDPointing/IOHIDPointingDevice',
    serialNumber: '',
    manufacturer: 'Apple Inc.',
    product: 'Apple Internal Keyboard / Trackpad',
    release: 549,
    interface: -1,
    usagePage: 1,
    usage: 2 },
  { vendorId: 1452,
    productId: 610,
    path: 'IOService:/IOResources/AppleUSBHostResources/AppleUSBLegacyRoot/AppleUSBXHCI@14000000/Apple Internal Keyboard / Trackpad@14400000/Touchpad@1/AppleUSBMultitouchDriver',
    serialNumber: '',
    manufacturer: 'Apple Inc.',
    product: 'Apple Internal Keyboard / Trackpad',
    release: 549,
    interface: -1,
    usagePage: 65280,
    usage: 1 },
  { vendorId: 1452,
    productId: 610,
    path: 'IOService:/IOResources/AppleUSBHostResources/AppleUSBLegacyRoot/AppleUSBXHCI@14000000/Apple Internal Keyboard / Trackpad@14400000/Apple Internal Keyboard@0/AppleUSBTCKeyboard',
    serialNumber: '',
    manufacturer: 'Apple Inc.',
    product: 'Apple Internal Keyboard / Trackpad',
    release: 549,
    interface: -1,
    usagePage: 1,
    usage: 6 } ]
```

todbot, 2017-03-21:
Also, I don't think node-hid can get reportIds any time soon, as it's just a wrapper for hidapi and I don't think hidapi supports getting that info.

rpastro, 2017-03-22:
I'm not getting separate objects for all the usages. I only get a single device with the first usage.

For instance, if I connect a Jabra EVOLVE LINK headset, I only get the following object back:

`
{ interface: -1,
  manufacturer: "GN Netcom A/S",
  path: "...",
  product: "Jabra EVOLVE LINK",
  productId: 774,
  release: 297,
  serialNumber: "000080FE56FC09",
  usage: 5,
  usagePage: 11,
 vendorId: 2830 }
`

This device, however, has three different usages. Below is the output for the same device using the chrome.hid API (https://developer.chrome.com/apps/hid)

`
   {
      "collections": [
         {
            "reportIds": [ 2 ],
            "usage": 5,
            "usagePage": 11
         },
         {
            "reportIds": [ 4, 5 ],
            "usage": 5,
            "usagePage": 65280
         },
         {
            "reportIds": [ 1 ],
            "usage": 1,
            "usagePage": 12
         }
      ],
      "deviceId": 76,
      "maxFeatureReportSize": 0,
      "maxInputReportSize": 32,
      "maxOutputReportSize": 32,
      "productId": 774,
      "productName": "Jabra EVOLVE LINK",
      "reportDescriptor": {},
      "serialNumber": "000080FE56FC09",
      "vendorId": 2830
   }
]
`

todbot, 2017-03-22:
And this is on Mac OS X?

rpastro, 2017-03-22:
Yes. Mac OS X version 10.11.6.

rpastro, 2017-03-24:
FYI, on Windows it works as you described, i.e. there is a different device for each usage.

closed 2019-11-23 by todbot