node-hid/node-hid #192: cannot use node-hid library in electron app

vnsharma-systango, 2017-03-22:
Hi @todbot ,

I had done all the needful mentioned under Using node-hid in Electron projects, package did get installed
but when i require it in my js file using :-

var HID = require('node-hid');
var devices = HID.devices();

And run my app , it gives an error 

> Error: Module version mismatch. Expected 50, got 51.

So can you please help me with this .

Thanks

todbot, 2017-03-22:
Hi,
Please give the exact steps you used to install node-hid in an Electron project (a link to a github repo with a package.json could be sufficient)

vnsharma-systango, 2017-03-23:
Hi,

I followed following steps -

In your electron project, add electron-rebuild and electron-prebuilt to
your devDependencies. Then in your package.json scripts add:

  "postinstall": "electron-rebuild --force"

If you want a specific version of electron, do something like:

electron-rebuild -v 0.36.5 --force -m . -w node-hid

and my package.json look's like this -

> {
>   "name": "Clientele",
>   "version": "1.0.0",
>   "description": "Clientele",
>   "homepage": "https://app.onepgr.com/",
>   "author": "OnePgr, Inc <amsystango@gmail.com>",
>   "private": true,
>   "scripts": {
>     "postinstall": "install-app-deps && electron-rebuild --force",
>     "start": "npm install && npm run compile && electron ./app",
>     "compile": "rimraf app/out && tsc",
>     "pack": "npm run compile && build --dir",
>     "dist": "npm run compile && build",
>     "release": "npm run compile && build",
>     "installer": "node ./app/installer.js",
>     "pack-win": "electron-packager app/  Clientele --platform=win32
> --arch=x64 --out dist/ --overwrite --icon=./app/assets/img/Clientele.ico
> --prune=true"
>   },
>   "build": {
>     "appId": "org.develar.onshape",
>     "category": "public.app-category.graphics-design",
>     "icon": "./app/assets/img/icon.icns",
>     "dmg": {
>       "contents": [
>         {
>           "x": 110,
>           "y": 150
>         },
>         {
>           "x": 240,
>           "y": 150,
>           "type": "link",
>           "path": "/Applications"
>         }
>       ]
>     },
>     "win": {
>       "publish": null,
>       "msi": true,
>       "icon": "./app/assets/img/icon.ico"
>     },
>     "linux": {
>       "publish": null,
>       "target": [
>         "deb",
>         "AppImage"
>       ]
>     }
>   },
>   "devDependencies": {
>     "asar": "^0.7.2",
>     "electron": "1.4.3",
>     "electron-builder": "^10.17.3",
>     "electron-packager": "^8.4.0",
>     "electron-prebuilt": "^0.31.0",
>     "electron-rebuild": "^1.5.7",
>     "electron-winstaller": "^2.4.1",
>     "fs-jetpack": "^0.7.0",
>     "grunt": "^1.0.1",
>     "gulp": "^3.9.0",
>     "gulp-copy": "0.0.2",
>     "gulp-uglify": "^1.3.0",
>     "gulp-usemin": "^0.3.12",
>     "q": "^1.4.1",
>     "rcedit": "^0.3.0",
>     "rimraf": "^2.5.4",
>     "typescript": "^2.0.3",
>     "node-hid": "^0.5.4"
>   }}

Thanks
<https://www.npmjs.com/package/node-hid#using-node-hid-in-nwjs-projects>

limpkin, 2017-04-05:
Hello there,

you may take inspiration from our own project here: https://github.com/limpkin/mooltiapp we have already done what you're trying to do.

omardavid55, 2017-04-24:
@vnsharma-systango Did you ever figure out this issue?  I am trying to use Node-HID library and am getting a dll error when I try to run the  var HID = require('node-hid'); line of code.  My issue is probably different from yours but I would like to learn from your problem.  Thanks.

mcammaert, 2017-05-04:
Hi, this is probably because your local electron version is different from what your application expects.
You can fix this by installing electron as dev dependency:

`npm install electron --save-dev`

and running this from node_modules instead of your local electron installation

`./node_modules/.bin/electron .`

closed 2018-01-01 by todbot