node-hid/node-hid #193: Xbox One controller not listed by 'HID.devices()'

lsevero, 2017-03-26:
My Xbox One controller is not listed using HID.devices(), and if I try to open it with its vendorID and productID I got the error: "Error: cannot open device with vendor id 0x45e and product id 0x2d1".

My Xbox One controller is working perfectly fine with other programs like emulators or antimicro. I am using Arch Linux with linux kernel 4.10.5 and node v7.7.3.

todbot, 2017-03-26:
Did you try running your program with 'sudo' or have you given that device non-root permissions with udev?

lsevero, 2017-03-26:
Yes, I tried to run HID.devices() with sudo as well, not listed, cannot open it with its vid and pid as well. This is the only controller that I have which was not listed, my x360 and ps3 controllers are ok.

Others programs like antimicro (map the controller buttons to the keyboard) that I use with this controller work just fine without sudo, so I don't think this is a permission issue.

todbot, 2017-03-26:
Hi,
If it's not listed in `HID.devices()` then node-hid can't access it.  This means either it's not a HID device or not in HID mode.

closed 2021-03-01 by todbot