node-hid/node-hid #194: process.exit triggers Node to hang

bminer, 2017-04-06:
After initializing a HID device and attaching a "data" event listener, Node can hang when `process.exit()` is called.  I believe this is because node-hid is doing a blocking read and keeping Node alive.  As soon as data is read from the HID device or the device is unplugged, Node will exit as expected.

For example, this code exhibits the problem:

```
const hid = require("node-hid");
var device = new hid.HID(...);
device.on("data", function() {});
process.nextTick(() => process.exit(0) );
```

I am running node-hid 0.5.4 on node 7.8.0, Linux Mint 18.1, Kernel 4.4.

todbot, 2017-04-06:
Yes, you need to close the device before exiting.

bminer, 2017-04-06:
This *feels* wrong, though, doesn't it?

The [libuv book](https://nikhilm.github.io/uvbook/threads.html#id1) says: "A well designed program would have a way to terminate long running workers that have already started executing. Such a worker could periodically check for a variable that only the main process sets to signal termination."

The question is: can this be easily done for node-hid?

bminer, 2017-04-06:
As a "workaround", can we just close the HID handle when Node's "exit" event is fired?  This should cause the read to fail and everything to clean-up nicely, right?

Or maybe this is just a documentation problem?  The docs for `process.exit` should probably read: "This might not actually cause Node to exit because of the thread pool, etc."

bminer, 2017-05-03:
I'm closing this issue since this is more of a libuv/node issue upstream.  `process.exit` doesn't really behave very well, since it doesn't immediately tell Node to exit... Node waits until all running threads complete their work... which could a very long time later.

closed 2017-05-03 by bminer