node-hid/node-hid #196: Electron Packager with node-hid

ovsokolov, 2017-04-19:
Hi,

I'm using Electron v1.6.5. 
Initially i was getting error with node-hid during development about version mismatch. node-hid requires version 50 and got 53.
It was fixed by rebuilding against different electron version 
npm rebuild --runtime=electron --target=1.4.4 --disturl=https://atom.io/download/atom-shell --build-from-source

but now after runing electron-packager the error comes back for windows executable it requires Node version 50 but electron 1.6.5 using version 53.

ovsokolov, 2017-04-19:
resolved. we can specify --version on the packager to match node-hid version build with

closed 2017-04-19 by ovsokolov

jsuiker, 2017-04-26:
@ovsokolov could you please provide an example?

ovsokolov, 2017-04-28:
you can specify specific electron version 
electron-packager . --overwrite --version=1.4.4 --asar=true --platform=win32 --arch=all --icon=assets/icons/win/icon.ico --prune=true --out=release-builds ......