node-hid/node-hid #197: Fairbanks scale available in Windows but invisible on Mac

wallyfoo, 2017-04-20:
I wanted to make sure I tested this as thoroughly as I could before I came to you guys. Thanks to google sunsetting chrome apps for the desktop, I'm working on getting an electron app to read a scale. I'd like it to be cross-platform, but that's looking less and less likely.

I have a Fairbanks scale, a HID device, (2919,21854). Node-hid will find and read from this scale in my simple electron app on Windows.

I admit that Fairbanks says their scale isn't supported on Mac, but here's the thing: on the Mac, System Profiler sees this scale when plugged in. A Python module I found will read the raw USB from this scale. Node-usb on the Mac will see this scale (I've not gone as far as trying to read from it yet). Node-hid ignores it completely, and if you try to feed it the correct vid, pid combo, the electron app crashes thusly:

`App threw an error during load
Error: cannot open device with vendor id 0xb67 and product id 0x555e
    at new HID (/helloworld/node_modules/node-hid/nodehid.js:28:17)
    at Object.<anonymous> (/helloworld/main.js:9:14)
    at Object.<anonymous> (/helloworld/main.js:32:3)
    at Module._compile (module.js:571:32)
    at Object.Module._extensions..js (module.js:580:10)
    at Module.load (module.js:488:32)
    at tryModuleLoad (module.js:447:12)
    at Function.Module._load (module.js:439:3)
    at loadApplicationPackage (/helloworld/node_modules/electron/dist/Electron.app/Contents/Resources/default_app.asar/
main.js:283:12)
    at Object.<anonymous> (/helloworld/node_modules/electron/dist/Electron.app/Contents/Resources/default_app.asar/main
.js:325:5)`

Node-hid is the perfect choice for the cross-platform nature of this app -- you don't have to install any extra dependencies on Windows for it -- but for some reason, the node-hid module refuses to admit that the scale is connected to my Mac, even when the OS says it is. Any guidance would be most appreciated.

todbot, 2017-04-21:
Hi,
What does `show-devices.js` show? (i.e. running `console.log(HID.devices())`)  If the device does not show up there, then node-hid cannot access it.  This could be caused by another device driver claiming it (even the OS if it appears as a keyboard HID device)

wallyfoo, 2017-04-21:
I absolutely appreciate where you're coming from here, but the OS sees it, and from the same sample application, node-usb sees it when I substitute that lib in place of node-hid. Substitute node-hid back in, and the device log does not display the device.

Here's the output from the node-usb (https://github.com/tessel/node-usb) on the Mac

`[ Device {
    busNumber: 20,
    deviceAddress: 20,
    deviceDescriptor: 
     { bLength: 18,
       bDescriptorType: 1,
       bcdUSB: 272,
       bDeviceClass: 0,
       bDeviceSubClass: 0,
       bDeviceProtocol: 0,
       bMaxPacketSize0: 64,
       idVendor: 2919,
       idProduct: 21854,
       bcdDevice: 4,
       iManufacturer: 1,
       iProduct: 2,
       iSerialNumber: 3,
       bNumConfigurations: 1 },
    portNumbers: [ 1 ] },
  Device {
    busNumber: 20,
    deviceAddress: 3,
    deviceDescriptor: 
     { bLength: 18,
       bDescriptorType: 1,
       bcdUSB: 512,
       bDeviceClass: 0,
       bDeviceSubClass: 0,
       bDeviceProtocol: 0,
       bMaxPacketSize0: 64,
       idVendor: 1452,
       idProduct: 628,
       bcdDevice: 1572,
       iManufacturer: 1,
       iProduct: 2,
       iSerialNumber: 3,
       bNumConfigurations: 1 },
    portNumbers: [ 4 ] },
  Device {
    busNumber: 20,
    deviceAddress: 2,
    deviceDescriptor: 
     { bLength: 18,
       bDescriptorType: 1,
       bcdUSB: 513,
       bDeviceClass: 239,
       bDeviceSubClass: 2,
       bDeviceProtocol: 1,
       bMaxPacketSize0: 64,
       idVendor: 1452,
       idProduct: 33424,
       bcdDevice: 326,
       iManufacturer: 1,
       iProduct: 2,
       iSerialNumber: 0,
       bNumConfigurations: 1 },
    portNumbers: [ 3 ] },
  Device {
    busNumber: 20,
    deviceAddress: 0,
    deviceDescriptor: 
     { bLength: 18,
       bDescriptorType: 1,
       bcdUSB: 256,
       bDeviceClass: 9,
       bDeviceSubClass: 255,
[... 138 lines cut ...]
    interface: 0,
    usagePage: 65280,
    usage: 11 },
  { vendorId: 1452,
    productId: 628,
    path: '\\\\?\\hid#vid_05ac&pid_0274&mi_01&col03#7&112d567b&0&0002#{4d1e55b2-f16f-11cf-88cb-001111000030}',
    serialNumber: 'D3H51655SF1FTV4AG6JS',
    manufacturer: 'Apple Inc.',
    product: 'Keyboard / Boot',
    release: 1572,
    interface: 1,
    usagePage: 65280,
    usage: 6 },
  { vendorId: 1452,
    productId: 628,
    path: '\\\\?\\hid#vid_05ac&pid_0274&mi_01&col04#7&112d567b&0&0003#{4d1e55b2-f16f-11cf-88cb-001111000030}',
    serialNumber: 'D3H51655SF1FTV4AG6JS',
    manufacturer: 'Apple Inc.',
    product: 'Keyboard / Boot',
    release: 1572,
    interface: 1,
    usagePage: 12,
    usage: 1 } ]`

And that's all the data I have. Node-hid sees it as a proper HID device on Windows, but not on the Mac even though other libraries will see it on the Mac. I simply don't understand the discrepancy.

todbot, 2017-04-22:
Hi, I fixed the formatting of your update so it should be readable now.

Can you do the `node-usb` output on Windows for your device?  Because in the output you provide above, while the `idProduct` and `idVendor` match, you can't see the interface config descriptors.  Specifically, we're looking for `bInterfaceClass` inside the interface descriptor of the config descriptor.  In order for a USB device to be a HID device, either its `bDeviceClass` is `3` or one of its interface's `bInterfaceClass` is `3`.

To look for that, here's a version of the original node-usb `lsusb.js` reworked for tessel's node-usb: https://gist.github.com/todbot/2fc331a48c4a7be86f1c8949def4c317
Try running it and creating a gist with the output, then link that output here.  (Please don't post full logs like that to git issues as it makes it really hard to have discussions otherwise)

wallyfoo, 2017-04-22:
Nice. Thanks for your patience in all this, by the way. After the adventure that is installing node-usb in Windows 10, I have the specific device output for you from the other library.

[Gist to the output](https://gist.github.com/wallyfoo/9744c65fe707951a56a0dae2cc023382)

So from what you say, since the bDeviceClass is zero on Windows, (though the Interface descriptor.bInterfaceClass is 3), then it shouldn't show up as a HID device. But in Windows, it does. In Windows, your lib reads input like a champ.

jakereps, 2018-06-27:
Any update on this? Running into the same issue here on macOS 10.13.5, Node v8.11.3, npm 5.6.0, node-hid ^0.7.2 and the device interface showing the following through `node-usb`:

```js
descriptor:
   { bLength: 9,
     bDescriptorType: 4,
     bInterfaceNumber: 0,
     bAlternateSetting: 0,
     bNumEndpoints: 2,
     bInterfaceClass: 3,
     bInterfaceSubClass: 141,
     bInterfaceProtocol: 141,
     iInterface: 0,
     extra: <Buffer 09 21 01 01 00 01 22 4f 00>,
     endpoints: [ [Object], [Object] ] },
```

wallyfoo, 2018-06-27:
I ended up having to create a hybrid solution that looks for the different library depending on which platform the application runs on. It's not perfect, but it works.

todbot, 2019-12-05:
I think this works now as of node-hid v1.1.0

closed 2019-12-05 by todbot