node-hid/node-hid #199: Handle device disconnect the reconnect

safinn, 2017-04-21:
Hello,

I'm wondering if node-hid does anything to keep reading the data from a device when its disconnected and reconnected again (removed from the usb slow and added again)?

I want to keep reading the device even after it's been removed and then plugged in.

I don't see any events fired such as device attached or removed for me to handle this functionality myself.

todbot, 2017-04-21:
Hi,
Node-hid doesn't support device insertion and detection support.   You will get an error if trying to write to a disconnected device and data will stop streaming if you're doing a read on a disconnected device.  In those cases, you should set up a process to close the existing device variable and re-open.

If you want proper USB plug/unplug detection, I recommend checking out https://github.com/MadLittleMods/node-usb-detection

safinn, 2017-04-21:
The recommended package looks good. Thanks for the help.

closed 2017-04-21 by safinn