node-hid/node-hid #201: Error: cannot open device with vendor id 0x46d and product id 0xc52b

karenYerevan, 2017-04-22:
in list usb device exist the device, but when I want open device it write error

E:\nodehid\node_modules\node-hid\nodehid.js:28
    this._raw = new (Function.prototype.bind.apply(binding.HID,
                ^

Error: cannot open device with vendor id 0x46d and product id 0xc52b
    at Error (native)
    at new HID (E:\nodehid\node_modules\node-hid\nodehid.js:28:17)
    at Object.<anonymous> (E:\nodehid\index.js:5:14)
    at Module._compile (module.js:570:32)
    at Object.Module._extensions..js (module.js:579:10)
    at Module.load (module.js:487:32)
    at tryModuleLoad (module.js:446:12)
    at Function.Module._load (module.js:438:3)
    at Module.runMain (module.js:604:10)
    at run (bootstrap_node.js:394:7)

todbot, 2017-04-22:
The device is opened by some other process then I think.

jay-cordo, 2017-05-23:
is there a way to close the device ?

dimitrisal, 2019-03-26:
I have the same issue did you find any solution?

todbot, 2019-03-26:
Is the device a mouse or a keyboard?  Those are grabbed by the OS for security reasons.

dimitrisal, 2019-03-27:
It is a scanner device, although it is recognised as a keyboard. Is there any workaround?

karenYerevan, 2019-03-27:
if it is a scanner device, you must catch event frm client side.
if you use electrone.js it is not dificult.
but if you anyway want catch data from server-side in nodejs, you must wrire  API on  C++

impactcolor, 2020-10-21:
How do you capture details with electron on client side?
@karenYerevan karenYerevan

todbot, 2020-10-21:
Closing this issue, as it's yet another "why can't I open a keyboard device with node-hid".

closed 2020-10-21 by todbot