node-hid/node-hid #202: Get an error in Electron trying to print to the console the devices object

omardavid55, 2017-04-24:
Hello @todbot,

I followed the instructions for electron:
[image]

Here is my package.json:

  "scripts": {
    "start": "electron .",
    "postinstall": "electron-rebuild --force"
  },
  "author": "david hernandez",
  "license": "ISC",
  "dependencies": {
    "electron": "^1.6.5",
    "node-hid": "^0.5.4"
  },
  "devDependencies": {
    "electron-prebuilt": "^1.4.13",
    "electron-rebuild": "^1.5.7"
  }

Here is my script:
  <script>
   var HID = require('node-hid');
   var devices = HID.devices();
   console.log("devices.vendorID");

Here is the error I see:
[image]

Any help would be greatly appreciated. Thank you.

omardavid55, 2017-04-24:
Looks like the error didn't show up in the original post.
[image]

todbot, 2017-04-24:
Hi,
Can you get this project to work?  https://www.npmjs.com/package/electron-hid-toy

omardavid55, 2017-04-24:
I tried that one among some other HID/USB apis and I couldn't get them up and running in electron.  This one found failed to install the dependencies.  I can't remember the error though.

todbot, 2017-04-24:
Yes, please try again.  'electron-hid-toy' is specifically designed to be an extremely simple example of `node-hid` used with Electron.  If it doesn't work for you, I think there's some larger issue.

omardavid55, 2017-04-24:
Ok, I downloaded it again and I already see that I was missing some files from the first download a few days ago.  I will report how it goes.  Thank you.

closed 2018-01-01 by todbot