node-hid/node-hid #207: Distributing with `pkg`

gabssnake, 2017-05-22:
Thanks for this great module!

Been willing to use it with [pkg](https://github.com/zeit/pkg). Here’s what I’ve got:
```
# On Darwin 16.6.0 - node v7.10.0 - npm v4.2.0

# Stealing pre-compiled binary
cp node_modules/node-hid/build/Release/HID.node ./bin/HID.node

# Ugly path setting
sed -ie "s@require(binding_path)@require(path.join(process.cwd(), 'bin', 'HID.node'))@" node_modules/node-hid/nodehid.js

# Compiling
pkg index.js -o ./bin/app

# Will throw
./bin/app
```

I’m seeing:
```
> ./bin/app

buffer.js:821
    throw new RangeError('Index out of range');
          ^

RangeError: Index out of range
    at checkOffset (buffer.js:821:11)
    at Buffer.readInt32LE (buffer.js:974:5)
    at evalmachine.<anonymous>:0
    at evalmachine.<anonymous>:0
    at Object.Module._extensions..node (evalmachine.<anonymous>:0)
    at Module.load (module.js:487:32)
    at tryModuleLoad (module.js:446:12)
    at Function.Module._load (module.js:438:3)
    at Module.require (module.js:497:17)
    at Module.require (evalmachine.<anonymous>:0)
```

Does `node-hid` has to be compiled for the same exact target?
What am I missing?

gabssnake, 2017-05-22:
Sorry, it was an issue with `pkg` itself: https://github.com/zeit/pkg/issues/104

closed 2017-05-22 by gabssnake

todbot, 2017-05-22:
No problem.  And if you find a way of using `pkg` with `node-hid` that creates a single executable file (instead of needing the HID.node file in the same directory), please let me know!