node-hid/node-hid #208: could not open by vid,pid

percinor, 2017-06-10:
i have a usb rfid reader,  it read  rfid tag work  well in notepad or ms word,
my environment are  os win10 64, npm version 3.10.10 ,node version 6.10.0.
var devices = HID.devices();   console.log(devices); its work ok and show result.
//
{"vendorId":1241,"productId":5890,"path":"\\\\?\\hid#vid_04d9&pid_1702#6&314f4e87&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}","manufacturer":"XAOJU LABORATORIES","product":"C8051F3xx Development","release":0,"interface":-1,"usagePage":1,"usage":6},
so i use the vid and pid like below ,
var devices = HID.devices();
var device = new HID.HID(1241,5890);
  device.on("data", function(data) {
	console.log(data);
});
but could not work ,error like below ,i use path get same error 

Error: cannot open device with vendor id 0x4d9 and product id 0x1702
    at Error (native)
    at new HID (D:\Dev\noexpress\exhid\node_modules\node-hid\nodehid.js:28:17)
    at Object.<anonymous> (D:\Dev\noexpress\exhid\routes\users.js:12:14)
    at Module._compile (module.js:570:32)
    at Object.Module._extensions..js (module.js:579:10)
    at Module.load (module.js:487:32)
    at tryModuleLoad (module.js:446:12)
    at Function.Module._load (module.js:438:3)
    at Module.require (module.js:497:17)
    at require (internal/module.js:20:19)

todbot, 2017-06-10:
Hi,
If this device is acting like a keyboard, then you will not be able to open it with non-OS programs like node-hid.  This is a security measure by the OS so you can't install a keytracker.

Many USB RFID readers have a set of switches to put it into into different USB modes.  One of these modes is a "HID custom" mode where it's a HID device but not a keyboard.  In that mode, you can use node-hid with it.

percinor, 2017-06-11:
thanks you answer ,it very helpful, could you tell me which model or type  usb-rfid reader  that have a set of switches ,so  i can buy it . thanks  again.

closed 2019-11-23 by todbot