node-hid/node-hid #209: Doesn't work in Node v8

DarrenMack-OD, 2017-06-13:
Hi,
We are using node-hid to interface with a BLINK device, but we have upgraded to node v8 and NPM v5 and node-hid seems to fall over... Any chance this will be supported soon?

_Error: The module '/Users/****/Workspace/blink/node_modules/node-hid/build/HID/Release/HID.node' was compiled against a different Node.js version using
NODE_MODULE_VERSION 51. This version of Node.js requires
NODE_MODULE_VERSION 57. Please try re-compiling or re-installing
the module (for instance, using `npm rebuild` or `npm install`).
    at Object.Module._extensions..node (module.js:598:18)
    at Module.load (module.js:503:32)
    at tryModuleLoad (module.js:466:12)
    at Function.Module._load (module.js:458:3)
    at Module.require (module.js:513:17)
    at require (internal/module.js:11:18)
    at Object.<anonymous> (/Users/****/Workspace/blink/node_modules/node-hid/nodehid.js:9:15)
    at Module._compile (module.js:569:30)
    at Object.Module._extensions..js (module.js:580:10)
    at Module.load (module.js:503:32)_

todbot, 2017-06-13:
Hi!  What OS are you on?  What process are you using to install node-hid?

DarrenMack-OD, 2017-06-14:
Hi, I am on Mac OS X Sierra 10.12.5 But I installed node manually from the node website. I have homebrew, but ran into complications with node when it was being controlled by homebrew.

I am installing node-hid as part of a process using package.json. It does the same if I try to install it manually as well...

ert78gb, 2017-06-28:
I have a same issue Mac 10.12.5 node 8.1.2
Npm log attached
[2017-06-28T09_53_10_075Z-debug.log.zip](https://github.com/node-hid/node-hid/files/1108263/2017-06-28T09_53_10_075Z-debug.log.zip)

KevinBeckers, 2017-09-26:
Same issue

ert78gb, 2017-09-26:
try to use `npm rebuild node-hid --force` or just simple delete the node_modules folder and `npm i`

KevinBeckers, 2017-09-26:
I used   `"postinstall": "electron-rebuild --force"` in my package.json

ert78gb, 2017-09-26:
yes electron-rebuild need if use in electron

todbot, 2017-09-26:
At least in the case of @ert78gb, it looks like you're using an old erversion (0.5.4) of `node-hid`.  Try the latest, 0.5.7, and see if that solves the problem.

For @DarrenMack-OD, are you using NW.js or Electron?  The version mismatch error is usually because of that.  Or similarly, are you using different versions of node, compiling `node-hid` using one version and then running a script that uses it in another version?  That will also cause the mismatch error.

Also, unfortunately npm-log doesn't capture all the error messages output when installing & compiling native modules.  Next time you have this issue, please make an note of the errors printed in the terminal and not just in the log.  Thanks!

ert78gb, 2017-09-26:
@todbot It is working for me. thx. I just forgot to signal it.

closed 2017-09-26 by todbot