node-hid/node-hid #210: Electron, node-hid, Windows 7, segmentation fault on close()

randomf, 2017-06-28:
Hello,

I'm getting segmentation fault upon `close()`. Searching, connecting and R/W to the device works fine. I even tried code suggested in [https://github.com/node-hid/node-hid/issues/129](url) without a luck. Same code works on Linux and Mac. Any ideas?

Node 6.11
Windows 7 SP1
Electron 1.4.16
node-hid 0.5.4

todbot, 2017-06-28:
Hi, could you show the code you're using both before and after `close()`?

randomf, 2017-06-28:
Hi,

here it goes:
```
device.write(Array.from(new Buffer('020000', 'hex')));

// some code is being processed in the meantime

device.pause();
device.removeAllListeners("data");
device.removeAllListeners("error");
device.close();
```

There is no node-hid relevant code executed after.

todbot, 2017-06-28:
What are you doing immediately after close?  If you're doing process.exit() or similar you will likely have problems because Node doesn't have time to clean up EventListeners

randomf, 2017-06-28:
I'm waiting for an user input in a React application. I don't call process.exit().

pablozandona, 2017-07-06:
Same problem here.

Node 6.10.3
Windows 7 SP12
node-hid 0.5.4

todbot, 2017-07-06:
Could someone experiencing this issue submit a small complete code example that illustrates the crash?

todbot, 2019-11-23:
Closing as I am suspecting this issue has been addressed by moving to libusb/hidapi.

closed 2019-11-23 by todbot