node-hid/node-hid #211: No devices found

micopiira, 2017-06-30:
Running the [show-devices.js](https://github.com/node-hid/node-hid/blob/master/src/show-devices.js) example returns an empty array, even though there are HID devices attached.

I'm using Windows 10 64-bit and Node version v.7.2.1 and tried also running from an Administrator command prompt.

vinay-kumar, 2017-10-16:
Have you solved it. This is happening with me also.

Everything install fine and Barcode reader connected. working fine on Linux but not on windows 10

todbot, 2017-10-16:
Hi,
On Windows you will not be able to access keyboards and mice.

vinay-kumar, 2017-10-18:
@todbot it's not keyboard or mice. I am using barcode scanner device

todbot, 2017-10-18:
Hi @vinay-kumar,

What kind of barcode scanner is it? 

Some of them appear as a keyboard to the OS (so that you can just open up Excel and have it "type" codes into a spreadsheet). Some appear as a "HID data" device, which is where you can use node-hid.  Some scanners have a switch to select between those two modes.

closed 2019-07-25 by todbot

kuangping-kp, 2022-07-20:
鼠标和键盘没法用node-hid？

todbot, 2022-07-20:
Node-hid cannot read keyboards or mice. 

https://github.com/node-hid/node-hid#devices-node-hid-cannot-read