node-hid/node-hid #212: webpack --target node fails: Can't resolve 'aws-sdk'

jcalfee, 2017-07-12:
I'm using `trezor.js-node` trying to webpack it for node but I'm getting an error from this module.  

It fails in the node-hid module: ` Error: Can't resolve 'aws-sdk'` .. 

This re-creates the error:
```bash
npm i webpack 
echo "require('node-hid')" > hid.js
webpack --target node hid.js out.js
```

There are some warnings then 4 errors.  It looks like this is the main issue:
```bash
ERROR in ./node_modules/node-hid/node_modules/node-pre-gyp/lib/info.js
Module not found: Error: Can't resolve 'aws-sdk'
```

I'm not sure about this one.  Is there something I can do in my webpack configuration or is there something we can do here to improve portability?

todbot, 2017-07-12:
Hi,
The `aws-sdk` is a  dependency of `node-pre-gyp`, the standard Node library used to compile native modules. `node-hid` doesn't use `aws-sdk` directly.

That's a weird error you're getting, I've never seen it before.  It looks like your copy of `npm` can't pull down the full npmjs directory or something.  But `aws-sdk` is there: https://www.npmjs.com/package/aws-sdk

jcalfee, 2017-07-13:
Loading the module works fine:
```bash
node ./hid.js
```

Something about the code is making it hard to package and use on the go (without npm)..  I suspect this is re-producible..
```
lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 16.04 LTS
Release:	16.04
Codename:	xenial
```

jcalfee, 2017-07-13:
I'll focus my attention on node-pre-gyp .. thank you

closed 2017-07-13 by jcalfee

GuillaumeCisco, 2017-10-19:
Hey @jcalfee , did you succeed fixing this issue?
I have the same problem, could you point me to a directino on node-pre-gyp, please.

Thank you, and sorry for commenting on this closed issue

todbot, 2017-10-20:
Hi,
`node-hid` doesn't use `node-pre-gyp` anymore just because of these kinds of issues. Please try the latest version of `node-hid` to see if it works for you.

GuillaumeCisco, 2017-10-20:
Thanks @todbot 
This error came from different errors when trying to work with electron and grpc.
For the full record and how I fixed it, here the full detailed issue: 
https://github.com/grpc/grpc/issues/13049#issuecomment-337677610

Maybe it will help some ;)