node-hid/node-hid #213: Pre-built binaries not found for node-hid@0.5.4 and node@8.1.4

jcalfee, 2017-07-16:
I'm trying to upgrade to Node 8 and start migrating to the new package-lock.json feature.  My project works under node v7 but not under v8.  Here is the error:

```bash
node-pre-gyp ERR! Tried to download(404): https://github.com/node-hid/node-hid/releases/download/0.5.4/HID-v0.5.4-node-v57-linux-x64.tar.gz 
node-pre-gyp ERR! Pre-built binaries not found for node-hid@0.5.4 and node@8.1.4 (node-v57 ABI) (falling back to source compile with node-gyp) 
node-pre-gyp http 404 status code downloading tarball https://github.com/node-hid/node-hid/releases/download/0.5.4/HID-v0.5.4-node-v57-linux-x64.tar.gz 
...
npm ERR! Failed at the node-hid@0.5.4 install script.
```

allotria-precio, 2017-08-08:
Same problem here. It would be nice if there existed builds for node 8.

gotleg, 2017-08-28:
Same here with node 6.11 : 
node-pre-gyp ERR! Pre-built binaries not found for node-hid@0.5.4 and node@6.11.2 (node-v48 ABI) (falling back to source compile with node-gyp) 
node-pre-gyp http 404 status code downloading tarball https://github.com/node-hid/node-hid/releases/download/0.5.4/HID-v0.5.4-node-v48-linux-arm.tar.gz

It's really a problem...

todbot, 2017-08-28:
Hi! Please update to 0.5.7. That should fix these issues.

closed 2018-01-01 by todbot

sgehrman, 2018-01-25:
same problem, what's the solution?