node-hid/node-hid #214: No error is reported on Windows

darkwave, 2017-07-24:
When a write() instruction is issued and **an error occurs** it does not generate any report or exception on windows.

For example using try/catch:
```javascript
try {
    device.write(command);
} catch (e) {
  console.log(e);
}
```
This code works (generate exception) on GNU/Linux system.

todbot, 2017-07-24:
Hi,
What version of Windows and Node are you on?

todbot, 2017-07-24:
Also, what is the exact command you're writing and to what type of HID device?

darkwave, 2017-07-28:
@todbot let me give you some more information... :nerd_face: 

I'm developing a **Blockly** based [**environment that uses a 2.4G dongle to control a robot**](https://github.com/darkwave/mbot-scratch-blocks/).

You can find full [source code using your module inside](https://github.com/darkwave/mbot-scratch-blocks/blob/master/hid_mbot.js#L105) in the repo.

I'm issuing commands like this:
```javascript
device.write([0, 7 ,0xff, 0x55, 0x04, 0x02, 0x01, 0x1, 0x03])
```

Official robot platform page about [dongle on MakeBlock website](http://learn.makeblock.com/en/2-4g-wireless-serial/)

dmesg output:
```
[91781.048605] usb 1-2: new full-speed USB device number 7 using xhci_hcd
[91781.190281] usb 1-2: New USB device found, idVendor=0416, idProduct=ffff
[91781.190292] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[91781.190298] usb 1-2: Product: RF UART
[91781.190303] usb 1-2: Manufacturer: MemsArt 
[91781.193778] input: MemsArt  RF UART as /devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2:1.0/0003:0416:FFFF.0002/input/input15
[91781.194225] hid-generic 0003:0416:FFFF.0002: input,hidraw0: USB HID v1.10 Device [MemsArt  RF UART] on usb-0000:00:14.0-2/input0
```

closed 2019-11-23 by todbot