node-hid/node-hid #216: device.setNonBlocking is not documented.

SchizoDuckie, 2017-07-27:
While experimenting with monitoring multiple RFID readers on a loop I found what turned out some weird blocking behavior, events queueing up and being batched.

Looking through issues and pull requests I found out about [device.setNonBlocking](https://github.com/node-hid/node-hid/blob/c05ae0bc08395eecfb2d70b17ba1382e8be2bfa7/src/HID.cc#L83) which solves this specific issue.

I'll craft a PR to address this, but to fully understand this issue, could anyone enlighten me on why this is not the default behavior?

peitschie, 2019-06-19:
Just bumping this as I was having issues with node hanging on exit unless I call `device.setNonBlocking(true)`.

Is this intended to be an internal-only function?

todbot, 2019-06-19:
As mentioned in another issue, `node-hid` is a slim wrapper around https://github.com/libusb/hidapi

I will update the README to reference `device.setNonBlocking()` though. Thanks!  

(I hope the recent changes happening to `hidapi` will fix some of the gremlins impacting a few `node-hid` users and let us understand better how this nonblocking stuff is messing up)

closed 2019-06-19 by todbot