node-hid/node-hid #217: Do not kill node process, if HID API initialization fails

RomanSaveljev, 2017-08-02:
This appears to be rather bold move to [kill](https://github.com/node-hid/node-hid/blob/master/src/HID.cc#L590) the whole node process, whenever HID API initialization fails.

Our product's CI runs in an USB-less environment (container). The application can not start at all, unless we workaround it by manually uninstalling the `node-hid` module.

Can `node-hid` handle this scenario more gracefully. It is fine, if `require('node-hid')` throws.

todbot, 2017-08-14:
Agreed, this is pretty drastic. Trying out some changes now.

todbot, 2017-08-16:
A fix for this is in the latest version. Let me know if you have any further issues and thanks for pointing it out.

closed 2017-08-16 by todbot