node-hid/node-hid #219: Fail to build on RASPBIAN JESSIE, node 6.11.0

sxwei123, 2017-08-08:
Hi, 
I tried to install node-hid on raspbian, can anyone help me find out what is the problem?
I am using RASPBIAN JESSIE: Linux raspberrypi 4.4.50-v7+ #970 SMP Mon Feb 20 19:18:29 GMT 2017 armv7l GNU/Linux
node version: v6.11.0 
I followed the guide of compiling from source, installed node-gyp, node-pre-gyp, gcc-4.8 g++-4.8, libusb-1.0-0, libusb-1.0-0-dev

npm install node-hid

> node-hid@0.5.4 install /home/pi/Desktop/smartBox/smartBox-client/node_modules/node-hid
> node-pre-gyp install --fallback-to-build

node-pre-gyp ERR! Tried to download(404): https://github.com/node-hid/node-hid/releases/download/0.5.4/HID-v0.5.4-node-v48-linux-arm.tar.gz 
node-pre-gyp ERR! Pre-built binaries not found for node-hid@0.5.4 and node@6.11.0 (node-v48 ABI) (falling back to source compile with node-gyp) 
make: Entering directory '/home/pi/Desktop/smartBox/smartBox-client/node_modules/node-hid/build'
  CC(target) Release/obj.target/hidapi/hidapi/libusb/hid.o
../hidapi/libusb/hid.c:47:20: fatal error: libusb.h: No such file or directory
 #include <libusb.h>
                    ^
compilation terminated.
hidapi.target.mk:94: recipe for target 'Release/obj.target/hidapi/hidapi/libusb/hid.o' failed
make: *** [Release/obj.target/hidapi/hidapi/libusb/hid.o] Error 1
make: Leaving directory '/home/pi/Desktop/smartBox/smartBox-client/node_modules/node-hid/build'
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/lib/node_modules/node-gyp/lib/build.js:258:23)
gyp ERR! stack     at emitTwo (events.js:106:13)
gyp ERR! stack     at ChildProcess.emit (events.js:191:7)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:215:12)
gyp ERR! System Linux 4.4.50-v7+
gyp ERR! command "/usr/bin/nodejs" "/usr/lib/node_modules/node-gyp/bin/node-gyp.js" "build" "--fallback-to-build" "--module=/home/pi/Desktop/smartBox/smartBox-client/node_modules/node-hid/build/Release/HID.node" "--module_name=HID" "--module_path=/home/pi/Desktop/smartBox/smartBox-client/node_modules/node-hid/build/Release"
gyp ERR! cwd /home/pi/Desktop/smartBox/smartBox-client/node_modules/node-hid
gyp ERR! node -v v6.11.0
gyp ERR! node-gyp -v v3.6.2
gyp ERR! not ok 
node-pre-gyp ERR! build error 
node-pre-gyp ERR! stack Error: Failed to execute '/usr/bin/nodejs /usr/lib/node_modules/node-gyp/bin/node-gyp.js build --fallback-to-build --module=/home/pi/Desktop/smartBox/smartBox-client/node_modules/node-hid/build/Release/HID.node --module_name=HID --module_path=/home/pi/Desktop/smartBox/smartBox-client/node_modules/node-hid/build/Release' (1)
node-pre-gyp ERR! stack     at ChildProcess.<anonymous> (/home/pi/Desktop/smartBox/smartBox-client/node_modules/node-hid/node_modules/node-pre-gyp/lib/util/compile.js:83:29)
node-pre-gyp ERR! stack     at emitTwo (events.js:106:13)
node-pre-gyp ERR! stack     at ChildProcess.emit (events.js:191:7)
node-pre-gyp ERR! stack     at maybeClose (internal/child_process.js:891:16)
node-pre-gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:226:5)
node-pre-gyp ERR! System Linux 4.4.50-v7+
node-pre-gyp ERR! command "/usr/bin/nodejs" "/home/pi/Desktop/smartBox/smartBox-client/node_modules/node-hid/node_modules/.bin/node-pre-gyp" "install" "--fallback-to-build"
node-pre-gyp ERR! cwd /home/pi/Desktop/smartBox/smartBox-client/node_modules/node-hid
node-pre-gyp ERR! node -v v6.11.0
node-pre-gyp ERR! node-pre-gyp -v v0.6.31
node-pre-gyp ERR! not ok 
Failed to execute '/usr/bin/nodejs /usr/lib/node_modules/node-gyp/bin/node-gyp.js build --fallback-to-build --module=/home/pi/Desktop/smartBox/smartBox-client/node_modules/node-hid/build/Release/HID.node --module_name=HID --module_path=/home/pi/Desktop/smartBox/smartBox-client/node_modules/node-hid/build/Release' (1)
npm WARN test-cloud-mqtt-client@1.0.0 No repository field.
npm ERR! Linux 4.4.50-v7+
npm ERR! argv "/usr/bin/nodejs" "/usr/bin/npm" "--registry=https://registry.npm.taobao.org" "--cache=/home/pi/.npm/.cache/cnpm" "--disturl=https://npm.taobao.org/dist" "--userconfig=/home/pi/.cnpmrc" "install" "node-hid"
npm ERR! node v6.11.0
npm ERR! npm  v3.10.10
npm ERR! code ELIFECYCLE

npm ERR! node-hid@0.5.4 install: `node-pre-gyp install --fallback-to-build`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the node-hid@0.5.4 install script 'node-pre-gyp install --fallback-to-build'.
npm ERR! Make sure you have the latest version of node.js and npm installed.
npm ERR! If you do, this is most likely a problem with the node-hid package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-pre-gyp install --fallback-to-build
npm ERR! You can get information on how to open an issue for this project with:
npm ERR!     npm bugs node-hid
npm ERR! Or if that isn't available, you can get their info via:
npm ERR!     npm owner ls node-hid
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /home/pi/Desktop/smartBox/smartBox-client/npm-debug.log

todbot, 2017-08-08:
Hi,
Those error messages show that libusb is not installed.

sxwei123, 2017-08-08:
Oh, Thanks!

closed 2017-08-08 by sxwei123