node-hid/node-hid #221: node-hid failing on Linux 7

mpab, 2017-08-16:


mpab, 2017-08-16:
Hi Tod - Not sure what's just changed but I seem to be getting the same build failure on multiple systems.
I'm doing an npm i parrot-minidrone

This warning appears briefly:
npm WARN deprecated node-hid@0.5.5: oops bad package.json

and here is the error in my log:
2902 verbose stack Error: node-hid@0.5.5 preinstall: `git submodule update --init`
2902 verbose stack Exit status 128

I know you aren't responsible for parrot-minidrone, but this was working yesterday and you seem to have made some commits which correspond to breaking the npm package.

Regards,
Mick

todbot, 2017-08-16:
Hi @mpab,
Apologies, I screwed up the package.json on a new version I'm preparing of node-hid. I deprecated it but apparently that wasn't enough. I just unpublished it now too.

mpab, 2017-08-16:
Hi Tod, thanks for the fix, I just successfully installed on my Debian (PC) and Raspberry Pi (ARM) systems.

--Mick

closed 2017-08-16 by mpab