node-hid/node-hid #222: Use of node-hid in nw.js 0.24.3

titous01, 2017-08-21:
Hello,
I´m trying to use node-hid in my project in Windows 10 with nw.js version 0.24.3. I can compile it with nw-gyp build and it went greate in the process of compilation. But when I try to use node-hid, with just this line(var hid=require(''node-hid")) my application crashes and no log is produced. I try to use chromium console to issue this and the result is the same. Any clues what´s happening?

Beste Regards,
Werther

todbot, 2017-08-21:
Hi Werther,
That's frustrating there's no log being produced. I don't quite know what to suggest to help debug. But that sort of hard crash sounds like it could only be the DLL being produced isn't being found or is the wrong format. (which normally node produces a readable error for)

One suggestion:  `node-hid` uses a very similar build process to `node-serialport` so it may be there are some hints on their github page about NW.js.  From my cursory inspection, it looks like serialport works with `nw-gyp`. (there is a quirk though in that both of our projects recently moved from `node-pre-gyp` to `prebuild` to handle pre-built binaries and that confuses things)

Two questions: 
* Does `npm install node-hid --build-from-source` work for you?
* Can you install if you use `node-hid@0.5.4`?

titous01, 2017-08-22:
Hello,
I try both of your sugestions but they didn´t work. What I tried was use node.js LTS (6.11.2 64bits), node-hid@0.5.4 and nw.js 0.14.7 and now I can use node-hid. I need to recompile other modules that my project uses like java, ref and ffi.
I´ll try to do that tomorow and let you know if it went succesful.
Thanks for your quick response.
Regards,
Werther

titous01, 2017-09-01:
Hello,
I still cannot make node-hid works wiht the newest version of nw (0.24.3). I recompile with nw-gyp and everything went fine but when I try to use it (require("node-hid")) the chromium crashes whithout any message. 
Do you know if is there anyone using it with this version of nw.js?

todbot, 2017-09-01:
Hi, can you give a link to your project so I could try? I don't have much experience with nw.js but I'd like to figure out what the issue is.

todbot, 2017-09-01:
Also, which OS are you on?

titous01, 2017-09-01:
I´m using Windows 10 64. On Ubuntu 64 is working.

todbot, 2017-09-01:
Oh good so at least you have an example on one OS of it working.
Did you follow all the Windows-specific steps noted here:
http://docs.nwjs.io/en/latest/For%20Users/Advanced/Use%20Native%20Node%20Modules/

titous01, 2017-09-01:
Yes I did nad the the nw-gyp works perfectly. Did you use a DLL to comunicate with? Where this supose to be?

todbot, 2017-09-01:
I don't understand your question.  Native modules in Node are all shared libraries (DLLs).

titous01, 2017-09-01:
I´m sorry. I´m asking if you use an external dll that I must have in some place in my machine.

titous01, 2017-09-01:
This is the output of nw-gyp:
gyp info it worked if it ends with ok
gyp info using nw-gyp@3.4.0
gyp info using node@8.4.0 | win32 | x64
gyp info spawn C:\Python27\python.EXE
gyp info spawn args [ 'C:\\Users\\Werther\\AppData\\Roaming\\npm\\node_modules\\nw-gyp\\gyp\\gyp_main.py',
gyp info spawn args   'binding.gyp',
gyp info spawn args   '-f',
gyp info spawn args   'msvs',
gyp info spawn args   '-G',
gyp info spawn args   'msvs_version=auto',
gyp info spawn args   '-I',
gyp info spawn args   'C:\\Projetos Grails\\EVAPress\\node_modules\\node-hid\\build\\config.gypi',
gyp info spawn args   '-I',
gyp info spawn args   'C:\\Users\\Werther\\AppData\\Roaming\\npm\\node_modules\\nw-gyp\\addon.gypi',
gyp info spawn args   '-I',
gyp info spawn args   'C:\\Users\\Werther\\.nw-gyp\\0.24.3\\common.gypi',
gyp info spawn args   '-Dlibrary=shared_library',
gyp info spawn args   '-Dvisibility=default',
gyp info spawn args   '-Dnode_root_dir=C:\\Users\\Werther\\.nw-gyp\\0.24.3',
gyp info spawn args   '-Dnode_gyp_dir=C:\\Users\\Werther\\AppData\\Roaming\\npm\\node_modules\\nw-gyp',
gyp info spawn args   '-Dnode_lib_file=nw.lib',
gyp info spawn args   '-Dmodule_root_dir=C:\\Projetos Grails\\EVAPress\\node_modules\\node-hid',
gyp info spawn args   '--depth=.',
gyp info spawn args   '--no-parallel',
gyp info spawn args   '-Dv13=1',
gyp info spawn args   '--generator-output',
gyp info spawn args   'C:\\Projetos Grails\\EVAPress\\node_modules\\node-hid\\build',
gyp info spawn args   '-Goutput_dir=.' ]
gyp info spawn C:\Program Files (x86)\MSBuild\14.0\bin\msbuild.exe
gyp info spawn args [ 'build/binding.sln',
gyp info spawn args   '/clp:Verbosity=minimal',
gyp info spawn args   '/nologo',
gyp info spawn args   '/p:Configuration=Release;Platform=x64' ]
Building the projects in this solution one at a time. To enable parallel build, please add the "/m" switch.
  hid.c
  win_delay_load_hook.cc
  hidapi.vcxproj -> C:\Projetos Grails\EVAPress\node_modules\node-hid\build\Release\\hidapi.lib
  HID.cc
  win_delay_load_hook.cc
..\src\HID.cc(67): warning C4290: C++ exception specification ignored except to indicate a function is not __declspec(nothrow) [C:\Projetos Grails\EVAPress\node_modules\node-hid\build\HID.vcxproj]
..\src\HID.cc(70): warning C4290: C++ exception specification ignored except to indicate a function is not __declspec(nothrow) [C:\Projetos Grails\EVAPress\node_modules\node-hid\build\HID.vcxproj]
..\src\HID.cc(152): warning C4290: C++ exception specification ignored except to indicate a function is not __declspec(nothrow) [C:\Projetos Grails\EVAPress\node_modules\node-hid\build\HID.vcxproj]
..\src\HID.cc(163): warning C4290: C++ exception specification ignored except to indicate a function is not __declspec(nothrow) [C:\Projetos Grails\EVAPress\node_modules\node-hid\build\HID.vcxproj]
..\src\HID.cc(207): warning C4996: 'v8::Function::NewInstance': was declared deprecated [C:\Projetos Grails\EVAPress\node_modules\node-hid\build\HID.vcxproj]
  C:\Users\Werther\.nw-gyp\0.24.3\deps\v8\include\v8.h(3788): note: see declaration of 'v8::Function::NewInstance'
..\src\HID.cc(398): warning C4530: C++ exception handler used, but unwind semantics are not enabled. Specify /EHsc [C:\Projetos Grails\EVAPress\node_modules\node-hid\build\HID.vcxproj]
..\src\HID.cc(488): warning C4244: 'argument': conversion from 'wchar_t' to 'char', possible loss of data [C:\Projetos Grails\EVAPress\node_modules\node-hid\build\HID.vcxproj]
     Creating library C:\Projetos Grails\EVAPress\node_modules\node-hid\build\Release\HID.lib and object C:\Projetos Grails\EVAPress\node_modules\node-hid\build\Release\HID.exp
  Generating code
  Finished generating code
  HID.vcxproj -> C:\Projetos Grails\EVAPress\node_modules\node-hid\build\Release\\HID.node
gyp info ok

titous01, 2017-09-01:
Just for info, in Ubuntu is wotking great even with cllbacks from my HID. This is a proprietary HID and is working perfectly on Ubuntu. Even with older versions of nw is working too (0.14) and node 6.

todbot, 2017-09-01:
Hmm, looks like HID.node (the native Node shared library for node-hid) got build okay.
But nw.js is not printing out any useful messages about what's going wrong?

titous01, 2017-09-01:
No, and when I try to use require("node-hid") it crashes. I Already try in another machine and with node an wn 32 bits with no luck.

todbot, 2017-09-01:
That's really frustrating.  I would ask someone who knows NW.js on Windows if there's a way to enable more verbose logging. It sounds like you're doing everything you're supposed to to rebuild node-hid for NW.js.  Maybe open up an issue on nw.js github? Or is there a forum?

titous01, 2017-09-01:
Hum, I try to find something like that. I see that you use others native modules inside your module. Which one needs rebuild too?

todbot, 2017-09-01:
The only native code node-hid uses is a C library called 'hidapi' that does the actual USB HID functionality. It is linked statically (no shared library) into the HID.node shared library.

titous01, 2017-09-01:
Can you read the dmp file that nw generate? I ddin´t find a program to read this kind of file, but it is being generated when the crashes occours. If you can read I can send to you.

todbot, 2017-09-01:
Yes at the very end of the dump, it says:
```
C:\Projetos Grails\EVAPress\node_modules\node-hid\build\Release\HID.node
gyp info ok
```
which means everything compiled correctly and the "HID.node" file was generated.

titous01, 2017-09-01:
I´m sorry not this dump. There is a file that is generated everytime nw crashes but is in dmp format. I cannot read this kind of file here. I think in there is a memory dump when the crash occour.

titous01, 2017-09-01:
I found that the error is in this line of nodehid.js:
var binding = require('bindings')('HID.node');
Do you think the problem may be in bindings?

titous01, 2017-09-01:
I just made node-hid and java nodes works in version 0.14.7 of nw.js. I´m looking forward to make these works in newer versions as well.

todbot, 2017-09-02:
Hey that's great news.  What did you have to do to get it working?

titous01, 2017-09-10:
Hello, in version 0.14.7 of nw.js everything works fine. This version of nw.js uses node 6. When I try to use newer versions of nw.js that uses node 8 and try to initiate node-hid(require ("node-hid")) it crashes imediately. Do you if anyone is using node-hid with nw.js that comes with node 8?
Regards,
Werther

ZackCoddington, 2017-09-14:
I am running into the same issue myself.  The last version of nw.js I was able to use node-hid with was version 0.22.3.  

With 0.25.0 it builds correctly, but when I try to utilize node-hid the program just quickly closes with an exit code of 3.

titous01, 2017-09-14:
Hi Zack ,
I didn´t manage to use it in 0.22.3, only in 0.14.7. Did you use in Windows?
Regards,
Werther

ZackCoddington, 2017-09-19:
Yes I am running in Windows.

porsager, 2017-11-27:
This appeared to be an issue with node::AtExit and should be fixed in the upcoming (v 0.26.7 of nwjs - https://github.com/nwjs/nw.js/issues/6047).
I've tested it with the nightly build and it seems to work fine.

todbot, 2017-11-27:
Hi @porsager.  Interesting!  Thank you for that find. I can't wait for others to try this out when it gets released.