node-hid/node-hid #223: Is it possible to bundle this module using webpack?

viorelAlexandrescu, 2017-08-25:
Hello there!

I have tried to bundle this module using webpack, but for some reason, it just can't make it work.

During the bundling, webpack warns me about the next issues.

[image]

These being just warnings, I decided to test the final bundle. This is the result

[image]

If someone has any idea, any help will be much appreciated!

todbot, 2017-08-25:
Hi!  I have used webpack with `node-hid` in https://github.com/todbot/Blink1Control2
I admit to knowing very little about webpack and found I had to add this to my [webpack.config.js](https://github.com/todbot/Blink1Control2/blob/master/webpack.config.js):
```
  externals: {
    "node-hid": 'commonjs node-hid'
  }
```

viorelAlexandrescu, 2017-08-28:
Thank you! That fixed it! Much abliged :D

todbot, 2017-08-28:
Awesome. Let me know if you have any further questions. I'll close this issue for now.

closed 2017-08-28 by todbot

mhuggins, 2018-03-24:
@todbot Thank you SO much for posting that!  I wasted hours trying to figure out how to get this fixed, and you saved the day. 😄