node-hid/node-hid #224: Unable to listen mouse - Windows 10

tonicfx, 2017-09-05:
Hello,

for a project, i would like to listen a USB Weight Balance. I don't have the hardware yet, so I just would like try this library, and listening my mouse USB on Port. I just make simple code : 

My configuration : Windows 10 / Node 6.11.2 / Node HID 0.5.7.

`var HID = require("node-hid");

var VENDOR_ID = 1133; // VID OF MY MOUSE
var PRODUCT_ID = 50479; // PID OF MY MOUSE
var deviceMouse = null;

try{
	console.log("create HID communication");	
	deviceMouse = new HID.HID(VENDOR_ID,PRODUCT_ID);
	deviceMouse.on('error', function(error){
		console.log("error : "+error);
	});;	
	deviceMouse.on('data', function(data){
		console.log("data : "+data);
	});;
}catch(error){
	console.log("catch erorr "+error);
}
`
The creation of HID seems OK, but any data is logged when I move my mouse.

I have listed devices : 
`
[  
   {  
      "vendorId":1133,
      "productId":50479,
      "path":"\\\\?\\hid#vid_046d&pid_c52f&mi_01&col01#8&1f5ee9b7&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}",
      "manufacturer":"Logitech",
      "product":"USB Receiver",
      "release":12288,
      "interface":1,
      "usagePage":12,
      "usage":1
   },
   {  
      "vendorId":1133,
      "productId":50479,
      "path":"\\\\?\\hid#vid_046d&pid_c52f&mi_00#8&372275fc&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}",
      "manufacturer":"Logitech",
      "product":"USB Receiver",
      "release":12288,
      "interface":0,
      "usagePage":1,
      "usage":2
   },
   {  
      "vendorId":1133,
      "productId":50479,
      "path":"\\\\?\\hid#vid_046d&pid_c52f&mi_01&col02#8&1f5ee9b7&0&0001#{4d1e55b2-f16f-11cf-88cb-001111000030}",
      "manufacturer":"Logitech",
      "product":"USB Receiver",
      "release":12288,
      "interface":1,
      "usagePage":65280,
      "usage":1
   },
   {  
      "vendorId":1133,
      "productId":50479,
      "path":"\\\\?\\hid#vid_046d&pid_c52f&mi_01&col03#8&1f5ee9b7&0&0002#{4d1e55b2-f16f-11cf-88cb-001111000030}",
      "manufacturer":"Logitech",
      "product":"USB Receiver",
      "release":12288,
      "interface":1,
      "usagePage":65280,
      "usage":2
   }
]
`
I have tested with HID.HID(path), but notion work too.

I have tested with à USB Listener Software, and I can see hexa data in the interface when I move my mouse.

Thanks for your help.

ert78gb, 2017-09-06:
what happening when you try the
`deviceMouse.read(function(err, data){console.log({err,data})})`

tonicfx, 2017-09-06:
I try with your code : 
```

var HID = require("node-hid");

var VENDOR_ID = 1133; // VID OF MY MOUSE
var PRODUCT_ID = 50479; // PID OF MY MOUSE

var deviceMouse = null;

var devices = HID.devices();

var str = JSON.stringify(devices);

//console.info(str);
try{
	deviceMouse = new HID.HID(VENDOR_ID,PRODUCT_ID);

	/*deviceMouse.on('error', function(error){
		console.log("error : "+error);
	});;
	
	deviceMouse.on('data', function(data){
		console.log("data : "+data);
	});;*/
	
	deviceMouse.read(function(err, data){
		console.info("read");
		console.log({err,data});
	});

}catch(error)
{
	console.log("catch erorr "+error);
}
```
But nothing is display. :(

todbot, 2017-09-07:
Hi,
It looks like that Logitech USB Receiver is a compound USB HID device, acting as multiple devices. (e.g. mouse, keyboard, media keys, etc).  These are what the "usage" and "usagePage" parameters are for.  (you can read all about usage tables here: http://www.usb.org/developers/hidpage/Hut1_12v2.pdf with an easier to read table here: http://www.freebsddiary.org/APC/usb_hid_usages.php )

For your particular case, you should open a different path, I think you want to open `devices[1]` , which has `usagePage==1` ("Generic Desktop") and `usage==2` ("Mouse")

tonicfx, 2017-09-07:
Hello, thanks for your help. 

I have implemented the logic, integrated the test with the "usage" and "usagePage", and use path to connect to the device : 
```
var HID = require("node-hid");
var VENDOR_ID = 1133; // VID OF MY MOUSE
var PRODUCT_ID = 50479; // PID OF MY MOUSE

var deviceMouse = null;
var devices = HID.devices();
var str = JSON.stringify(devices);
var deviceFound = false;

devices.forEach(function(device, index, records) {

    deviceFound = (device.vendorId == VENDOR_ID && device.usagePage === 1 && device.usage === 2);
	
	if (deviceFound && deviceMouse == null) {
		
		deviceMouse = new HID.HID(device.path);
		deviceMouse.on('error', function(error){
			console.log("error : "+error);
		});;
		deviceMouse.on('data', function(data){
			console.log("data : "+data);
		});;
	}
});
```
But I have this error 
```

C:\Users\Anthony\Desktop\node_usb_test>node test_usb_3.js
C:\Users\Anthony\Desktop\node_usb_test\node_modules\node-hid\nodehid.js:24
    this._raw = new (Function.prototype.bind.apply(binding.HID,
                ^

Error: cannot open device with path \\?\hid#vid_046d&pid_c52f&mi_00#8&372275fc&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}
    at Error (native)
    at new HID (C:\Users\Anthony\Desktop\node_usb_test\node_modules\node-hid\nodehid.js:24:17)
    at C:\Users\Anthony\Desktop\node_usb_test\test_usb_3.js:20:17
    at Array.forEach (native)
    at Object.<anonymous> (C:\Users\Anthony\Desktop\node_usb_test\test_usb_3.js:14:9)
    at Module._compile (module.js:570:32)
    at Object.Module._extensions..js (module.js:579:10)
    at Module.load (module.js:487:32)
    at tryModuleLoad (module.js:446:12)
    at Function.Module._load (module.js:438:3)
```

todbot, 2017-09-07:
Oh I did not notice you were on Windows.  You cannot open mice or keyboards on Windows. The OS will not allow it.

tonicfx, 2017-09-07:
Oh thanks !
So, I plugged à PS4 joystick, install driver, and seems to work :)

```
data : �|�    a   ������            �    �   �    �   �    �
data : �|� $  P����������            �    �   �    �   �    �
data : �|� (  3 ��  ����            �    �   �    �   �    �
data : �|� ,   �� ������            �    �   �    �   �    �
data : �|� 0  ����������            �    �   �    �   �    �
data : �|� 4  �  ��  ����            �    �   �    �   �    �
```
how to translate this in ascii or comprehensive hex string ?

todbot, 2017-09-07:
I think if you `JSON.stringify(data)` you'll get something you can at least visually inspect

tonicfx, 2017-09-08:
Thanks a lot, its working : 

`{"type":"Buffer","data":[1,129,124,129,130,8,0,128,0,0,140,181,248,0,0,253,255,254,255,169,255,191,31,241,6,0,0,0,0,0,27,0,0,0,0,128,0,0,0,128,0,0,0,0,128,0,0,0,128,0,0,0,0,128,0,0,0,128,0,0,0,0,128,0]}`

I have a last question, the weight balance I want to plug is based on port RS232, I imagine to use a RS232 to USB adaptor to plug on USB Port (computer not have RS232 port). Do you think I can read and write data from Weight balance with this adaptor, based on this HID Library ?

todbot, 2017-09-08:
For serial you will need a USB-to-serial adapter (these are pretty cheap).  If you need proper RS232 (using a DB9 or DB25 connector) then be sure to search for "USB RS232" on Amazon or similar. If it's 5V serial (aka "TTL serial" ), then search for "USB TTL serial". And then you would use `node-serialport` instead of `node-hid`.

tonicfx, 2017-09-08:
Thanks for your precious help :)

I will be back to tell you if it works.

tonicfx, 2017-12-07:
Hello, I'm back with my serial adpater, RS232 to USB :) I have tested the cable, and my computer recognize the cable as "usb serial converter". So the HID library not recognize the device (HID.devices() doesn't show the device), because I think it's not a HID USB specification.
Is it possible to listen / write a non HID USB Device ?

todbot, 2017-12-07:
Hi @tonicfx,
Yes, it is possible to listen / write to a USB-to-serial adapter. 
You need to do two things:
- Install the OS-specific device driver for your USB-to-serial adapter 
- Use `node-serialport` to access the device as a serial port

Closing this issue because it's not `node-hid` relevant.

closed 2017-12-07 by todbot

tonicfx, 2017-12-07:
Oh thanks todbot, I tested that !