node-hid/node-hid #226: Do not execute code on `require`.

MicahZoltu, 2017-09-26:
I have a project that has a transitive dependency on node-hid.  Node-hid is used in one particular code path in my application yet I can't run _any_ of my tests in Docker or Travis because node-hid is throwing an error when I do `require('node-hid')`.  The specifics of the error are not the topic of this issue (that is a separate problem that has been brought up a couple of times in other issues).  This issue is because `node-hid` should not be executing code during a `require` outside of what is required to export some variables.

In my project my tests do not exercise any node-hid code path (intentionally) yet as soon as I require a library that transitively requires `node-hid` I am unable to run _any_ of my tests in Docker or in CI.

Please change the pattern to have an `init` function or similar that the user must call if they want to initialize node-hid rather than doing initialization as a side effect of `require`.

todbot, 2017-09-26:
What is the error that is being thrown? Perhaps we can find a solution to that.

`node-hid` currently must execute code when loaded to ensure its underlying native libraries can be found. It seems like it would break more systems than fix if that were to change.

MicahZoltu, 2017-09-26:
I do recognize that a change to make initialization happen on a function call rather than on require would likely be a sem-ver major.  At the moment `node-hid` can't be anywhere in the dependency tree for any project that will be run/installed on a system without libusb (at the bare minimum) even if the features provided by `node-hid` are unused by the top level application.

As for my specific problem, I'll open a separate issue for it but I would rather keep troubleshooting of it out of this issue as this issue is meant to address the more fundamental problem of having a nodejs library that has a hard dependency on OS features/libraries at require time.

MicahZoltu, 2017-09-26:
Note: One of the problems I ran into was that I didn't have libusb.  This is an example of a problem that can't really be fixed by `node-hid`, but would be mitigated by not depending on libusb at require time but instead by depending on it at usage time.  That way if there is some code path that I never exercise 
deep in my application's dependency hierarchy I don't need to install libusb just so that my app starts.

todbot, 2017-09-26:
I'm looking into it now. I think there may be a way to do it without changing the historic API much. 

But wouldn't the more appropriate approach to stub out the upstream package that depends on `node-hid`, since it is assuming `node-hid` (and thus libusb if Linux) will work?

MicahZoltu, 2017-09-26:
I effectively do stub out the whole library via dependency injection.  However, in order to provide a good experience for my users I have some factories for creating the appropriate objects so the user doesn't need to instantiate those libraries themselves (or know/understand how they work) for the common case (which is that they want to actually connect to a USB device).  You can see those factories here:
https://github.com/ethereumjs/ethereumjs-ledger/blob/master/source/connection-factories.ts

In order to reference the factories, I need to reference the file that has the factories, which then internally references `node-hid`.  A situation where one might do this (I don't believe I do at the moment) is when you want to have the real library as a default that can be overridden via dependency injection.  In this case I might do something like `const thing = (typeof factory === 'undefined') ? factory() : defaultFactory();`.  In this case, I would have a hard dependency on the underlying library even though the tests will never hit that code path because they will always supply an alternative factory.

The situation is further complicated by the fact that the TypeScript type definitions for the library I am using are tied to the library, which means in order to get the static analyzer to work I need to reference the `ledgerco` library (which then brings in `node-hid`)..

MicahZoltu, 2017-09-26:
🤔 about possible solutions, one option could be a lazy initialization where the library will go through the initialization code on first use/access.  This could allow the public surface area to remain with no explicit `initialize` call (meaning no breaking change) while also not executing initialization during `require`.  Depending on how large the public surface area of `node-hid` is, this could be a large undertaking though.

todbot, 2017-09-27:
Hi @MicahZoltu, this is exactly what I was thinking.  I just checked in a change to master that implements lazy-loading of the C++ binding.  Is it easy for you to try out?

MicahZoltu, 2017-09-27:
Master appears to resolve the issue of not having libusb installed.  :)  I assume this also means it resolves the issue of running in an environment without access to USB devices (like Travis/Docker) though testing those before a version on NPM is a bit more tricky.

todbot, 2017-09-27:
Excellent.  Let me do some more tests and I'll push a new version in the next few days.  Thanks for helping out!

MicahZoltu, 2017-10-23:
I'm curious if this ever made it to NPM?  If so, which version is it in?

todbot, 2017-10-23:
Hi, not on NPM yet. Some personal issues came up that's limiting my github time.
I'd bump the version to 0.6.0 for this change, so when you see 0.6.0, that'll be this change.
Thanks for your patience.

bartekn, 2017-12-12:
Hey, any chances for `0.6.0` release any time soon? :)

todbot, 2017-12-30:
Hi, I'm working on publishing `node-hid@0.6.0` now.  It is published now and should be good to go. Let me know if you have any problems. 

Apologies for the very lengthy delay and thanks for your patience.

MicahZoltu, 2017-12-31:
For anyone coming along after me, 0.6.0 now allows me to successfully run unit tests in node on systems that don't have USB devices.  I did have to teach Travis/Docker how to install libusb though:
Travis:
```
addons:
  apt:
    packages:
      - libusb-1.0-0-dev
```
Docker:
```
FROM node:8
RUN apt-get update \
	&& apt-get install --yes libusb-1.0-0-dev
```

Thanks a lot @todbot!

closed 2017-12-31 by MicahZoltu