node-hid/node-hid #228: I cannot read proper barcode data.

tsadak, 2017-10-24:
Hello

I cannot tarnsform raw data read by node-hid to what it should really read. ı have tried many buffoer.to String conversions to no avail. These are some examples of what it should read and what it reads instead:
What node-hid should read: 4*9
What it reads instead:
<Buffer 00 00 21 00 00 00 00 00>
<Buffer 00 00 00 00 00 00 00 00>
<Buffer 02 00 00 00 00 00 00 00>
<Buffer 02 00 25 00 00 00 00 00>
<Buffer 02 00 00 00 00 00 00 00>
<Buffer 00 00 00 00 00 00 00 00>
<Buffer 00 00 26 00 00 00 00 00>
<Buffer 00 00 00 00 00 00 00 00>
<Buffer 00 00 28 00 00 00 00 00>
<Buffer 00 00 00 00 00 00 00 00>

What node-hid should read: 4*8
What it reads instead:
<Buffer 00 00 21 00 00 00 00 00>
<Buffer 00 00 00 00 00 00 00 00>
<Buffer 02 00 00 00 00 00 00 00>
<Buffer 02 00 25 00 00 00 00 00>
<Buffer 02 00 00 00 00 00 00 00>
<Buffer 00 00 00 00 00 00 00 00>
<Buffer 00 00 25 00 00 00 00 00>
<Buffer 00 00 00 00 00 00 00 00>
<Buffer 00 00 28 00 00 00 00 00>
<Buffer 00 00 00 00 00 00 00 00>

What node-hid should read: 4*1
What it reads instead:
<Buffer 00 00 21 00 00 00 00 00>
<Buffer 00 00 00 00 00 00 00 00>
<Buffer 02 00 00 00 00 00 00 00>
<Buffer 02 00 25 00 00 00 00 00>
<Buffer 02 00 00 00 00 00 00 00>
<Buffer 00 00 00 00 00 00 00 00>
<Buffer 00 00 1e 00 00 00 00 00>
<Buffer 00 00 00 00 00 00 00 00>
<Buffer 00 00 28 00 00 00 00 00>
<Buffer 00 00 00 00 00 00 00 00>

What node-hid should read: 1*01
What it reads instead:
<Buffer 00 00 1e 00 00 00 00 00>
<Buffer 00 00 00 00 00 00 00 00>
<Buffer 02 00 00 00 00 00 00 00>
<Buffer 02 00 25 00 00 00 00 00>
<Buffer 02 00 00 00 00 00 00 00>
<Buffer 00 00 00 00 00 00 00 00>
<Buffer 00 00 27 00 00 00 00 00>
<Buffer 00 00 00 00 00 00 00 00>
<Buffer 00 00 1e 00 00 00 00 00>
<Buffer 00 00 00 00 00 00 00 00>
<Buffer 00 00 28 00 00 00 00 00>
<Buffer 00 00 00 00 00 00 00 00>

What node-hid should read: 1*04
What it reads instead:
<Buffer 00 00 1e 00 00 00 00 00>
<Buffer 00 00 00 00 00 00 00 00>
<Buffer 02 00 00 00 00 00 00 00>
<Buffer 02 00 25 00 00 00 00 00>
<Buffer 02 00 00 00 00 00 00 00>
<Buffer 00 00 00 00 00 00 00 00>
<Buffer 00 00 27 00 00 00 00 00>
<Buffer 00 00 00 00 00 00 00 00>
<Buffer 00 00 20 00 00 00 00 00>
<Buffer 00 00 00 00 00 00 00 00>
<Buffer 00 00 28 00 00 00 00 00>
<Buffer 00 00 00 00 00 00 00 00>

What node-hid should read: 3*BAKIM
What it reads instead:
<Buffer 00 00 20 00 00 00 00 00>
<Buffer 00 00 00 00 00 00 00 00>
<Buffer 02 00 00 00 00 00 00 00>
<Buffer 02 00 25 00 00 00 00 00>
<Buffer 02 00 00 00 00 00 00 00>
<Buffer 02 00 05 00 00 00 00 00>
<Buffer 02 00 00 00 00 00 00 00>
<Buffer 02 00 04 00 00 00 00 00>
<Buffer 02 00 00 00 00 00 00 00>
<Buffer 02 00 0e 00 00 00 00 00>
<Buffer 02 00 00 00 00 00 00 00>
<Buffer 02 00 0c 00 00 00 00 00>
<Buffer 02 00 00 00 00 00 00 00>
<Buffer 02 00 10 00 00 00 00 00>
<Buffer 02 00 00 00 00 00 00 00>
<Buffer 00 00 00 00 00 00 00 00>
<Buffer 00 00 28 00 00 00 00 00>
<Buffer 00 00 00 00 00 00 00 00>

What node-hid should read: 3*SANZIMAN
What it reads instead:
<Buffer 00 00 20 00 00 00 00 00>
<Buffer 00 00 00 00 00 00 00 00>
<Buffer 02 00 00 00 00 00 00 00>
<Buffer 02 00 25 00 00 00 00 00>
<Buffer 02 00 00 00 00 00 00 00>
<Buffer 02 00 16 00 00 00 00 00>
<Buffer 02 00 00 00 00 00 00 00>
<Buffer 02 00 04 00 00 00 00 00>
<Buffer 02 00 00 00 00 00 00 00>
<Buffer 02 00 11 00 00 00 00 00>
<Buffer 02 00 00 00 00 00 00 00>
<Buffer 02 00 1d 00 00 00 00 00>
<Buffer 02 00 00 00 00 00 00 00>
<Buffer 02 00 0c 00 00 00 00 00>
<Buffer 02 00 00 00 00 00 00 00>
<Buffer 02 00 10 00 00 00 00 00>
<Buffer 02 00 00 00 00 00 00 00>
<Buffer 02 00 04 00 00 00 00 00>
<Buffer 02 00 00 00 00 00 00 00>
<Buffer 02 00 11 00 00 00 00 00>
<Buffer 02 00 00 00 00 00 00 00>
<Buffer 00 00 00 00 00 00 00 00>
<Buffer 00 00 28 00 00 00 00 00>
<Buffer 00 00 00 00 00 00 00 00>

How can i convert these buffers to proper text?

todbot, 2017-10-24:
Hi @tsadak, I think you can do the standard Buffer to string conversions, i.e. do `buf.toString()` or `JSON.stringify(buf)`

tsadak, 2017-10-26:
It’s no use. What returns is:

As data and Cumulative data, with ‘---‘ in between:

--- %'‑%‑&''#&$!$(%

--- %'‑%‑&''#&$!$(%

' --- %'‑%‑&''#&$!$(%'

--- %'‑%‑&''#&$!$(%'

‑ --- %'‑%‑&''#&$!$(%'‑

--- %'‑%‑&''#&$!$(%'‑

% --- %'‑%‑&''#&$!$(%'‑%

--- %'‑%‑&''#&$!$(%'‑%

‑ --- %'‑%‑&''#&$!$(%'‑%‑

--- %'‑%‑&''#&$!$(%'‑%‑

& --- %'‑%‑&''#&$!$(%'‑%‑&

--- %'‑%‑&''#&$!$(%'‑%‑&

' --- %'‑%‑&''#&$!$(%'‑%‑&'

--- %'‑%‑&''#&$!$(%'‑%‑&'

' --- %'‑%‑&''#&$!$(%'‑%‑&''

--- %'‑%‑&''#&$!$(%'‑%‑&''

# --- %'‑%‑&''#&$!$(%'‑%‑&''#

--- %'‑%‑&''#&$!$(%'‑%‑&''#

& --- %'‑%‑&''#&$!$(%'‑%‑&''#&

--- %'‑%‑&''#&$!$(%'‑%‑&''#&

$ --- %'‑%‑&''#&$!$(%'‑%‑&''#&$

--- %'‑%‑&''#&$!$(%'‑%‑&''#&$

! --- %'‑%‑&''#&$!$(%'‑%‑&''#&$!

--- %'‑%‑&''#&$!$(%'‑%‑&''#&$!

$ --- %'‑%‑&''#&$!$(%'‑%‑&''#&$!$

--- %'‑%‑&''#&$!$(%'‑%‑&''#&$!$

( --- %'‑%‑&''#&$!$(%'‑%‑&''#&$!$(

--- %'‑%‑&''#&$!$(%'‑%‑&''#&$!$(

dazhbog, 2017-11-05:
Hi @tsadak, you will need to convert the incoming data using HID usage tables. 

See here http://www.usb.org/developers/hidpage/Hut1_12v2.pdf

For example you gave this below

> What node-hid should read: 4*1
> What it reads instead:
> <Buffer 00 00 21 00 00 00 00 00>
> <Buffer 00 00 00 00 00 00 00 00>
> <Buffer 02 00 00 00 00 00 00 00>
> <Buffer 02 00 25 00 00 00 00 00>
> <Buffer 02 [...]

Looking at the third field in each line, you can see that the data is actually there. (I used the HID keyboard table. See pdf.)

- 21 -> 4
- 25 -> * (shift was pressed)
- 1e -> 1
- 28 -> \n

Hope this helps.

usernametaken1, 2018-07-29:
Just in case anybody else is trying to figure this out, this is what worked for me in the end. 
Also not sure its the best way.

This wouldn't work for different length barcodes. In my case im reading a usb hid nfc reader (14 characters lenght)

the fourth octet holds the character
<Buffer 02 00 00 **27** 00 00 00 00 00>

So i make a variable for the result a counter element and a key map
```
var nfcBuffs = ''
var buffsCount = 1
var keymap = {'04':'A','05':'B','06':'C','07':'D','08':'E','09':'F','0a':'G','0b':'H','0c':'I','0d':'J','0e':'K','0f':'L','10':'M','11':'N','12':'O','13':'P','14':'Q','15':'R','16':'S','17':'T','18':'U','19':'V','1a':'W','1b':'X','1c':'Y','1d':'Z','1e':'1','1f':'2','20':'3','21':'4','22':'5','23':'6','24':'7','25':'8','26':'9','27':'0','00':''}
```

then the data
```
nfcReader.on("data", function(data) {

    var nfcBuf = Buffer.from([data[3]]);
    nfcBuffs += keymap[nfcBuf.toString('hex')]

    if (buffsCount == 28) {
        nfcDecodeBuffer()
    }

    buffsCount+=1
    
})
```
then the function to do something with the result and reset counter/buffer
```
function nfcDecodedBuffer() {

    console.log(nfcBuffs)
    nfcBuffs = []
    buffsCount = 0

}
```

imyelo, 2019-01-08:
FYI, the HID Usage Tables Document (`Hut1_12v2.pdf` mentioned by @dazhbog ) is currently on:

https://www.usb.org/document-library/hid-usage-tables-112

graze1234, 2019-01-27:
thanks @usernametaken1 - with a bit of tweaking that worked for an RFID reader I was trying to get working on a Pi.

imyelo, 2019-01-28:
Hey guys. I wrote [a script](https://github.com/imyelo/hid-scanner#keymap) that automatically generates key mapping tables from [Android documents](https://source.android.com/devices/input/keyboard-devices.html#hid-keyboard-and-keypad-page-0x07). It might help you with this same issue, I think. 

A generated mapping table is here:
- https://github.com/imyelo/hid-scanner/blob/master/lib/keymap.raw.json

In addition, you may also be interested in this library [hid-scanner](https://github.com/imyelo/hid-scanner), which is based on `node-hide` but gives you the mapped key information.

closed 2019-07-25 by todbot