node-hid/node-hid #229: Occasional "Error in `node': double free or corruption (fasttop)"

mondalaci, 2017-10-29:
From time to time, node-hid crashes node. Any way to resolve this?

```
*** Error in `node': double free or corruption (fasttop): 0x00007f8afc000a50 ***
======= Backtrace: =========
/lib/x86_64-linux-gnu/libc.so.6(+0x777e5)[0x7f8b11b737e5]
/lib/x86_64-linux-gnu/libc.so.6(+0x8037a)[0x7f8b11b7c37a]
/lib/x86_64-linux-gnu/libc.so.6(cfree+0x4c)[0x7f8b11b8053c]
/lib/x86_64-linux-gnu/libusb-1.0.so.0(+0x8a6f)[0x7f8b0f6d3a6f]
/lib/x86_64-linux-gnu/libusb-1.0.so.0(libusb_handle_events_timeout_completed+0xd3)[0x7f8b0f6d4b53]
/lib/x86_64-linux-gnu/libusb-1.0.so.0(libusb_handle_events+0x2f)[0x7f8b0f6d4c3f]
/home/laci/projects/uhk/firmware/lib/agent/packages/usb/node_modules/node-hid/build/Release/HID.node(+0xe45e)[0x7f8b0f8f145e]
/lib/x86_64-linux-gnu/libpthread.so.0(+0x76ba)[0x7f8b11ecd6ba]
/lib/x86_64-linux-gnu/libc.so.6(clone+0x6d)[0x7f8b11c033dd]
======= Memory map: ========
00400000-01ff4000 r-xp 00000000 fd:01 192591                             /home/laci/.nvm/versions/node/v8.3.0/bin/node
021f3000-0220d000 rw-p 01bf3000 fd:01 192591                             /home/laci/.nvm/versions/node/v8.3.0/bin/node
0220d000-02225000 rw-p 00000000 00:00 0 
03fad000-041f8000 rw-p 00000000 00:00 0                                  [heap]
694d8d00000-694d8d80000 ---p 00000000 00:00 0 
807f3200000-807f3280000 rw-p 00000000 00:00 0 
9ac5773b000-9ac5773c000 ---p 00000000 00:00 0 
9ac5773c000-9ac57744000 rw-p 00000000 00:00 0 
9ac57744000-9ac57747000 ---p 00000000 00:00 0 
a4126280000-a4126300000 rw-p 00000000 00:00 0 
d1f3b2c7000-d1f3b480000 ---p 00000000 00:00 0 
d1f3b480000-d1f3b483000 rw-p 00000000 00:00 0 
d1f3b483000-d1f3b484000 ---p 00000000 00:00 0 
d1f3b484000-d1f3b4ff000 rwxp 00000000 00:00 0 
d1f3b4ff000-d1f3b500000 ---p 00000000 00:00 0 
d1f3b500000-d1f3b503000 rw-p 00000000 00:00 0 
d1f3b503000-d1f3b504000 ---p 00000000 00:00 0 
d1f3b504000-d1f3b57f000 rwxp 00000000 00:00 0 
d1f3b57f000-d1f3b580000 ---p 00000000 00:00 0 
d1f3b580000-d1f3b583000 rw-p 00000000 00:00 0 
d1f3b583000-d1f3b584000 ---p 00000000 00:00 0 
d1f3b584000-d1f3b5ff000 rwxp 00000000 00:00 0 
d1f3b5ff000-d1f5b2c7000 ---p 00000000 00:00 0 
e6670180000-e6670200000 rw-p 00000000 00:00 0 
151ba8c80000-151ba8d00000 rw-p 00000000 00:00 0 
1b56f1600000-1b56f1680000 rw-p 00000000 00:00 0 
1f4f51380000-1f4f51400000 ---p 00000000 00:00 0 
21f65b080000-21f65b100000 ---p 00000000 00:00 0 
2d4d31600000-2d4d31680000 ---p 00000000 00:00 0 
2d4db5000000-2d4db5080000 ---p 00000000 00:00 0 
39f28b600000-39f28b680000 ---p 00000000 00:00 0 
3cf775600000-3cf775680000 ---p 00000000 00:00 0 
3e0eed380000-3e0eed400000 ---p 00000000 00:00 0 
7f8af0000000-7f8af0021000 rw-p 00000000 00:00 0 
7f8af0021000-7f8af4000000 ---p 00000000 00:00 0 
7f8af8000000-7f8af8021000 rw-p 00000000 00:00 0 
7f8af8021000-7f8afc000000 ---p 00000000 00:00 0 
7f8afc000000-7f8afc021000 rw-p 00000000 00:00 0 
7f8afc021000-7f8b00000000 ---p 00000000 00:00 0 
7f8b00000000-7f8b00021000 rw-p 00000000 00:00 0 
7f8b00021000-7f8b04000000 ---p 00000000 00:00 0 
7f8b04000000-7f8b04021000 rw-p 00000000 00:00 0 
7f8b04021000-7f8b08000000 ---p 00000000 00:00 0 
7f8b08000000-7f8b08055000 rw-p 00000000 00:00 0 
7f8b08055000-7f8b0c000000 ---p 00000000 00:00 0 
7f8b0e4c5000-7f8b0e4c6000 ---p 00000000 00:00 0 
7f8b0e4c6000-7f8b0ecc6000 rw-p 00000000 00:00 0 
7f8b0ecc6000-7f8b0ecc9000 r-xp 00000000 fd:01 1451639                    /usr/lib/x86_64-linux-gnu/gconv/UTF-16.so
7f8b0ecc9000-7f8b0eec8000 ---p 00003000 fd:01 1451639                    /usr/lib/x86_64-linux-gnu/gconv/UTF-16.so
7f8b0eec8000-7f8b0eec9000 r--p 00002000 fd:01 1451639                    /usr/lib/x86_64-linux-gnu/gconv/UTF-16.so
7f8b0eec9000-7f8b0eeca000 rw-p 00003000 fd:01 1451639                    /usr/lib/x86_64-linux-gnu/gconv/UTF-16.so
7f8b0eeca000-7f8b0eecb000 ---p 00000000 00:00 0 
7f8b0eecb000-7f8b0f6cb000 rw-p 00000000 00:00 0 
7f8b0f6cb000-7f8b0f6e2000 r-xp 00000000 fd:01 787566                     /lib/x86_64-linux-gnu/libusb-1.0.so.0.1.0
7f8b0f6e2000-7f8b0f8e1000 ---p 00017000 fd:01 787566                     /lib/x86_64-linux-gnu/libusb-1.0.so.0.1.0
[... 37 lines cut ...]
7f8b1298b000-7f8b12b8a000 ---p 00007000 fd:01 787754                     /lib/x86_64-linux-gnu/librt-2.23.so
7f8b12b8a000-7f8b12b8b000 r--p 00006000 fd:01 787754                     /lib/x86_64-linux-gnu/librt-2.23.so
7f8b12b8b000-7f8b12b8c000 rw-p 00007000 fd:01 787754                     /lib/x86_64-linux-gnu/librt-2.23.so
7f8b12b8c000-7f8b12b8f000 r-xp 00000000 fd:01 787760                     /lib/x86_64-linux-gnu/libdl-2.23.so
7f8b12b8f000-7f8b12d8e000 ---p 00003000 fd:01 787760                     /lib/x86_64-linux-gnu/libdl-2.23.so
7f8b12d8e000-7f8b12d8f000 r--p 00002000 fd:01 787760                     /lib/x86_64-linux-gnu/libdl-2.23.so
7f8b12d8f000-7f8b12d90000 rw-p 00003000 fd:01 787760                     /lib/x86_64-linux-gnu/libdl-2.23.so
7f8b12d90000-7f8b12db6000 r-xp 00000000 fd:01 787738                     /lib/x86_64-linux-gnu/ld-2.23.so
7f8b12f7c000-7f8b12f83000 rw-p 00000000 00:00 0 
7f8b12f87000-7f8b12f88000 rw-p 00000000 00:00 0 
7f8b12f88000-7f8b12f8f000 r--s 00000000 fd:01 1452785                    /usr/lib/x86_64-linux-gnu/gconv/gconv-modules.cache
7f8b12f8f000-7f8b12fad000 r-xp 00000000 fd:01 815853                     /lib/x86_64-linux-gnu/libudev.so.1.6.4
7f8b12fad000-7f8b12fae000 r--p 0001d000 fd:01 815853                     /lib/x86_64-linux-gnu/libudev.so.1.6.4
7f8b12fae000-7f8b12faf000 rw-p 0001e000 fd:01 815853                     /lib/x86_64-linux-gnu/libudev.so.1.6.4
7f8b12faf000-7f8b12fb0000 ---p 00000000 00:00 0 
7f8b12fb0000-7f8b12fb5000 rw-p 00000000 00:00 0 
7f8b12fb5000-7f8b12fb6000 r--p 00025000 fd:01 787738                     /lib/x86_64-linux-gnu/ld-2.23.so
7f8b12fb6000-7f8b12fb7000 rw-p 00026000 fd:01 787738                     /lib/x86_64-linux-gnu/ld-2.23.so
7f8b12fb7000-7f8b12fb8000 rw-p 00000000 00:00 0 
7fff90558000-7fff90579000 rw-p 00000000 00:00 0                          [stack]
7fff905d6000-7fff905d8000 r--p 00000000 00:00 0                          [vvar]
7fff905d8000-7fff905da000 r-xp 00000000 00:00 0                          [vdso]
ffffffffff600000-ffffffffff601000 r-xp 00000000 00:00 0                  [vsyscall]
Aborted
```

todbot, 2017-10-29:
Hi @mondalaci,
Yes, that's a very bad error.
Can you create a small demo program that recreates the problem?

mondalaci, 2017-10-29:
@todbot I could reproduce the issue a couple of times using [this script](https://github.com/UltimateHackingKeyboard/agent/blob/eac9341c992c2d5a02722af9930fc8a05a282d04/packages/usb/send-kboot-command-to-slave.js) and [this one](https://github.com/UltimateHackingKeyboard/agent/blob/eac9341c992c2d5a02722af9930fc8a05a282d04/packages/usb/eeprom-stress-test.sh). Unfortunately, these scripts rely on our keyboard hardware.

The main problem is that I cannot reproduce it deterministically, and I doubt it's possible. This looks like a memory leak of some sort. Any ideas how to reproduce it deterministically?

todbot, 2017-10-29:
Hmmm.  A couple things you might want to try:

1. If you're using a prebuilt version of the node-hid native lib (most people are), you can build your own with `npm install node-hid  --build-from-source `  This is in case there's some slight oddity between the the node-hid CI's distro and your system.  I doubt this will have any effect, but it's good to check.

2. Can you change your code to use `device.read()` instead of `device.readSync()`?  The `readSync` method is a less-traveled code path and I can see it interacting badly with the callback way of doing reads .  (This is part of the code I'm not too familiar with, but I'm looking at it now)

jiteshsojitra, 2019-10-31:
We are facing this issue very frequently with node 12.13 now, everything was fine with 12.11.

docker:
      - image: circleci/node:13.0-browsers

```
*** Error in `node': double free or corruption (out): 0x00000000033c75e0 ***
======= Backtrace: =========
/lib/x86_64-linux-gnu/libc.so.6(+0x70bfb)[0x7ff7e2b8abfb]
/lib/x86_64-linux-gnu/libc.so.6(+0x76fc6)[0x7ff7e2b90fc6]
/lib/x86_64-linux-gnu/libc.so.6(+0x7780e)[0x7ff7e2b9180e]
node[0x15bb2fc]
node(asn1_item_embed_free+0x33b)[0x147dceb]
node(ASN1_item_free+0x17)[0x147de47]
node(OPENSSL_sk_pop_free+0x34)[0x1596714]
node[0x141f85c]
node(_ZN4node7TLSWrap10DestroySSLERKN2v820FunctionCallbackInfoINS1_5ValueEEE+0x83)[0xaf6c63]
node[0xba3c19]
node(_ZN2v88internal21Builtin_HandleApiCallEiPmPNS0_7IsolateE+0xb7)[0xba5a07]
node[0x13726d9]
======= Memory map: ========
00400000-027d9000 r-xp 00000000 00:4e 161852                             /usr/local/bin/node
029d8000-029db000 r--p 023d8000 00:4e 161852                             /usr/local/bin/node
029db000-029ec000 rw-p 023db000 00:4e 161852                             /usr/local/bin/node
029ec000-02a0b000 rw-p 00000000 00:00 0 
02e3c000-038f3000 rw-p 00000000 00:00 0                                  [heap]
4181c00000-4181c40000 rw-p 00000000 00:00 0 
4479800000-4479840000 rw-p 00000000 00:00 0 
52fb100000-52fb140000 rw-p 00000000 00:00 0 
7f0bd80000-7f0bdc0000 rw-p 00000000 00:00 0 
d060080000-d0600c0000 rw-p 00000000 00:00 0 
d75f640000-d75f680000 rw-p 00000000 00:00 0 
127a1640000-127a1680000 rw-p 00000000 00:00 0 
15ab7600000-15ab7640000 rw-p 00000000 00:00 0 
16456740000-16456780000 rw-p 00000000 00:00 0 
196ae180000-196ae1c0000 rw-p 00000000 00:00 0 
1c068cc0000-1c068d00000 rw-p 00000000 00:00 0 
1fe3c280000-1fe3c2c0000 rw-p 00000000 00:00 0 
```

todbot, 2019-10-31:
Hi,
1. Do you have a test case exhibiting the problem that I could try to replicate?
2. What is the code that is causing this issue?
3. What device is this?
4. What OS is this on?

Julusian, 2025-06-03:
The library has been largely rewritten since this was opened, if this is still a problem please open a new issue

closed 2025-06-03 by Julusian