node-hid/node-hid #230: Possible race condition on close

FrogTheFrog, 2017-11-04:
Hello,

like some other people, I sometimes get an error when closing a device. It's a very rare error on my system, however I managed to catch it and it was about access violation at 0x00000000. Basically, this means that something that has been freed is being used again (could be related to #229 and maybe #210).

After analyzing code I found a possible race condition with `HID::close` and `HID::recvAsync`. If `_hidHandle` is freed by `HID::close` just before `hid_read(hid->_hidHandle, buf, sizeof buf)` is called, we will try to access already freed memory. `hid_read` itself contains no `null` handle check whatsoever.

Given that `hid_read` is not thread safe, I believe that the only possible solution is an active pooling with `hid_read_timeout(dev, data, length, 0)` on a separate thread with some sort of timeout or thread sleep duration.

Using separate threads would also solve a problem with node's 3 remaining threads being locked up by `node-hid` when reading more than 3 devices that are not sending data.

todbot, 2017-11-04:
Hi @FrogTheFrog, your analysis sounds spot on.  Thank you.

Your proposed change sounds like the better one long-term but it's a pretty big change that I won't have time to implement and test any time soon.

As a short-term fix, what about if we put a null test on `_hidHandle` just before any `hid_read()`s?  It won't eliminate the race condition but could shrink the window of a null pointer chance

FrogTheFrog, 2017-11-04:
@todbot It should work. I'll try to work on a solution myself. Already got separate threads on my fork, just need to finish properly implement pooling and closing. Basically, I don't know what should be the interval for a `while` loop or, in other words, for how long should I put thread to sleep?

herrjemand, 2018-03-07:
Hey guys. I am getting electron crashing with error 3221225477(ACCESS_VIOLATION). I think it is related to the current issue. Any idea what I should do?

neppi, 2019-02-07:
Hey all,

i have a windows service using version 0.7.4 of node-hid
I did not debugged the c part of the node module. But we often have the case that the node service kills itself when calling **close()** on the HID.device. 

I get **no** exception at all. I just kills the whole node service **without any error**. Try catch does not help.

We bypass the issue by **not closing** the device. But thats not good i think.

Maybe this is related to the race condition mentioned by @FrogTheFrog 
Any suggestions?

nine-fox, 2019-04-19:
I got the same issue here. By calling close(), segment fault occurred. Just like @neppi said, I removed the close() call, and everything works fine.

My only concern is, will it cause the memory growing issue, if I am running it in a long running electron application? That is, lots of hid devices are allocated and used then unused, but never closed, due to plug/unplug usb devices several times. These unused devices are waiting for node.js free I guess. 

My current testing shows fine.

Newbrict, 2019-12-17:
I am also experiencing this issue...

todbot, 2019-12-17:
Hi, if someone can create a small Node program that exhibits this issue repeatedly on easily-available hardware (e.g. a Teensy or Trinket M0 running a RawHID implementation), we'll be able to diagnose it and find a solution.  Thanks!

FrogTheFrog, 2019-12-17:
@todbot The problem is that on windows this piece or code is being used:
```
void HID_API_EXPORT HID_API_CALL hid_close(hid_device *dev)
{
	if (!dev)
		return;
	CancelIo(dev->device_handle);
	free_hid_device(dev);
}
```
and CancelIo does not support multi-threading whereas CancelIoEx does.

todbot, 2019-12-17:
That is a `hidapi` piece of code, not `node-hid`.  I would recommend opening an issue there.

FrogTheFrog, 2019-12-17:
I forgot I had this issue open until today :smile: . Anyway, a quick summary - a PR in hidapi was refused, because it would bring min. Windows requirements up to Windows Vista I believe... Long live the bug :frowning_face:

todbot, 2019-12-17:
Which hidapi PR is this? With libusb taking over hidap in the last few monthsi it's actually getting improved fairly regularly.

FrogTheFrog, 2019-12-17:
I kinda mixed up and issue and PR. Anyway, the "refusing" (but not really) part is https://github.com/signal11/hidapi/issues/48 and the never merged PR is https://github.com/signal11/hidapi/pull/208.

todbot, 2019-12-17:
Yes, the original hidapi owner seemingly has abandoned the project. But a new fork is being updated regularly at [libusb/hidapi](https://github.com/libusb/hidapi)  If you have time, please re-open the issue there. (I'll put some effort toward it too but it really helps if other users come forward).  Thanks for your patience.

FrogTheFrog, 2019-12-17:
@todbot done https://github.com/libusb/hidapi/issues/133

todbot, 2019-12-18:
Hey @FrogTheFrog, if you have a repeatable test case for the issue you were originally seeing, can you try it out on latest `node-hid` repo?  I think I may have finally fixed the bug Windows users were having when using `device.on('data', ...)`