node-hid/node-hid #231: Is it possible to get data from a mouse without taking it over?

ptcher, 2017-11-07:
I am using node-hid to register multiple mice plugged into the same computer as different devices, so I can produce input events with the device id attached. My issue is that when I listen to a mouse with `device.on()`, I consume all its input, and I can no longer control the OS cursor. I can't figure out from the documentation whether there is any way around this. 

It have tried some different configurations where I periodically close and re-acquire the device, pause and resume reading, or try to read synchronously. None of those approaches result in the behavior that I want.

Am I out of luck if I want to listen to HID devices without monopolizing them, or am I missing something?

Cheers,
Philip Tchernavskij

ptcher, 2017-11-17:
I may not have described my issue clearly.
Can someone please tell me whether it is possible to read input data from a USB mouse with node-hid without preventing that mouse from being used regularly?

ert78gb, 2017-11-18:
Do you use windows? We have a similar problem in win, but everything works fine on Mac and Linux

ptcher, 2017-11-20:
Nope, I'm using Mac.  
I have managed to read from the mouse in two ways, `device.on('data', processData)`  
and by periodically doing this:
```
device.resume();  
device.read(processData);  
device.pause();
```
where device is created with `HID.HID(vid, pid)`.
In both cases, the OS cursor stops moving while the program is running.

ptcher, 2017-12-04:
Can anyone show me a working example of reading from a mouse without blocking it from communicating with OSX? 
According to @ert78gb, it is supposed to be possible, so I must have misunderstood how I'm supposed to use the library.

closed 2019-11-23 by todbot