node-hid/node-hid #232: Hi DLL link initialization routine failed with electron

ColacX, 2017-11-16:
Hi can I get some help with this problem? I encountered this error when using Electron. I followed the readme guide the best I could.

I am running on Windows 10 with Node v8.6.0 installed on the system.
npm install electron@1.6.0 --save-dev --save-exact
npm install electron-rebuild@1.6.0 --save-dev --save-exact 
npm install electron-prebuilt@1.4.13 --save-dev --save-exact
npm install node-hid@0.5.7 --save-dev --save-exact

```
//test.js
const HID = require('node-hid');
console.log(HID.devices());
```

electron-rebuild --force
electron ./test.js"

Am I missing a step? according to the readme this version of electron should work with node-hid.

todbot, 2017-11-16:
what error are you seeing? I don't see it in your issue above.

ColacX, 2017-11-17:
Sorry I forgot about that. I updated the issue. Thanks for taking interest.

todbot, 2017-11-17:
Perhaps not have `node-hid` as a devDependency but as a normal dependency?

ColacX, 2017-11-20:
That was an error in my description. I double checked it is saved as a normal dependency.

ColacX, 2017-11-28:
I found this guy on github who got a working project
https://github.com/todbot/electron-hid-toy

The magic seems to happen with "electron-builder": "^19.35.1",
npm install electron-builder@19.35.1 --save-dev-exact

and
"postinstall": "electron-builder install-app-deps"

todbot, 2019-07-25:
Fixed as of many versions ago. :)

closed 2019-07-25 by todbot