node-hid/node-hid #233: Electron 1.8.x crash

mkazlauskas, 2017-11-20:
It works fine with 1.7.x. When I try to `electron-rebuild` it with 1.8.x it will build without errors, but at runtime throw the following:

```electron.exe: c:\projects\electron\vendor\node\src\node.cc:4451: Assertion `(env) != (nullptr)' failed.```

ChaosRambo, 2017-11-27:
I have the same Issue but on Electron 1.7.9...
If I want to start Electron I get this Error Message:

/home/pathtoproject/electron/node_modules/electron/dist/electron[19378]: ../../vendor/node/src/node.cc:4451:void node::AtExit(node::Environment , void ()(void *), void *): Assertion `(env) != (nullptr)' failed.
1: node::Abort() [/home/pathtoproject/electron/node_modules/electron/dist/libnode.so]
2: 0x7f594375c2f4 [/home/pathtoproject/electron/node_modules/electron/dist/libnode.so]
3: 0x7f59437659bc [/home/pathtoproject/electron/node_modules/electron/dist/libnode.so]
4: HID::Initialize(v8::Localv8::Object) [/home/pathtoproject/electron/node_modules/node-hid/build/Release/HID.node]
5: 0x7f591c5725d9 [/home/pathtoproject/electron/node_modules/node-hid/build/Release/HID.node]
6: 0x7f594376330d [/home/pathtoproject/electron/node_modules/electron/dist/libnode.so]
7: 0x7f594306aaa0 [/home/pathtoproject/electron/node_modules/electron/dist/libnode.so]
8: 0x7f59434b4d08 [/home/pathtoproject/electron/node_modules/electron/dist/libnode.so]
9: 0x7f59434b427e [/home/pathtoproject/electron/node_modules/electron/dist/libnode.so]
10: 0x38ac5d4843fd

I use node-hid 0.5.7, with node 8.9.1 and I'm currently working on Ubuntu 16.04.

It would be perfect if someone could be help us.
Cheers,
ChaosRambo

todbot, 2017-11-28:
Hi I'm attempting to test this right now.
Do you have a small example Electron app that shows this behavior?

mkazlauskas, 2017-11-28:
Sorry, I do not have an example app. I think you just need to try to build for 1.8.x. (or maybe just Node 8.2.1). By the way, the support table in README doesn't include 1.8.x.

todbot, 2017-11-28:
Yeah. Electron 1.8 is still in beta so we don't officially support it yet.

todbot, 2017-12-01:
This appears to be a problem with Electron 1.8.
I've opened an issue for it here: https://github.com/electron/electron/issues/11299

closed 2019-01-31 by todbot