node-hid/node-hid #236: Opening a device more than once  is not working on (Linux) but works on (Windows platform)...

riodel27, 2017-11-29:
Basically we want to be able to open the same device more than once or Instantiate a "new HID.HID(path or [vid,pid]) on linux.  I am wondering if this issue could be somehow related to the the linux platform running on libusb drivers? and should it be "hidraw"?  or maybe because of the udev permissions in which I think I tried to address , but with no luck. 

I did some simulation both on windows and and linux.

======================================================
Windows Platform:

var HID = require('../');
console.log('devices:', HID.devices());

var hiddevice = new HID.HID(vid,pid)
hiddevice.on("data", data=>{
    let buffer = new Uint16Array(data)
    console.log("BUFFER: " + buffer);
})

var hiddevice1 = new HID.HID(vid,pid)
hiddevice1.on("data", data=>{
    let buffer = new Uint16Array(data)
    console.log("BUFFER new connection: " + buffer);
})

! It is working and will log both connections
BUFFER new connection: [ 255,255,255,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0...]
BUFFER:   [ 255,255,255,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,]
======================================================

-

======================================================
Linux Mint/Ubuntu

var HID = require('../');
console.log('devices:', HID.devices());

var hiddevice = new HID.HID('0001:0006:00')

hiddevice.on("data", data => {
 
 let buffer = new Uint16Array(data)
console.log("Buffer: " + buffer);

 var otherHidDeviceConnection = new HID.HID(pid,vid){
  otherHidDeviceConnection .on("data",data=>{

   let buffer = new Uint16Array(data){
    console.log("New Buffer Connection: " + buffer)

 }
}
})

! I changed the code structure from the above(windows platform) to make sure that the first open/connection will log something and that it works and that it is on the second connection/instance will give me the error.

Output: 

dev@dev:~/Desktop/dev/node-hid$ sudo node src/show-devices.js
devices: [ { vendorId: vid,
    productId: pid,
    path: '0001:0006:00',
    manufacturer: '',
    product: '',
    release: 256,
    interface: 0 },
  { vendorId: 33006,
    productId: 33,
    path: '0001:0002:00',
    manufacturer: 'VirtualBox',
    product: 'USB Tablet',
    release: 256,
    interface: 0 } ]
BUFFER: 0,1,9,255,255,255,255,255,255,1,0,1,0,0,0,0,0,27,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0
/home/dev/Desktop/dev/node-hid/nodehid.js:32
    this._raw = new (Function.prototype.bind.apply(binding.HID,
                ^

Error: cannot open device with vendor id 0x4d8 and product id 0x3f    at new HID (/home/dev/Desktop/dev/node-hid/nodehid.js:32:17)
    at HID.hiddevice.on.data (/home/dev/Desktop/dev/node-hid/src/show-devices.js:15:22)
    at emitOne (events.js:115:13)
    at HID.emit (events.js:210:7)
    at readFunc (/home/dev/Desktop/dev/node-hid/nodehid.js:103:22)
=====================================================

-Any Idea how I could make it work on Linux or is it even possible or maybe some limitation with the hidapi on non windows platform machine? Thx.

todbot, 2017-11-29:
You cannot open an already open device.

riodel27, 2017-11-30:
@todbot , right sorry about the formatting, I just copy and pasted it and weird because the formatting before I posted the issue was different. But anyways I am just wondering if that is the case that we cannot open an already open device, then how and why is it working on windows? or maybe I am just using the wrong term about "opening the same device more than once" ?  but Instead use multiple instance of  "new HID.HID(vid,pid)" ?

Basically I can do this on my code and will work on windows platform,

var hiddevice = new HID.HID(vid,pid)
var hiddevice1 = new HID.HID(vid,pid)
var hiddevice2 = new HID.HID(vid,pid)
var hiddevice3 = new HID.HID(vid,pid)
and so on....

but if I try to do that in Linux only one instance or one connection of that will work and the rest will  give me that "Error: cannot open device with vendor id 0x4d8 and product id 0x3f at new HID ".

Because basically what I wanted to happen is that I could do  " var hiddevice = new HID.HID(vid,pid) "  on **Electron Main process**. And I could do " var hiddevice = new HID.HID(vid,pid) " on **Renderer process** on Electron as well.  We wanted to do the hid.write() on renderer process and do the reading/streaming of the data on Electron main process, that way we have two processes, and we needed to separate the logic as there is a lot of recursion going on inside the **Renderer process** and we don't want it to block the data coming in from the device. Basically the concept that we wanted to achieve is like the multi threaded functionality of another programming language.  

I am wondering if this could be a limitation of hidapi on non windows platform, but I really don't know.

Thx..

herrjemand, 2018-03-07:
@riodel27 I just have a small device status manager

let openedDevices = {};

/**
 * Opens device defined by given deviceInfo
 * @param  {DeviceInfo} deviceInfo - Node HID device info object
 * @return {NodeHIDDevice}
 */
let openDevice = (deviceInfo) => {
    if (deviceInfo && deviceInfo.path) {
        if (!openedDevices[deviceInfo.path]) {
            let newDevice = new hid.HID(deviceInfo.path);
            openedDevices[deviceInfo.path] = newDevice;
            return newDevice;
        } else
            return openedDevices[deviceInfo.path];
    } else
        throw new Error('Given DeviceInfo is not a valid DeviceInfo object!');
}

/**
 * Closes device defined by given deviceInfo
 * @param  {DeviceInfo} deviceInfo - Node HID device info object
 */
let closeDevice = (deviceInfo) => {
    if (deviceInfo && deviceInfo.path) {
        if (openedDevices[deviceInfo.path]) {
            openedDevices[deviceInfo.path].close();
            openedDevices[deviceInfo.path] = undefined;
        }
    } else
        throw new Error('Given DeviceInfo is not a valid DeviceInfo object!');
}

riodel27, 2018-03-08:
@Ackermann Yuriy Thank you for this update. I will sure look in to it.

todbot, 2019-06-19:
Thanks for the help @herrjemand.  The re-open problem is an upstream `hidapi` issue and is very platform-dependent.

closed 2019-06-19 by todbot