node-hid/node-hid #238: Error: Could not locate the bindings file

LawKong, 2017-12-13:
sudo git clone https://github.com/node-hid/node-hid.git
cd node-hid                                        # must change into node-hid directory
sudo npm run prepublish                                 # get the needed hidapi submodule
sudo npm install --build-from-source                    # rebuilds the module with C code

I used the commands above to get start with logitech g29 but after keying in the last command: sudo node ./src/show-devices.js , error happens (as shown below).

/node-hid/node_modules/bindings/bindings.js:96
  throw err
  ^

Error: Could not locate the bindings file. Tried:
 → /node-hid/build/HID.node
 → /node-hid/build/Debug/HID.node
 → /node-hid/build/Release/HID.node
 → /node-hid/out/Debug/HID.node
 → /node-hid/Debug/HID.node
 → /node-hid/out/Release/HID.node
 → /node-hid/Release/HID.node
 → /node-hid/build/default/HID.node
 → /node-hid/compiled/8.9.3/linux/arm/HID.node
    at bindings (/node-hid/node_modules/bindings/bindings.js:93:9)
    at loadBinding (/node-hid/nodehid.js:9:38)
    at Object.showdevices [as devices] (/node-hid/nodehid.js:110:5)
    at Object.<anonymous> (/node-hid/src/show-devices.js:3:29)
    at Module._compile (module.js:635:30)
    at Object.Module._extensions..js (module.js:646:10)
    at Module.load (module.js:554:32)
    at tryModuleLoad (module.js:497:12)
    at Function.Module._load (module.js:489:3)
    at Function.Module.runMain (module.js:676:10)
ps: New github user here, please tell me if I did anything wrong or this issue is duplicated. Thank you in advance.

todbot, 2017-12-13:
Hi,

A few questions to help diagnose your problem:
1. What OS are you trying this on?
2. What version of Node are you using?
3. Why are you using 'sudo'?  Normally npm and Node does not like to be used with 'sudo'.

LawKong, 2017-12-13:
1. Raspbian Jessie Version 8 (flashed on Raspberry pi 3)
2. Nodejs version 8.9.3
3. I didn't follow the fixing npm permissions steps at (https://docs.npmjs.com/getting-started/fixing-npm-permissions) thus I used sudo for the commands.

todbot, 2017-12-30:
If you have a bad npm permissions (as you mention in (3)) then you'll see the sort of problem you're experiencing.  Fix that and try again.

LawKong, 2018-01-15:
I actually re-do from the beginning as I noticed I actually missed out many
of the steps. https://github.com/node-hid/node-hid#compiling-from-source,
which command (under  "All OSes" or "Linux") should I use? As I mentioned
I'm using raspbian jessie on raspberry pi 3.

todbot, 2018-01-15:
Just do the Linux part.  You can install node-gyp globally if you want but it's not necessary.

LawKong, 2018-01-18:
I finally finished setting up everything and it run nicely on ubuntu desktop but when I used this command "node src/test-bigredbutton.js" on raspberry pi, an error occurs.
/home/pi/node_modules/node-hid/nodehid.js:24
    this._raw = new (Function.prototype.bind.apply(binding.HID,
                ^

Error: cannot open device with vendor id 0x1d34 and product id 0xd
    at new HID (/home/pi/node_modules/node-hid/nodehid.js:24:17)
    at Object.<anonymous> (/home/pi/node_modules/node-hid/src/test-bigredbutton.js:16:14)
    at Module._compile (module.js:660:30)
    at Object.Module._extensions..js (module.js:671:10)
    at Module.load (module.js:573:32)
    at tryModuleLoad (module.js:513:12)
    at Function.Module._load (module.js:505:3)
    at Function.Module.runMain (module.js:701:10)
    at startup (bootstrap_node.js:193:16)
    at bootstrap_node.js:617:3
I also did "node src/show-devices.js" to check the devices, it doesn't seem correct.
devices: [ { vendorId: 1133,
    productId: 49743,
    path: '0001:000d:00',
    release: 35072,
    interface: 0 },
  { vendorId: 1133, (tried on linux desktop, works properly showing this is logitech g29)
    productId: 49743,
    path: '0001:000d:01',
    release: 35072,
    interface: 1 },
  { vendorId: 1121,
    productId: 16,
    path: '0001:0005:00',
    release: 260,
    interface: 0 },
  { vendorId: 1121,
    productId: 16,
    path: '0001:0005:01',
    release: 260,
    interface: 1 },
  { vendorId: 1121,
    productId: 20010,
    path: '0001:0004:00',
    release: 256,
    interface: 0 } ]

todbot, 2018-01-18:
The error is `Error: cannot open device with vendor id 0x1d34 and product id 0xd`
So it looks like that device is not plugged in. 
1. Does it show up in `lsusb`?
2. Do you have proper udev rules for it or are running the script as the `root` user?
3. What exactly is this device?

LawKong, 2018-01-18:
1. How do you make it show up in `lsusb`?
2. I'm not sure as well for the udev rules. I'm running it in `pi@raspberrypi:~ $`
3. This is the Logitech G29 Steering wheel and Gas Pedal

todbot, 2018-01-18:
Hi @LawKong,

1. `lsusb` is a common Linux command-line tool for showing what USB devices are connected to your computer. I think it may be installed by default on Raspbian but if not you can get it by doing `sudo apt install usbutils`.

2. You should read up a bit on how to set udev permissions for usb devices.  You can see an example here: https://github.com/node-hid/node-hid#udev-device-permissions

3. For the Logitech G29 wheel you should check out the `logitech-g29` Node.js package.: https://github.com/ForestMist/logitech-g29  He also provides an example udev rule.

devyaz, 2018-04-04:
i get the same Error
`C:\Users\Martin Mtawali\Desktop\Electron\Maravitech-POS\node_modules\bindings\bindings.js:96 Uncaught Error: Could not locate the bindings file. Tried:
 → C:\Users\Martin Mtawali\Desktop\Electron\Maravitech-POS\node_modules\node-hid\build\HID.node
 → C:\Users\Martin Mtawali\Desktop\Electron\Maravitech-POS\node_modules\node-hid\build\Debug\HID.node
 → C:\Users\Martin Mtawali\Desktop\Electron\Maravitech-POS\node_modules\node-hid\build\Release\HID.node
 → C:\Users\Martin Mtawali\Desktop\Electron\Maravitech-POS\node_modules\node-hid\out\Debug\HID.node
 → C:\Users\Martin Mtawali\Desktop\Electron\Maravitech-POS\node_modules\node-hid\Debug\HID.node
 → C:\Users\Martin Mtawali\Desktop\Electron\Maravitech-POS\node_modules\node-hid\out\Release\HID.node
 → C:\Users\Martin Mtawali\Desktop\Electron\Maravitech-POS\node_modules\node-hid\Release\HID.node
 → C:\Users\Martin Mtawali\Desktop\Electron\Maravitech-POS\node_modules\node-hid\build\default\HID.node
 → C:\Users\Martin Mtawali\Desktop\Electron\Maravitech-POS\node_modules\node-hid\compiled\8.2.1\win32\x64\HID.node
    at bindings (C:\Users\Martin Mtawali\Desktop\Electron\Maravitech-POS\node_modules\bindings\bindings.js:93)
    at loadBinding (C:\Users\Martin Mtawali\Desktop\Electron\Maravitech-POS\node_modules\node-hid\nodehid.js:16)
    at Object.showdevices [as devices] (C:\Users\Martin Mtawali\Desktop\Electron\Maravitech-POS\node_modules\node-hid\nodehid.js:132)
    at window.onload (index.js:6)`
in an Electron app  any ideas?

todbot, 2018-04-04:
Hi @devyaz,
Your issue sounds like you're not running `electron-rebuild` as described here:
https://github.com/node-hid/node-hid#using-node-hid-in-electron-projects

closed 2018-04-04 by todbot

devyaz, 2018-04-06:
Hi @todbot  i did as per your instruction here is the output
`> electron-rebuild --force

An unhandled error occurred inside electron-rebuild
Warning: while converting VCCLCompilerTool/BasicRuntimeChecks to MSBuild, index value (-1) not in expected range [0, 4)
Warning: while converting VCCLCompilerTool/RuntimeLibrary to MSBuild, index value (-1) not in expected range [0, 4)
Warning: while converting VCCLCompilerTool/RuntimeTypeInfo to MSBuild, expected bool; got '-1'
Warning: while converting VCCLCompilerTool/RuntimeLibrary to MSBuild, index value (-1) not in expected range [0, 4)
Warning: while converting VCCLCompilerTool/RuntimeTypeInfo to MSBuild, expected bool; got '-1'
Warning: while converting VCCLCompilerTool/BasicRuntimeChecks to MSBuild, index value (-1) not in expected range [0, 4)
Warning: while converting VCCLCompilerTool/BasicRuntimeChecks to MSBuild, index value (-1) not in expected range [0, 4)
Warning: while converting VCCLCompilerTool/RuntimeTypeInfo to MSBuild, expected bool; got '-1'
Warning: while converting VCCLCompilerTool/RuntimeLibrary to MSBuild, index value (-1) not in expected range [0, 4)
Warning: while converting VCCLCompilerTool/RuntimeTypeInfo to MSBuild, expected bool; got '-1'
Warning: while converting VCCLCompilerTool/BasicRuntimeChecks to MSBuild, index value (-1) not in expected range [0, 4)
Building the projects in this solution one at a time. To enable parallel build, please add the "/m" switch.
C:\Users\Martin Mtawali\Desktop\Electron\Maravitech-POS\node_modules\electron-edge\build\edge_nativeclr.vcxproj(20,3): error MSB4019: The imported project "C:\Microsoft.Cpp.Default.props" was not found. Confirm that the path in the <Import> declaration is correct, and that the file exists on disk.
C:\Users\Martin Mtawali\Desktop\Electron\Maravitech-POS\node_modules\electron-edge\build\edge_coreclr.vcxproj(20,3): error MSB4019: The imported project "C:\Microsoft.Cpp.Default.props" was not found. Confirm that the path in the <Import> declaration is correct, and that the file exists on disk.
gyp ERR! build error
gyp ERR! stack Error: `C:\Windows\Microsoft.NET\Framework\v4.0.30319\msbuild.exe` failed with exit code: 1
gyp ERR! stack     at ChildProcess.onExit (C:\Users\Martin Mtawali\Desktop\Electron\Maravitech-POS\node_modules\node-gyp\lib\build.js:258:23)
gyp ERR! stack     at emitTwo (events.js:106:13)
gyp ERR! stack     at ChildProcess.emit (events.js:192:7)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:215:12)
gyp ERR! System Windows_NT 6.1.7601
gyp ERR! command "C:\\Program Files\\nodejs\\node.exe" "C:\\Users\\Martin Mtawali\\Desktop\\Electron\\Maravitech-POS\\node_modules\\node-gyp\\bin\\node-gyp.js" "rebuild" "--target=1.8.4" "--arch=x64" "--dist-url=https://atom.io/download/electron" "--build-from-source"
gyp ERR! cwd C:\Users\Martin Mtawali\Desktop\Electron\Maravitech-POS\node_modules\electron-edge
gyp ERR! node -v v7.6.0
gyp ERR! node-gyp -v v3.6.2
gyp ERR! not ok

Failed with exit code: 1

Error: Warning: while converting VCCLCompilerTool/BasicRuntimeChecks to MSBuild, index value (-1) not in expected range [0, 4)
Warning: while converting VCCLCompilerTool/RuntimeLibrary to MSBuild, index value (-1) not in expected range [0, 4)
Warning: while converting VCCLCompilerTool/RuntimeTypeInfo to MSBuild, expected bool; got '-1'
Warning: while converting VCCLCompilerTool/RuntimeLibrary to MSBuild, index value (-1) not in expected range [0, 4)
Warning: while converting VCCLCompilerTool/RuntimeTypeInfo to MSBuild, expected bool; got '-1'
Warning: while converting VCCLCompilerTool/BasicRuntimeChecks to MSBuild, index value (-1) not in expected range [0, 4)
Warning: while converting VCCLCompilerTool/BasicRuntimeChecks to MSBuild, index value (-1) not in expected range [0, 4)
Warning: while converting VCCLCompilerTool/RuntimeTypeInfo to MSBuild, expected bool; got '-1'
Warning: while converting VCCLCompilerTool/RuntimeLibrary to MSBuild, index value (-1) not in expected range [0, 4)
Warning: while converting VCCLCompilerTool/RuntimeTypeInfo to MSBuild, expected bool; got '-1'
Warning: while converting VCCLCompilerTool/BasicRuntimeChecks to MSBuild, index value (-1) not in expected range [0, 4)
Building the projects in this solution one at a time. To enable parallel build, please add the "/m" switch.
C:\Users\Martin Mtawali\Desktop\Electron\Maravitech-POS\node_modules\electron-edge\build\edge_nativeclr.vcxproj(20,3): error MSB4019: The imported project "C:\Microsoft.Cpp.Default.props" was not found. Confirm that the path in the <Import> declaration is correct, and that the file exists on disk.
C:\Users\Martin Mtawali\Desktop\Electron\Maravitech-POS\node_modules\electron-edge\build\edge_coreclr.vcxproj(20,3): error MSB4019: The imported project "C:\Microsoft.Cpp.Default.props" was not found. Confirm that the path in the <Import> declaration is correct, and that the file exists on disk.
gyp ERR! build error
gyp ERR! stack Error: `C:\Windows\Microsoft.NET\Framework\v4.0.30319\msbuild.exe` failed with exit code: 1
gyp ERR! stack     at ChildProcess.onExit (C:\Users\Martin Mtawali\Desktop\Electron\Maravitech-POS\node_modules\node-gyp\lib\build.js:258:23)
gyp ERR! stack     at emitTwo (events.js:106:13)
gyp ERR! stack     at ChildProcess.emit (events.js:192:7)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:215:12)
gyp ERR! System Windows_NT 6.1.7601
[... cut ...]
npm ERR! errno 4294967295
npm ERR! Maravitech-POS@1.0.0 postinstall: `electron-rebuild --force`
npm ERR! Exit status 4294967295
npm ERR!
npm ERR! Failed at the Maravitech-POS@1.0.0 postinstall script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     C:\Users\Martin Mtawali\AppData\Roaming\npm-cache\_logs\2018-04-06T07_11_44_415Z-debug.log
` maybe i should remove the node modules folder and  do `npm i`?