node-hid/node-hid #239: I can't get any device!

mahdi-gh1373, 2017-12-19:
I write my code like this:

```
var HID = require('node-hid');
var devices = HID.devices();
console.log(devices)
```
But my log Show: [ ]
What is the reason?

todbot, 2017-12-19:
Hi,
We'll need more information to diagnose the issue your'e seeing.
For instance,
- What kind of computer are you running?
- What specific Linux distribution are you running?
- What USB HID devices do you have plugged in?
- What does 'lsusb' print out?

mahdi-gh1373, 2017-12-20:
Hi todbot,
I use linux on my odroid device and I want to write some data on my usb flash drive.

```
root@odroid:~# lsusb
Bus 001 Device 010: ID 1976:6025 Chipsbrand Microelectronics (HK) Co., Ltd. Flash Drive 512 MB
Bus 001 Device 003: ID 0458:708f KYE Systems Corp. (Mouse Systems)
Bus 001 Device 002: ID 05e3:0610 Genesys Logic, Inc. 4-port hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

todbot, 2017-12-20:
Hi, I'm still unclear what Linux distribution you're running, but it seems like perhaps the native library part of node-hid isn't compiled correctly for your odroid.

So I would try to install node-hid in your project with:
```
  npm install node-hid --build-from-source
```
This will cause npm to build node-hid from the ground up instead of relying on pre-built modules.  Note you will need all the pre-requisites for compiling as described here:
https://github.com/node-hid/node-hid#compiling-from-source

todbot, 2018-01-17:
Closing due to no activity.

closed 2018-01-17 by todbot

Kainos-SU, 2022-02-24:
I am sory, maybe I am not undestand some thing, but I have connected thermal usb printer to computer, entering command "lsusb" I can see that device is connected? but when I run script to find it via node I get a lot of duplocated devices but none of them is device I am looking for.

Code  I am running:
`
import HID from "node-hid";

HID.devices().forEach((element, index) => {
    console.log(`element number ${index} - 0x${element.vendorId.toString(16)}\n`, element);
});
`
My package.json:
`
{
  "name": "usb_test",
  "version": "1.0.0",
  "main": "index.js",
  "type": "module",
  "author": "",
  "license": "ISC",
  "description": "",
  "devDependencies": {},
  "scripts": {
    "test": "echo \"Error: no test specified\" && exit 1"
  },
  "dependencies": {
    "node-hid": "^2.1.1"
  },
}
`

I try to run
`npm install node-hid --build-from-source`
but it doesn work. Please I dont understand what I am doing wrong/

todbot, 2022-02-24:
Printers are not HID devices, so `node-hid` cannot see them.  If you're on Linux, you should see in `dmesg` what driver  gets attached to the printer when you plug it in. For HID devices, that is HIDRAW and you'll see a device `/dev/hidraw*` be created.  

Perhaps you should look at `node-usb` instead or for printing, just use standard unix printing techniques

Kainos-SU, 2022-02-24:
Thank for answer, I try to use `node-usb` but get error of access, so I try alternative library. Sorry for distubing.