node-hid/node-hid #240: Build node-hid for use with electron, but now jest fails.

obiwanjacobi, 2017-12-23:
Hi,

As I was building up my app I had some integration tests working (jest) that tested my logic against the connected device - using node-hid. Worked fine.
Then I wanted to integrate that logic into the electron app and I had to rebuild - using the instructions in the read-me (that are not quite up-to-date anymore, btw) I got that working. My app logic execute fine in electron's main process.

However, the jest integration tests are now failing with a message "The specified module could not be found" followed by a path to a file: "\\?\C:[full path to my project root]\node_modules\node-hid\build\Release\HID.node". The funny thing is that the file does exist, so my thought was that perhaps it could not read it correctly (because of the rebuild?)...?

Anyway, as this is my first TypeScript/JavaScript/electron/node-hid/react/etc. project, I am a bit lost as to where to look next. Any suggestions?

PS: I am aware that this may not even be a node-hid issue (but a jest issue) but this still seemed to logical place to start my investigation.

I'm on Windows 10.

todbot, 2017-12-23:
Hi @obiwanjacobi,

Do you have a link to your project I can check out and try to help debug?

I assume you are using `electron-rebuild`?  You'll need that to get the appropriate native module for `node-hid` in Electron. And if you're using a packager (is that what jest is?), that packager must have a way of finding the native module.

Also, which bits of the README are out-of-date?  I'll get that fixed ASAP and apologies for the confusion.

obiwanjacobi, 2017-12-25:
No sorry, no public git repo yet. But if you drop me an email (obiwanjacobi@hotmail.com) I can zip you the project (I could not find your email). 

The readme says:
> In your electron project, add electron-rebuild and electron-prebuilt to your devDependencies

But electron-prebuild did not work / was not available. But without it, it still works. I did have 'electron' as a dependency though.

obiwanjacobi, 2017-12-28:
I have isolated the problem in a separate project:
https://github.com/obiwanjacobi/electron-hid-jest-problem

HTH

todbot, 2017-12-28:
Hi @obiwanjacobi,
Thanks this is a great base to start from. I just tried cloning and building it and a big problem I see is that there's no `Program.js` (the `main`  in package.json) and I don't see a script to build it.  So the error I get is "Unable to find Electron app  at electron-hid-jest-problem".

todbot, 2017-12-28:
Okay I think I found a solution to your issue.  
Add a new script to your `package.json`:
```
    "build" : "tsc && electron-rebuild",
```

This will both build the Typescript and run electron-rebuild.  I'm not sure why you have `__postinstall` as a script (it should be just `postinstall`) but I've found having an explicit rebuild script to be super handy.

obiwanjacobi, 2017-12-29:
I don't understand how that works. I execute 'npm run build' (because 'npm build' is reserved) but the problem still exists.
Also, do I have to run this every time I want to run the app?
I use tsc -w to transpile on the fly and have a jest plugin in vscode to continuously run the unit tests...
Can you tell more about the background of this problem, so I may understand and perhaps make a different choice in unit testing framework. I am still early in the project so that is still an option.

Thanks.

todbot, 2017-12-29:
The `tsc` compile needs to be done any time you edit your .ts files to turn them into .js for use by Node or Electron.  I'm sure there are boilerplate examples showing how to do 'hot-reloading" for Typescript.  Here's one for Typescript + React + Electron https://github.com/iRath96/electron-react-typescript-boilerplate   

But it seems like most Electron + Typescript examples I see have an explicit "build" step that invokes `tsc` like this: https://github.com/electron/electron-quick-start-typescript

The `electron-rebuild` step is only done once, after npm has installed Electron and all other Node modules.  In theory this should go in the special `"postinstall"` script (and it's what I do in the [electron-hid-toy](https://github.com/todbot/electron-hid-toy) example.  For a while the special "postinstall" step didn't work correctly for `node-hid` depending on what version of npm/Node/Electron you had, but now I think it's okay. 

The reason you need `electron-rebuild` at all is there are two different execution environments that node modules are "built" in: the Node version used by `npm` and the Node built into Electron.  In a different universe, there could have been an "electron-npm" that built everything using the Electron Node. But most Node modules don't care about which version of Node they run in so most people don't care.

obiwanjacobi, 2017-12-30:
Ok, so I can have a postinstall script which does the tsc trans/compile and the electron-rebuild. Do I have to do an npm install for that to run (automatically after npm is finished installing) -or- can I just trigger it separately (and just execute tsc & electron-rebuild)?
I have ran it separately and that didn't fix the jest issue...

todbot, 2017-12-30:
Yes, the problem is that jest is using your system Node and Electron is using its own version of Node.  You need to tell jest to use Electron for its tests.

It looks like someone had a similar issue here: https://github.com/facebook/jest/issues/3698
and a solution was offered. 

I made this change to my own checkout of `electron-hid-jest-problem` and now my "scripts" look like:
```
  "scripts": {
    "start": "electron .",
    "test": "ELECTRON_RUN_AS_NODE=true ./node_modules/.bin/electron ./node_modules/.bin/jest",
    "build": "tsc",
    "postinstall": "electron-rebuild --force"
  },

And the jest tests pass when I do "npm run test".  I can edit files, run `npm run build` , and then `npm run test` again or `npm run start` and everything seems to work as expected.

closed 2017-12-30 by todbot