node-hid/node-hid #241: Cannot open device with vendor id 0x57e and product id 0x337

fudgepop01, 2018-01-04:
What I'm trying to do is read data from a mayflash Gamecube Controller adapter in WiiIU mode, which has the following data when read in my terminal when I get a list of the HID devices with the `ioreg -p IOUSB -w0 -l` command:

>+-o WUP-028@14600000  <class AppleUSBDevice, id 0x100001d58, registered, matched, active, busy 0 (44 ms), retain 14>
        {
          "sessionID" = 53926682243641
          "iManufacturer" = 1
          "bNumConfigurations" = 1
          "idProduct" = 823
          "bcdDevice" = 256
          "Bus Power Available" = 250
          "USB Address" = 13
          "bMaxPacketSize0" = 64
          "iProduct" = 2
          "iSerialNumber" = 3
          "bDeviceClass" = 0
          "Built-In" = No
          "locationID" = 341835776
          "bDeviceSubClass" = 0
          "bcdUSB" = 512
          "USB Product Name" = "WUP-028"
          "PortNum" = 6
          "non-removable" = "no"
          "IOCFPlugInTypes" = {"9dc7b780-9ec0-11d4-a54f-000a27052861"="IOUSBFamily.kext/Contents/PlugIns/IOUSBLib.bundle"}
          "bDeviceProtocol" = 0
          "IOUserClientClass" = "IOUSBDeviceUserClientV2"
          "IOPowerManagement" = {"DevicePowerState"=0,"CurrentPowerState"=3,"CapabilityFlags"=65536,"MaxPowerState"=4,"DriverPowerState"=3}
          "kUSBCurrentConfiguration" = 1
          "Device Speed" = 1
          "USB Vendor Name" = "Nintendo"
          "idVendor" = 1406
          "IOGeneralInterest" = "IOCommand is not serializable"
          "USB Serial Number" = "15/07/2014"
          "IOClassNameOverride" = "IOUSBDevice"
        }

Here's the stack trace:

>/Users/Dominick/Desktop/projects/GCC-display/node_modules/node-hid/nodehid.js:51
    this._raw = new (Function.prototype.bind.apply(binding.HID,
                ^

>Error: cannot open device with vendor id 0x57e and product id 0x337
    at new HID (/Users/Dominick/Desktop/projects/GCC-display/node_modules/node-hid/nodehid.js:51:17)
    at Object.<anonymous> (/Users/Dominick/Desktop/projects/GCC-display/index02.js:5:14)
    at Module._compile (module.js:635:30)
    at Object.Module._extensions..js (module.js:646:10)
    at Module.load (module.js:554:32)
    at tryModuleLoad (module.js:497:12)
    at Function.Module._load (module.js:489:3)
    at Function.Module.runMain (module.js:676:10)
    at startup (bootstrap_node.js:187:16)
    at bootstrap_node.js:608:3

I'm working with the 'electron-react-boilerplate' to create an electron application. I've received the data before. IIRC the data it sends is a buffer of 21 hexadecimal values which i've been able to break down into binary and decode in the past. 

Strangely enough, this was working just fine yesterday, but as soon as I went to rebuild with electron-rebuild, it stopped working across all of my programs that utilize it.

todbot, 2018-01-04:
Hi,
Does the device show up to node-hid with `HID.devices()`?  If it doesn't `node-hid` can't access it.

What OS are you on?  If Linux, then you'll need to run as root or set up a udev rule.

If the device worked before but doesn't know, that sounds like another process is grabbing hold of the device? Do you use this controller with any other software?  Or perhaps a zombie process of yours from previously?  Do a "ps auwwx" or look through Activity Monitor for process remnants.

The only thing `electron-rebuild` does is rebuild the native part of `node-hid` so it works with the Node inside of Electron.  If you think that's an issue, you can use eliminate it as a possibility with a separate project that just uses Node.

todbot, 2018-01-17:
Closing from non-activity. Sounds like it worked before.

closed 2018-01-17 by todbot

edwincabezas7, 2020-06-19:
> Hi,
> Does the device show up to `node-hid` with `HID.devices()`? If it doesn't `node-hid` can't access it.
>
> What OS are you on? If Linux, then you'll need to run as root or set up a udev rule.
>
> If [...]

In my case I have to run my program with priviligies (sudo). Thanks !

smohanty92, 2023-07-16:
I'm also trying read from GCM controller. I can get it to open, but I just get a random stream of bytes coming in without me even pressing a btn