node-hid/node-hid #243: Card reader connected using USB serial converter cable not listed in device list

camal4u, 2018-01-17:
I need to read data from a 2D barcode reader and mag stripe reader connected over USB. I use this device [E-SEEK MODEL 250 ](http://store.idscan.net/ESeek-M250-ID-2D-Barcode-Magnetic-Stripe-Card-Reader-USB-REFURBISHED_p_2.html) which is provided with a USB-Serial converter cable.

I have installed the necessary drivers on windows and the device is listed in device manager when connected.

But its not listed in the below line of code
`var devices = HID.devices();`

Can you please help me out ?

todbot, 2018-01-17:
Hi @camal4u,
That device is a USB Serial device not a USB HID device.  `node-hid` cannot see it or use it.
After you've installed the drivers you mentioned, you should see a virtual serial port in the device manager. Find out the COM port number and use it with `node-serialport`.

closed 2018-01-17 by todbot

camal4u, 2018-01-18:
hi @todbot 
Thanks a ton for your help.